ksheumaker/homeassistant-apsystems_ecur #4: Add plugin to HACS

ksheumaker, 2021-03-09:
Investigate how HACS works, and get this integration included as part of HACS to make it easier to install and update for end users.

willemstoker, 2021-03-29:
Why not make it a usual integration in Home Assistant?

bgbraga, 2021-04-11:
Please, if you could change this integration folders we could add it as extra repository at HACS.

The structure should be something like:
custom_components/apsystems_ecur/__init_.py
custom_components/apsystems_ecur/sensor.py
custom_components/apsystems_ecur/manifest.json
info.md
README.md

https://hacs.xyz/docs/publish/integration/

ksheumaker, 2021-04-12:
@bgbraga Ok, this should be easy enough - i'll get an update done tonight to move everything

ksheumaker, 2021-07-13:
Directory structure changed, will work with HACS now.

closed 2021-07-13 by ksheumaker