ksheumaker/homeassistant-apsystems_ecur #6: Yc1000 compatibility

Shadyville08, 2021-04-04:
Support for Yc1000 would be great.

86hakan, 2021-04-04:
I'm also interested in YC1000 compatibility. Now when I enter my IP number for ECU-C, the home assistant does not want to start after that.

So support for YC1000 and ECU-C would be great.

HAEdwin, 2021-04-11:
Kyle and I do not own a YC1000 and my knowledge of Python is limited. Despite that based on the frame of the YC1000, I made an attempt to make the integration compatible with the YC1000. Would you like to test the integration from my fork? https://github.com/HAEdwin/homeassistant-apsystems_ecur

86hakan, 2021-04-11:
Absolutly i can test it, can i download from the link?

//
Håkan Gundstedt
0735 - 64 34 58

HAEdwin, 2021-04-11:
Follow hyperlink and choose Download Code (in zip format for example) and follow instructions.

Shadyville08, 2021-04-11:
> Kyle and I do not own a YC1000 (or ECU-C) and my knowledge of Python is limited. Despite that based on the frame of the YC1000, I made an attempt to make the ECU-R integration compatible with the YC1000. Would you like to test the integration from my fork? https://github.com/HAEdwin/homeassistant-apsystems_ecur

Thank you! I'm currently not producing any power (snowing outside..) but get lifetime energy and today energy in HA. I get the voltages from the inverters but frequency, temperature and signal says zero. Those values maybe only update when the panels are producing?

HAEdwin, 2021-04-11:
I am not sure. If you take a look at your entities and filter on "ecu" you will notice a entity "ECU Querying Enabled". We use a service call which en- or disables the querying service depending on sunrise and sunset. Service call is: apsystems_ecur.start_query or apsystems_ecur.stop_query. "ECU Querying Enabled" shows the status.

Shadyville08, 2021-04-11:
I'm not seeing any "ECU Querying Enabled" in entities. I see "current power" and so on and also "ECU Using Cached Data".
However I do find the Querying service in developer tools. Guess I might just wait until tomorrow to see if anything updates when the system are producing.

HAEdwin, 2021-04-12:
And? does it work? @86hakan does this work for you also (having a ECU-C)?

Shadyville08, 2021-04-12:
It works! Thank you!!!

86hakan, 2021-04-12:
Im testing tomorrow. Have lot of work today.

//
Håkan Gundstedt

HAEdwin, 2021-04-12:
@Shadyville08 Great to hear that, so if all is working and checked I will do a pull request.

ksheumaker, 2021-04-12:
PR approved, thanks @HAEdwin for adding that!

86hakan, 2021-04-13:
It doesnt work with my ecu. Home assistant wont start with this.

HAEdwin, 2021-04-13:
@86hakan Too bad, that does surprise me because the compatibility table and the use of the same ECUapp to configure the ECU-R or ECU-C suggest otherwise. 
[image]

If all requirements have been met, we conclude that this ECU-R integration does not work for the ECU-C, thank you for testing. This ECU-R integration was made possible because contributers all own a ECU-R (and a YC600, QS1 or YC1000).

However, I do want to help you, which is why I am raising this in the HA forum. it seems to me the appropriate medium for now to investigate this new wish. Feel free to discuss and contribute to that discussion: https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835 Because this issue in Github is officially closed I'll continue in the HA forum.

HAEdwin, 2021-04-13:
@86hakan You could try the following: Download and install packet sender (packetsender.com) on windows. Add the following parameters 
[image]
Then press [send].
You should get te following:
[image]
Does this work for you until this stage?

86hakan, 2021-04-13:
It works!
[image: image.png]

HAEdwin, 2021-04-13:
Image is missing, could you post the entire string in response to string APS1100160001END?  That string should start with: APS11009.... and ends with END

HAEdwin, 2021-04-13:
Oh, I forgot to warn you! Don't try random commands here. It can really mess up the ECU configuration.

86hakan, 2021-04-14:
APS120115000121500000392501\00\02@\c4\00\00$\e4\00\00\14\f2
!\04\14\13GW\00\0b\00\0b10014ECU-C-Z_C1.1.8016Europe/Stockholm\80\97\1b\01b\15`\c5\a8\e6\d5\fc00\00\00\00\00\00\00\00\00\00\00END\n

HAEdwin, 2021-04-14:
Great! Your ECUC-ID = 215000003925
Now try to send this command to query the ECU-C for registered inverters: APS1100280002215000003925END
Can you post me the results again?

86hakan, 2021-04-14:
APS11032600020001\00\0b !\04\14\15\08\05P \00\00b#\0102\01\f5\00\8c\01\02\00\ec\01\01\00\ee\00\fe\00\ea\00\fdP \00\00g\15\0102\01\f4\00\8b\01\02\00\ed\01\02\00\ef\00\f9\00\ed\00\fcP \00\00iU\0102\01\f4\00\92\00\fe\00\ed\00\fa\00\ef\00\f9\00\ea\00\f8P \00\00rs\0102\01\f3\00\8f\00\fe\00\ec\01\01\00\ef\01\02\00\eb\01\00P \00\00\83I\0102\01\f4\00\97\00\fc\00\eb\01\00\00\ef\00\fb\00\eb\00\fbP \00\00\87)\0102\01\f4\00\8d\00\fa\00\ec\00\fc\00\ee\00\f7\00\ea\00\f8P \00\00cB\0102\01\f4\00\8d\00\fa\00\ee\01\02\00\f0\01\00\00\ec\00\f9P \00\00gW\0102\01\f4\00\8f\01\01\00\ed\01\07\00\ee\00\fd\00\ec\00\faP \00\00rc\0102\01\f4\00\92\01\01\00\ee\01\03\00\ee\01\00\00\ec\00\feP \00\00t2\0102\01\f3\00\8a\01\04\00\ed\01\05\00\ee\01\01\00\eb\00\feP \00\00\85A\0102\01\f4\00\8c\00\e9\00\ed\00\ed\00\f0\00\e3\00\ec\00\e4END\n

HAEdwin, 2021-04-14:
Ok, sorry.. for interpretation I now need the hex data, not the ASCII data. I think it's easier to save the log and post that here.

86hakan, 2021-04-14:
[trafficlog.log](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/6311171/trafficlog.log)

HAEdwin, 2021-04-14:
The tests previously done by Hakan of sending the commands to the ECU-C prove that querying the ECU-C works with the same commands. The query on the ECU brought valid data which I analyzed to check that. It turns out that there are 11 YC1000 inverters (44 panels) configured in the ECU-C. As a result, the string length of inverter_raw_data becomes 326 bytes. The data frame also meets all the requirements of the integration on part of the YC1000 dataframe specs. Actually, it should work. Is it likely that the data arrives fragmented and is therefore incomplete? Of course it is not much, but TCP transfers can still be fragmented to smaller chunks of data [code line: self.ecu_raw_data = sock.recv(self.recv_size)]. More on receive methods can be found here: https://stackoverflow.com/questions/17667903/python-socket-receive-large-amount-of-data

Here is the raw data converted to hex. To debug, I think assigning this to ecu_raw_data would prove that the data can be interpreted by the integration. 4150533131303332363030303230303031000B2021041415380550200000622301303201F50089006A00E6006A00E7006900E3006850200000671501303201F40085006500E7006600E8006200E5006350200000695501303201F4008E006700E7006500E9006500E4006450200000727301303201F4008C006900E7006900E9006900E4006850200000834901303201F50092006800E7006800E9006600E5006650200000872901303201F4008B006600E7006700E8006400E4006450200000634201303201F40089006700E8006900E9006800E4006550200000675701303201F40089006800E8006900E8006600E4006550200000726301303201F5008E006900E7006900E8006800E4006750200000743201303201F40087006A00E8006900E9006800E4006750200000854101303201F50089006500E8006500EA006400E40064454E440A

HAEdwin, 2021-04-16:
The string containing the hex data is ultimately much larger after conversion of the received data than should be. So the checksum fails. Somewhere the conversion of the received data to hex data is not going well for  all pairs in case of the ECU-C.

What is being received:
APS11032600020001\00\0b !\04\14\158\05P \00\00b#\0102\01\f5\00\89\00j\00\e6\00j\00\e7\00i\00\e3\00hP ..... etc
What is being converted by the integration:
41 50 53 31 31 30 33 32 36 30 30 30 32 30 30 30 31 00 00 62 ..... etc
What it should be (and how Packet Sender converts it):
41 50 53 31 31 30 33 32 36 30 30 30 32 30 30 30 31 00 0B ..... etc

Somewhere down the line of binascii.b2a_hex things must go wrong.

HAEdwin, 2021-04-16:
I noticed that the signal is zero on all inverters. Same goes for that part I guess.

HAEdwin, 2021-04-19:
@86hakan If you want, you can download APSystemsECUR.py from my GitHub fork (https://github.com/HAEdwin/homeassistant-apsystems_ecur) to try if the integration now works with your ECU-C. Hopefully it will work but it is hard to really test because I don't own a ECU-C. If things go wrong please post the home-assistant.log here. To provide more feedback you can add:
`logger:
    default: warning
    logs:
        custom_components.apsystems_ecur: debug`
to the configuration.yaml file. After installation restart core to take changes into effect.

86hakan, 2021-04-25:
I have test now, but i get error-message for configfile for the log.
Invalid config for [logger]: expected a dictionary for dictionary value @
data['logger']['logs']. Got None extra keys not allowed @
data['logger']['custom_components.apsystems_ecur']. Got 'debug'. (See
/config/configuration.yaml, line 110).
my code is:
logger:
    default: warning
    logs:
    custom_components.apsystems_ecur: debug

apsystems_ecur:
  host: 192.168.9.138

HAEdwin, 2021-04-25:
I'm not sure but the indent may be wrong here. Make sure indent is correct, like example below.
[image]

86hakan, 2021-04-25:
Thanks. That was the problem.

the only error-message i can se when im activate the plugin is
2021-04-25 12:00:39 WARNING (Recorder)
[homeassistant.components.recorder.util] The system could not validate that
the sqlite3 database at //config/home-assistant_v2.db was shutdown cleanly
2021-04-25 12:00:49 WARNING (MainThread) [homeassistant.setup] Setup of
recorder is taking over 10 seconds.
2021-04-25 12:01:39 WARNING (MainThread) [homeassistant.bootstrap] Waiting
on integrations to complete setup: recorder
2021-04-25 12:02:23 WARNING (Recorder) [homeassistant.components.recorder]
Ended unfinished session (id=20 from 2021-04-25 09:54:45.371389)

I dont know if this is a problem for the ap-system component.

HAEdwin, 2021-04-25:
No, it is not related. Always shutdown HA cleanly, do not pull the plug but do a proper shutdown. Now I hope you will be able to see a lot of entities which you can add to the dashboard (and that they have the correct values).

86hakan, 2021-04-25:
Ok. I just restart the core, but now im do a correct shutdown and a
powerup, im runnning this in a virtual box on a mac and my kernel_task is
going for 2,39G RAM  when im start the homeassistant.

HAEdwin, 2021-04-25:
And? I'm curious of course. Can you share the log if it doesn't go well?

86hakan, 2021-04-25:
It doesnt got well. Do you know how i can read the log even if homeassist doesnt start up?
Can logger write to a specific file?

//
Håkan Gundstedt
0735 - 64 34 58

HAEdwin, 2021-04-25:
See if you can manage to access the home-assistant.log, it should be in the root of the config folder. Running HA on a virtual box on the MAC does bring a whole new dimension with it (trouble shooting I mean).

normanp2pl, 2021-08-10:
Hi @HAEdwin
I've a 6 QS1 (with 4 panels on each) with ECU-C. I just installed your extension and not working. HA won't start. Hangs on :

> 2021-08-10 14:24:13 DEBUG (MainThread) [custom_components.apsystems_ecur] Querying ECU
> 2021-08-10 14:24:13 INFO (MainThread) [custom_components.apsystems_ecur.APSystemsECUR] Connected to 192.168.1.220 8899

When I change address to invalid one, I have :

> 2021-08-10 14:29:26 WARNING (MainThread) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Error: [Errno 113] Connect call failed ('192.168.1.210', 8899)
> 2021-08-10 14:29:26 DEBUG (MainThread) [custom_components.apsystems_ecur] Returning {'data_from_cache': True, 'querying': True}
> 2021-08-10 14:29:26 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 4.701 seconds

when I'm querying my ECU-C with 1st python script from https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835 I have (i added printout of response from ECU-C) :

> root@hasiu:/srv/homeautomation/hass-config# python3 /home/norman/test.py
> b'415053313130313637303030323030303100062021081014294180600000461401303301f400a2009c00eb009d00a1009f80600000365801303301f40097007b00ea007c0083008180600000396601303301f4009a007900eb007c0080007e80600000582901303301f40090007800eb00780000000080600000433601303301f40099007600ea0077007b007c80600000587901303301f4009d007400ea0078007a007b454e440a'
> { "Time": "b'20210810142941'", "inv1": "b'806000004614'", "inv1on": "True", "inv1freq": "50.0", "inv1temp": "62", "inv1powerA": "156", "inv1volt": "235", "inv1powerB": "157"}

So my ECU-C is working well ?

HAEdwin, 2021-08-10:
Yes, probably. The integration was initially written for the ECU-R (by @ksheumaker) not for the ECU-C. However, I have noticed that a small adjustment of the data conversion could ensure compatibility with the ECU-C. However, I lack the time and Python knowledge to tackle that. Much of the analysis I've already done, maybe someone else can contribute to this. A binascii.hexlify(aStr) function and then trimming the datastring can probably make things compatible. Contrary to the ECU-R the received string of the ECU-C does not use the \x escape sequence.

ksheumaker, 2021-08-10:
The latest version already supports the 806 QS1 inverter, we had someone with an ECU-R report that.  We also have code in that shutdown and reopens connections when using the ECU_R_PRO device instead of the ECU_R.  If you take a look at line 115 of APSystemsECUR.py

`
        if "ECU_R_PRO" in self.firmware:
            self.writer.close()
            _LOGGER.info(f"Re-connecting to ECU_R_PRO on {self.ipaddr} {self.port}")
            self.reader, self.writer = await asyncio.open_connection(self.ipaddr, self.port)
`

That if statement could easily be altered to include the ECU_C if you let me know what value you get from self.firmware.

normanp2pl, 2021-08-11:
I changed that line yesterday and everything was working fine until now.
It's not related to this topic but my ECU-C won't respond anymore on any type of queries :/
I got:

> Error getting checksum int from 'ECU Query' data=b''

My debug says : 

> 1st send - sending     : APS1100160001END
> 
> 1st send - response : b''

Now I have sky full of sun so I won't do a restart. I'll restart ECU later and maybe it want to start work again :/

sk-net, 2021-08-22:
I am testing with ECU-C. I installed ECU-R integration, I modified lines 115 and 124 of APSystemsECUR.py to this:
if "ECU_R_PRO" in self.firmware or "ECU-C" in self.firmware:

First day it works and start collecting 3 parameters:
sensor.ecu_current_power
sensor.ecu_today_energy
sensor.ecu_lifetime_energy
and:
the stat of querying and cached data.
The next day there was no data at all and I am getting a status of all ecu entities as unavailable.

I enabled debugging and I have this logs:

Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/APSystemsECUR.py", line 210, in check_ecu_checksum
checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/__init__.py", line 72, in update
data = await self.ecu.async_query_ecu()
File "/config/custom_components/apsystems_ecur/APSystemsECUR.py", line 113, in async_query_ecu
self.process_ecu_data()
File "/config/custom_components/apsystems_ecur/APSystemsECUR.py", line 236, in process_ecu_data
self.check_ecu_checksum(data, "ECU Query")
File "/config/custom_components/apsystems_ecur/APSystemsECUR.py", line 213, in check_ecu_checksum
raise APSystemsInvalidData(f"Error getting checksum int from '{cmd}' data={debugdata}")
custom_components.apsystems_ecur.APSystemsECUR.APSystemsInvalidData: Error getting checksum int from 'ECU Query' data=b''
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 187, in _async_refresh
self.data = await self._async_update_data()
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 147, in _async_update_data
return await self.update_method()
File "/config/custom_components/apsystems_ecur/__init__.py", line 94, in update
raise Exception(f"Error using cached data for more than {self.cache_max} times.")
Exception: Error using cached data for more than 5 times.
2021-08-22 10:00:15 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 1.681 seconds
2021-08-22 10:05:15 DEBUG (MainThread) [custom_components.apsystems_ecur] Querying ECU
2021-08-22 10:05:15 INFO (MainThread) [custom_components.apsystems_ecur.APSystemsECUR] Connected to 192.168.1.65 8899
2021-08-22 10:05:15 WARNING (MainThread) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Error: Error getting checksum int from 'ECU Query' data=b''
2021-08-22 10:05:15 ERROR (MainThread) [custom_components.apsystems_ecur] Unexpected error fetching apsystems_ecur data: Error using cached data for more than 5 times.

normanp2pl, 2021-08-22:
> I am testing with ECU-C. I installed ECU-R integration, I modified lines 115 and 124 of APSystemsECUR.py to this:
> if "ECU_R_PRO" in self.firmware or "ECU-C" in self.firmware:
>
> First day it works and [...]

same story as mine. After couple of hours of normal ECU-C stops responds. I decided to gather data via HTTP which is available on ECU-C. More info at https://community.home-assistant.io/t/apsystems-aps-ecu-r-local-inverters-data-pull/260835/414?u=roman_piatkowski

sk-net, 2021-08-22:
I made one more test.
I removed ecur integration and restarted HA. Installed once again ecur integration with the modification in APSystemsECUR.py.
I set in the configuration file: scan_interval: 300
I am getting now :
binary_sensor.ecu_querying_enabled - on
binary_sensor.ecu_using_cached_data - on
sensor.ecu_current_power - unknown
sensor.ecu_today_energy - unknown
sensor.ecu_lifetime_energy – unknown

Log:

2021-08-22 10:38:09 DEBUG (MainThread) [custom_components.apsystems_ecur] Querying ECU
2021-08-22 10:38:24 INFO (MainThread) [custom_components.apsystems_ecur.APSystemsECUR] Connected to 192.168.1.65 8899
2021-08-22 10:38:24 WARNING (MainThread) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Error: Error getting checksum int from 'ECU Query' data=b''
2021-08-22 10:38:24 DEBUG (MainThread) [custom_components.apsystems_ecur] Returning {'data_from_cache': True, 'querying': True}
2021-08-22 10:38:24 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.305 seconds

normanp2pl, 2021-08-22:
Problem is in ECU-C not in integration or HA. It stops to respond after period of time. You must restart ECU-C in order to start work it again.
In my version of extension you got responses all time :) You don't have signal strength for inverters coz' http website dosn't provide this kind of data.
In the future we will be able to choose the version we want to use (I strongly believe in that :)) - socket 8899 version or http version.

HAEdwin, 2022-01-05:
You are right @normanp2pl. Although the ECU-C also has a web interface and your solution works fine, it would complete the integration if the 8899 method also works for the ECU-C. Temperature and signal strength data are then added as an advantage. How do you feel about my findings in this thread about the second and third responses to the query strings? _Contrary to the ECU-R the received string of the ECU-C does not use the \x escape sequence._

normanp2pl, 2022-01-06:
Hi @HAEdwin,
I'm using my own modified version of component since August 10, 2021 and it's works fine. I don't use 8899 method anymore.
I think that "\x escape sequence" is not a problem here. It's firmware problem in ECU-C - after couple of hours communication doesn't work anymore :/

nielssteenvoorde, 2022-02-02:
I have the same issue as @normanp2pl . I did explain in: [https://community.home-assistant.io/t/apsystems-sensor-for-home-assistant-not-visible/370041/20](url). My ECU-C stops after the night and doesn't come up. Interval in script is now 60. Should I change it in 300?  I was really impressed it al works the first time.  Really great job ;-)

ksheumaker, 2022-02-02:
I don't have an ECU C to test with, but if the ECU_R_PRO behavior works better for the C devices as well, I can certainly make it do that for both the C and R_PRO.  Can you provide me the exact text from the firmware version?  You should be able to see that in the new integration under "Devices > ECU XXXXX". Here's mine for an example:

[image]

When you add the integration you can adjust the interval, if it converted it from your configuration.yaml whatever `scan_interval` was will be migrated over.  If you need to adjust the interval you'll have to remove the integration and re-add it.  I will attempt to add a reconfiguration option in an upcoming release.

ksheumaker, 2022-02-03:
I released a new beta version 1.1.2 that changes the default interval to 300 seconds, and makes the ECU-C devices take the same code path as ECU_R_PROs and close the socket, sleep for a second and re-open when we do our 3 queries to get all the data required for the integration.  

Hopefully this will make it more stable for ECU-C users.

nielssteenvoorde, 2022-02-03:
The integration has worked for one day. Next day all integrations where off.  The firmware version of the ECU-C is: C1.2.0
Did upload a picture  of the firmware version on the HA forum:
[https://community.home-assistant.io/t/apsystems-sensor-for-home-assistant-not-visible/370041/24](url)

In HA it's not working anymore And i can't get it to fix. HA recently updated tot v.2022.2. I will try your beta version and review it ;-). Thanks again for this hard work @ksheumaker

nielssteenvoorde, 2022-02-03:
Did make a phonecall to APSystems here in the netherlands for which firmware my device should have. There are more firmware versions available for APSystem ECU's:
If I write down correct, it should be like this:
- ECU-B latest firmware: v1.2.19 (Cheaper ECU-R without a Ethernetport and limited for 4 solarpanels) 
- ECU-R latest firmware 1.2.19 (for models with serial ending 2160) (two antenna's)
- ECU-R latest firmware 2.0.4 (for models with serial ending 2162) (one antenna)
- ECU-C latest firmware C.1.2.0 (more models available. Older version has no powerbutton. Firmware updates itself when connected)
There are no ECU-R Pro's if I been wel informed. Is this helpful for making things more compatible?

ksheumaker, 2022-02-03:
Thanks for the info, very interesting.

I think people with ECU_R_PRO are showing up with the 2.0.4 firmware.  We are just pulling the firmware from what the device returns, so they might not call it a "PRO", but that's what the device reports.

I assumed the PRO with 2.0.4 firmware was a newer model, but they told you it's an older one?

For reference, my ECU-R is on the 1.2.19 firmware and was installed about a year and half ago.  The device reports itself as `ECU_R_1.2.19`

HAEdwin, 2022-02-03:
Although the ECU-R and ECU-R-PRO look exactly the same, they hold different firmware.
ECU-R
[image]
ECU-R-Pro
[image]
Difference is the certification for Sunspec modbus. We've seen that behaviours are very different between these two versions.

HAEdwin, 2022-02-03:
ECU-R-Pro is somewhere at firmware ECU_R_PRO_2.0.1016

ksheumaker, 2022-02-03:
Since the ECU-C does host it's own web-page that has all the production information creating a different integration that scrapes the web page might be more reliable for those devices.  My device doesn't have a webpage, so this reverse engineered protocol is all I've got.

Since the webpage is meant to be used by the customer (where this port we are connecting to clearly isn't) maybe APSsystems has put more effort in it being reliable and stable.

HAEdwin, 2022-02-03:
The ECU-R-Pro firmware also has it's own webpage where you can configure the RS485 modbus. It should not take long to make use of that although there is no serial modbus add-on available for HA yet. Look here for more info: https://gathering.tweakers.net/forum/list_messages/2032302/last

HAEdwin, 2022-02-03:
cmos6502 writes:
I have a still working integration with the ecu-r-pro. Has been running for a few days.
- I'm querying the lan port, not the wifi.
- In addition, I have my own 'version' of the sync_query in an async setup.
- and I have set a start/stop query on the sun status
- query per 300 seconds

nielssteenvoorde, 2022-02-03:
> 

Maybe I switched this two, was a litte bit confusing conversation. The diffence between this two was the antenna. The newer one has two antenna's i believe.   For the ECU-R there could be 2 diffent firmware versions (old and new), for the ECU-C only one and for the ECU-B als one.

HAEdwin, 2022-02-04:
I removed my comments because they where off topic, sorry about that,

ksheumaker, 2022-02-04:
Multiple changes were done in 1.1.2 to hopefully increase stability with ECU-C and ECU_R_PRO devices.  Please update to the latest version and if you still have issues please open a new github issue.

closed 2022-02-04 by ksheumaker