ksheumaker/homeassistant-apsystems_ecur #14: Unsupported inverter type 804

fdalto, 2021-07-10:
Hello,

I have just instaled a APSystem device. It´s running OK on their phone APP.
My microinverter model is QS1A
My ECU is  216000084416

I have installed custom_components Folder and received this LOG message:

Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:84
Integration: APSystems PV solar ECU-R (documentation)
First occurred: 12:23:26 PM (3 occurrences)
Last logged: 12:25:27 PM

Using cached data from last successful communication from ECU. Error: Unsupported inverter type 804

Could you please Help me?

thanks
[image]

HAEdwin, 2021-07-10:
Hi, try to add the first three digits to line 35 from the /config/custom_components/apsystems_ecur/APSystemsECUR.py file maybe it turns out well for this new model.

HAEdwin, 2021-07-13:
Solved by adding the QS1A id (804) to /config/custom_components/apsystems_ecur/APSystemsECUR.py line 35 to self.qs1_ids = [ “802”, “801” ,“804”] save the file and restart HA.

ksheumaker, 2021-07-13:
PR #15 merged with 804 ECU added.

closed 2021-07-13 by ksheumaker