ksheumaker/homeassistant-apsystems_ecur #21: Home Assistant 2021.9.1 energy dashboard broken

royvanmanen, 2021-09-03:
in sensor.py please change following:

line 16-18 original:
from homeassistant.components.sensor import (
    STATE_CLASS_MEASUREMENT,
 )
line 16-19 modified:
from homeassistant.components.sensor import (
    STATE_CLASS_MEASUREMENT,
    STATE_CLASS_TOTAL_INCREASING
)

after modification this is line 60:
devclass=DEVICE_CLASS_ENERGY, icon=SOLAR_ICON, stateclass=STATE_CLASS_MEASUREMENT),
change this to:
devclass=DEVICE_CLASS_ENERGY, icon=SOLAR_ICON, stateclass=STATE_CLASS_TOTAL_INCREASING),

Levdbas, 2021-09-04:
I can confirm this fixes the energy dashboard. Great work @royvanmanen

HAEdwin, 2021-12-17:
@royvanmanen This issue should be solved and can be closed, correct?

royvanmanen, 2021-12-17:
jazeker, bedankt!

closed 2021-12-17 by royvanmanen

HAEdwin, 2021-12-17:
Jij ook bedankt!