ksheumaker/homeassistant-apsystems_ecur #26: RuntimeWarning on HA 2021.12.6

dclobato, 2021-12-29:
During startup, the following message is printed on the logfile

```
/config/custom_components/apsystems_ecur/__init__.py:141: RuntimeWarning: coroutine 'DataUpdateCoordinator.async_refresh'
was never awaited coordinator.async_refresh()
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```
No further error occurs...

HAEdwin, 2021-12-29:
Does this occur consistently when restarting HA? I can't reproduce this so expect it to be a confluence of HA startup conditions. Possible performance or conflict. Are you calling the service to stop the query on HA startup?

dclobato, 2021-12-29:
Yes, I am...  There is an automation to call stop query when the HA starts and the sun is below horizon :-/

HAEdwin, 2021-12-29:
Is it an option for you to add a delay to stop the query also at startup? This way HA can startup and servicecall to stop the query will execute after that? It is known that you can let the integration query the ECU without interruption 24/7. This only results in error log lines because the ECU does some maintenance between 02:45 and 03:15 in the early morning and closes port 8899 but this does not result in further problems, the integration picks it up later on when 8899 becomes available again.

dclobato, 2021-12-29:
I was starting/stopping to suppress errors on log.   I'll keep the service querying... thanks

closed 2021-12-29 by dclobato