ksheumaker/homeassistant-apsystems_ecur #31: Weird spikes in the data

Roald28, 2022-01-17:
Hi, 

I am having weird spikes, as you can see in HA there is more than 9.5Kw of energy produced.  

[image]

But the total lifetime is not more than 8.5Kw

[image]

How can this be explained? Where do I have to look for a solution?

ksheumaker, 2022-01-17:
What sensor do you have under Solar Panels in your energy dashboard, it should be sensor.ecu_lifetime_energy?  and what does the graph look like if you view the history of that sensor inside HA?

Roald28, 2022-01-17:
This is the only sensor I see. So I think it is the right one?

[image]

ksheumaker, 2022-01-17:
Yes, it's the correct sensor.  What if you go to the history page in the sidebar, and then select sensor.ecu_lifetime_energy and pick the last day, what does that show?

Roald28, 2022-01-17:
Got a weird drop, around the same time the energy dashboard peaks looks like it. I already have set the scan_interval at 180 because the ECU crashes otherwise. Has it something to do with this?

[image]

HAEdwin, 2022-01-21:
I noticed the screendump on the EMA site. The integration does not modify the ECU data (or buffer) but only reads it. It is therefore striking that the ECU did not send correct data to the EMA site. At night, when the inverters are not working, the ECU makes sure that the internal data buffer is checked again against the EMA data that has been pushed during the day. Did you not do a new ECU or reinstall of the ECU?

Roald28, 2022-01-21:
Hi @HAEdwin,

Thanks for the info! Didn't know that. 

It is exactly the same ECU, did a powercycle because the ECU was not reporting. 

For now it is doing well the last 4 days. Did put the scan_interval on 300.

HAEdwin, 2022-01-21:
Great to hear that it is running well for four days, what kind of ECU model and firmware to you own? You mentioned spikes (more than ones), does it occur more often?

n0liver, 2022-01-21:
I've experienced the same issue today, and had it happened also on jan 11. The symptom is that the sensor value changes to 0 and then jumps back up to the total generated value (and in my case twice in quick succession apparently). See screenshots:
Energy panel:
[image]
The sensor
[image]
ECU's webpage ( [ECU-IP]/index.php/realtimedata/energy_graph )
[image]

As for my setup:
- HA 2021.12.5 running via the [docker 2.5-alpine image](https://hub.docker.com/_/haproxy) -- no useful logs
- Intergation is up to date with HACS (3b7148a) (scaninterval @ 300s, had connection issues previously)
- ECU-R with firmware ECU_R_PRO_2.0.3012 (I just noticed this -- might be reason of the issue? )
- 3x QS1 and 1x YC600

Would love to help in any way possible to fix this issue, and happily provide more info if necessary!

ksheumaker, 2022-01-21:
What do the other sensors from the integration report during the 0 reporting period?  Like sensor.ecu_current_power or sensor.ecu_today_energy?

n0liver, 2022-01-21:
sensor.ecu_current_power seems normal, ecu_today_energy appears to show the same drop. I also noticed that this _seems_ to go paired with a signal loss, but maybe that could be because of bad/failed parsing of ECU data?
[image]

ksheumaker, 2022-01-22:
Didn't mean to close this, but I created a new version which will attempt to fix this by detecting a '0' value for ecu_lifetime_energy and throw an exception, which causes the integration to pull it's last cached value.  Hopefully this will fix the issue, it also will log the raw data to the HA log so we can see what was returned.

n0liver, 2022-01-22:
Awesome, I have updated and will keep an eye on this :+1:

Roald28, 2022-01-24:
I have updated it as well! Thanks for your time!

renini, 2022-01-25:
Sorry to bump this issue, not sure if completely related, but shouldn't the other two sensor's get the right right stateclass? And not use lifetime_energy, but actually today_energy or the current_power options to calculate solar production with the current version of Home Assistant.

so in sensor.py:
```
     sensors = [
         APSystemsECUSensor(coordinator, ecu, "current_power",
             label="Current Power", unit=POWER_WATT,
-            devclass=DEVICE_CLASS_POWER, icon=SOLAR_ICON),
+            devclass=DEVICE_CLASS_POWER, icon=SOLAR_ICON, stateclass=STATE_CLASS_MEASUREMENT),
         APSystemsECUSensor(coordinator, ecu, "today_energy",
             label="Today Energy", unit=ENERGY_KILO_WATT_HOUR,
-            devclass=DEVICE_CLASS_ENERGY, icon=SOLAR_ICON),
+            devclass=DEVICE_CLASS_ENERGY, icon=SOLAR_ICON, stateclass=STATE_CLASS_TOTAL_INCREASING),
         APSystemsECUSensor(coordinator, ecu, "lifetime_energy",
             label="Lifetime Energy", unit=ENERGY_KILO_WATT_HOUR,
             devclass=DEVICE_CLASS_ENERGY, icon=SOLAR_ICON, stateclass=STATE_CLASS_TOTAL_INCREASING),
```

This way in the energy configuration you can use the sensor.ecu_today_energy and not get the error: "The following entities do not have the expected state class:" on the Energy configuration page.

ksheumaker, 2022-01-25:
These state classes are is a newish feature and were not around when the integration was first made, so yes these sensors probably do need to be updated with additional flags on them.  I'll do that in the next release, or feel free to submit a pull request for the change.

If you read the docs here for STATE_CLASS_TOTAL_INCREASING:  https://developers.home-assistant.io/blog/2021/08/16/state_class_total/ it will consider the meter reset whenever it goes down in value.  In the end it won't really matter if you pick the lifetime or the today as the energy panel looks at the difference between the last value and the current value to figure out the energy.  That's the point of the STATE_CLASS_TOTAL_INCREASING flag. 

The current power sensor on the other hand is the total power being generated at the moment, so that number will fluctuate depending on the position of the sun, clouds, etc.  So this is not the correct value to be used in the energy panel.

ksheumaker, 2022-02-04:
This should be addressed in the recently released v1.1.2.  It will not allow for a 0 value in the sensor.ecu_lifetime_energy so we shouldn't see the  graphs go crazy.  ecu_today_energy is also available as to the energy panel if you prefer to use that sensor.

PR #32 (2022-01-22): Detect 0 lifetime
Attempt to fix #16  problems with ECU_R_PRO by adding additional waits when closing sockets, and sleep for 1 second to help the ECU not get "overloaded" 

Attempt to fix #31 by detecting lifetime_energy = 0 and throwing an exception, which should cause integration to pull from it's last cached value instead of sending a 0 through to home assistant and screwing up the graph/calculations.
closed 2022-02-04 by ksheumaker