ksheumaker/homeassistant-apsystems_ecur #38: Can't add ECU-R error - APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw

fdr76, 2022-02-06:
First I removed my old entry from the configuration.yaml and the old APSystem entities and restart HA. 

Then I install via HACS 1.1.2 I can't get register the ECU. 

When I use the new config flow GUI to register the integration I get the error below in the logging. In the frontend I get the  error "can't add ECU-R". 

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsECUR.py:118 
Integration: APSystems PV solar ECU-R ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur)) 
First occurred: 09:05:22 (5 occurrences) 
Last logged: 09:45:53

APSystemsInvalidData exception: **ECU returned 0 for lifetime energy, raw** data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\x18\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\x17\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\x1b\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 40, in async_step_user
    test_query = await ap_ecu.async_query_ecu()
  File "/config/custom_components/apsystems_ecur/APSystemsECUR.py", line 118, in async_query_ecu
    raise APSystemsInvalidData(f"ECU returned 0 for lifetime energy, raw data={self.ecu_raw_data}")
custom_components.apsystems_ecur.APSystemsECUR.APSystemsInvalidData: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\x18\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'

ksheumaker, 2022-02-06:
Has your system been up and running for a while (and have you generated any power)?  We try and not allow 0 for lifetime energy since that can cause issues with the energy dashboard.  

If it's a brand new system, that could be 0 and let it run for a while before adding the integration.  If it has generated power, then I'll have to do some further investigation.

fdr76, 2022-02-07:
Thanks for your reaction! The APSystem / ECU-r solution is used about 2 years. Yesterday at the and of the day te intergration works. I have nothing changed since this report. 

Only this morning my entities are again unavailable. In the logging of HA I see that the same error is between 2:26 and 9:31. Around 9:45 I see produced power in HA. 

It looks that the ECU-r doesn't give information back when the production is minimal. After a while there is information and then the integration works. But in the UI I see that the entiteit are unavailable. 

Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:95 
Integration: APSystems PV solar ECU-R ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur)) 
First occurred: 02:26:07 (86 occurrences) 
Last logged: 09:31:07
Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00e\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\xef\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00\x84\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x008\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'
Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'APS110094000121600004446301\x00\x00\x00\x00\x00\x00\x00a\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x02\x00\x0210012ECU_R_1.2.19009Etc/GMT-8\x80\x97\x1b\x01\xa9k\x00\x00\x00\x00\x00\x00END\n'

ksheumaker, 2022-02-07:
Did the old integration give you correct data for the entity `sensor.ecu_lifetime_energy` ?

If I decode the data string it's showing it comes back as:

```
{'ecu_id': '216000044463', 'qty_of_inverters': 2, 'qty_of_online_inverters': 2, 'lifetime_energy': 0.0, 'current_power': 101, 'today_energy': 0.0, 'firmware': 'ECU_R_1.2.19', 'timezone': 'Etc/GMT-8' }
```

I've never seen that happen.  Does the EMA website show the correct energy production numbers?

ksheumaker, 2022-02-07:
On my system the lifetime energy number just continually increases.  Here you can see what it looks like over the last week.  It's never been back down to 0.

[image]
4

fdr76, 2022-02-07:
Yes, the EMA site show the right information. Only tomorrow it shows 0 Watt production till around 9:45. And then  the integration works. 

My traffic looks like the:
[image]

ksheumaker, 2022-02-07:
We've never had a report of lifetime energy being reset every night.  Daily energy yes, but not the lifetime.  Is APSystems EMA site shows something like this?

[image]

And under report menu a timeline like this?

[image]

fdr76, 2022-02-13:
Thanks for analyzing. In summary it seems that my ECU-R uses the value sensor.ecu_lifetime_energy to store the daily value. The value sensor.ecu_today_energy is not populated at all.

The integration works each day as expected after the inverter has supplied power for approximately 1/2 hour. Thus the sensor.ecu_lifetime_energy contains information. Every night this value is reset to 0, making the integration unavailable. When you want to add the integration again, it also has the value 0, which makes it appear to be a new system.

This may have been caused by a firmware update of the ECU-R. Or by a setting on the ECU-R itself. I just can't find a setting that does this. I hope someone has a idee.

For now I have added an S0 meter and am disappointed how closed APsystem is. It seems that they do not want you to read this data automatically yourself. Especially with the new catcha barrier on their webpage.

fdr76, 2022-02-20:
I can't find a setting in the ECU-r which switched total energy measured en daily energy measured. Because I'm the only person with this problem it's not a problem in integration code so I close the issue.

closed 2022-02-20 by fdr76