ksheumaker/homeassistant-apsystems_ecur #49: Can not connect to ECU. Check log

dclobato, 2022-02-22:
After upgrading HA and the integration, I'm getting an error when I try to add my ECU.

After click on "Add Integration", select "APSystems PV solar ECU" and inform the IP address of the WiFi interface, I get a "Can not connect to the ECU" error message. I tried twice, and same error on both...

Below, the log:

```bash
2022-02-21 21:14:37 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b''
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 212, in check_ecu_checksum
    checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 287, in process_inverter_data
    self.check_ecu_checksum(data, "Inverter data")
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 217, in check_ecu_checksum
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Error getting checksum int from 'Inverter data' data=b''
2022-02-21 21:18:19 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] Unhandled exception: [Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 150, in query_ecu
    self.inverter_raw_data = self.send_read_from_socket(cmd)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 105, in send_read_from_socket
    return self.read_from_socket()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 93, in read_from_socket
    data = self.sock.recv(self.recv_size)
ConnectionResetError: [Errno 104] Connection reset by peer
```

My ECU is on version 2.0.5017 (I think)

```bash
[dclobato@appserver] ~$ nc 10.0.1.19 8899
APS1100160001END
APS120117000121620000686001R▒
? "!!110015ECU_R_PRO_2.0.5017America/Sao_Paulo▒▒▒`Ũy▒00END
^C
[dclobato@appserver] ~$
```

Emile86, 2022-02-22:
Same problem here sinds today

tv3, 2022-02-22:
If you both are using an ECU_R_PRO with firmware 2.0.x, you might be running into same kind of issue as reported in https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/42

You could follow the suggestions in that issue to get integration working.

dclobato, 2022-02-23:
> You could follow the suggestions in that issue to get integration working.

Yeah!  It's working again! 🥇

closed 2022-02-23 by dclobato