ksheumaker/homeassistant-apsystems_ecur #51: ECU-R-BR MOdel: ECU-R

clifforsf, 2022-02-28:
Hello!
It's freezing, and it doesn't come back until after the power cycle.
The strange thing is that the diagnosis works!
Thank you.

          "online": true,
          "unknown": "01",
          "frequency": 60.0,
          "temperature": 39,
          "signal": 76,
          "model": "YC600",
          "channel_qty": 2,
          "power": [
            112,
            111
          ],

tv3, 2022-02-28:
What version firmware is ECU running?

HAEdwin, 2022-04-18:
@clifforsf Should be solved with the latest 1.2.12 release

HAEdwin, 2022-04-27:
@clifforsf Also look at issue #71 please let me know what the results are.

HAEdwin, 2022-05-19:
@clifforsf Can you provide us feedback? Can you close this issue? Also have a look here https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/75#issuecomment-1128781446 it might add to further stability but I need testers and feedback. Thnx!

HAEdwin, 2022-08-30:
Does this command also work on the ECU-R with pro firmware (UID starting with 2162xxxxx)
`curl "http:// < IP > /index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttpRequest" --data-raw "SSID=ECU-WIFI_local&channel=0&method=0&psk_wep=&psk_wpa="`

HAEdwin, 2022-09-13:
@tv3 Unfortunately no feedback has been given. Can you tell us if the command also applies to the ECU-R-Pro and works? Suggestion for an automation is here: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/96#issuecomment-1244235935

ViperRNMC, 2022-09-13:
I have an ECU-R and a family member also, de command works to reset the device.

HAEdwin, 2022-09-14:
@ViperRNMC Thanks for the feedback. This is great news because it also makes the integration applicable to ECU-R owners with SunSpec logo (UIDs starting with 2162xxxxxxx). 
To document things for the readme:
Have you used my suggested automation or do you have any suggestions for improvement? 
Is your ECU-R connected via ethernet cable?

ViperRNMC, 2022-09-14:
Yes, this automation is what I use now.

`
trigger:
  - platform: state
    entity_id:
      - switch.ecu_query_device
    to: "off"
condition:
  - condition: state
    entity_id: binary_sensor.ecu_using_cached_data
    state: "on"
action:
  - service: shell_command.shell_reboot_ecu
    data: {}
  - delay: "00:01:00"
  - service: switch.turn_on
    entity_id: switch.ecu_query_device
`

For me it is not clear, is it wise to disable the query switch in the night (in automation)? what does the switch do?

HAEdwin, 2022-09-14:
The switch simply skips querying the ECU, no further action is taken but does not stop the 5 minute job interval to scan for tasks. The switch was initially build to stop queries during the time the ECU is checking with the EMA site if data is correct or needs to be corrected and firmware updates are there. Querying during this maintenance resulted for previous versions of the integration in time-outs and error entries in the log. In the new versions time-outs are surpressed in the log because we know these can occur and do not add to a [silver level of integration quality Scale](https://developers.home-assistant.io/docs/integration_quality_scale_index/).

In fact the switch could be replaced by a sensor which indicates that an ECU reboot is needed (after five failed attempts to get a response from the ECU). Though for testing purposes it might be handy to trigger the cache. 

Your automation uses the shell command but the RESTful command is more native to HA's software stack and is better in handling results. Shell will use bash of underlying operating system, that's an external dependency. I want to be as close to compatibility with all installations as possible. Would you like to try my automation from this thread (https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/96#issuecomment-1244235935) and provide me some feedback?

ViperRNMC, 2022-09-14:
Changed to rest, disable query and reset won't work. 

'''
rest_command:
  reset_ecu_c:
    url: "http://<ip-address>/index.php/management/set_wlan_ap"
    payload: "-H 'X-Requested-With: XMLHttpRequest' --data-raw 'SSID=ECU-WIFI_local&channel=0&method=0&psk_wep=&psk_wpa='"
    method: post
'''

HAEdwin, 2022-09-14:
Have to do some research on REST I guess, thanks for testing!

HAEdwin, 2022-09-18:
Though REST may be preferred, I'll stick to Shell first and close this issue since it is solved with this automation for ECU-R pro firmware devices also.

closed 2022-09-18 by HAEdwin