ksheumaker/homeassistant-apsystems_ecur #58: Can't connect to ap

pbabilas, 2022-03-09:
As in topic can't connect to my ap ecu-r:

https://share.merce.com/files/XgQpEtm1oo.png

Sunno what means the err msg.

HAEdwin, 2022-03-09:
have you tried this? https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/55

pbabilas, 2022-03-09:
I understand that. I've got configured all. My ip is set https://share.merce.com/files/YcCxRhXzzv.png and online. but ports are closed :(

`(☸ |prod:kids51015) b-code@bcode-XPS  ~/workspace/amba/ratelimit   main  nmap 192.168.1.100
Starting Nmap 7.80 ( https://nmap.org ) at 2022-03-09 17:13 CET
Nmap scan report for 192.168.1.100
Host is up (0.0022s latency).
All 1000 scanned ports on 192.168.1.100 are closed
`

Maybe it's becouse I've got model with zigbee?

https://thepowerstore.com/apsystem-ecu-residential-monitoring-gateway-zigbee-comm-wifi-no-cts-no-rgm-yc600-sku-part-number-ecu-r-gateway.html

HAEdwin, 2022-03-09:
No, Zigbee is not the issue - the ECU communication with inverters is zigbee (in mesh network if you've got multible inverters). You only scanned the first 1000 ports? The apps use port 8899 to communicate with the ECU-R. Try: nmap 192.168.1.100 -p 8899

pbabilas, 2022-03-09:
tried also port:

(☸ |prod:partnertele) b-code@bcode-XPS  ~/workspace/amba/ratelimit   main  nc -zv 192.168.1.100 8899
nc: connect to 192.168.1.100 port 8899 (tcp) failed: Connection refused

:(

HAEdwin, 2022-03-09:
Does Packet Sender work for you or are you on MAC?

pbabilas, 2022-03-09:
Tried it already, no luck :(

https://share.merce.com/files/fIwDgz6avZ.png

PS. Restarted router so there is now ip from dhcp ;]

pbabilas, 2022-03-09:
And using ubuntu 21.10.

HAEdwin, 2022-03-09:
So your ECU now got 192.168.1.150 due to DHCP instead of 192.168.1.100 - it's weird you're not able to connect. Not even with Packet Sender. I wonder if there's something in your network preventing a connection, are you able to ping the ECU? Did you allready powercycle the ECU? What is your firmware version (visible when using the ECUapp).
I'm able to connect:
[image]
[image]

pbabilas, 2022-03-09:
ping is fine:

`(☸ |prod:partnertele) b-code@bcode-XPS  ~/workspace/amba/ratelimit   main  ping 192.168.1.150
PING 192.168.1.150 (192.168.1.150) 56(84) bytes of data.
64 bytes from 192.168.1.150: icmp_seq=1 ttl=255 time=3.87 ms
64 bytes from 192.168.1.150: icmp_seq=2 ttl=255 time=1.60 ms
64 bytes from 192.168.1.150: icmp_seq=3 ttl=255 time=3.70 ms
64 bytes from 192.168.1.150: icmp_seq=4 ttl=255 time=1.38 ms
64 bytes from 192.168.1.150: icmp_seq=5 ttl=255 time=0.996 ms
^C
--- 192.168.1.150 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4007ms
rtt min/avg/max/mdev = 0.996/2.306/3.865/1.219 ms
(☸ |prod:partnertele) b-code@bcode-XPS  ~/workspace/amba/ratelimit   main  nc -zv 192.168.1.150 8899
nc: connect to 192.168.1.150 port 8899 (tcp) failed: Connection refused
`

What do u mean "powercycle the ECI?"

Hmm I cant find that workspace on my app.
All i've got is:

[image]
[image]

HAEdwin, 2022-03-09:
You have firmware 1.2.20 also (like I have). The IP 192.168.1.150 is bound to you LAN (ethernet/cabled) connection. Using this connection is not possible, port 8899 is not open on the ethernet NIC. What is the WiFi IP-address on the ECU? Configure and use that.

pbabilas, 2022-03-10:
Gr8 its working now!, thans a lot. Wlan has other mc address so other ip was given. Need to scan all ips and find one with 8899 port open. Now it working. Thanks again! ;]

HAEdwin, 2022-03-10:
No problem, glad I could help. Please mark the post as answered so this issue can be closed.

closed 2022-03-10 by pbabilas

HAEdwin, 2023-06-13:
Yes correct, wireless only will do as well.

Smiggel, 2023-06-13:
> Yes correct, wireless only will do as well. But keep in mind that both interfaces each use a different IP-address. So make sure you enter the correct one in the integration's settings.

Ok, thank you. I have to wait for some info from my supplier, before I can login with the app, etc. I will wait for that info and connect the AP to my WiFi network.