ksheumaker/homeassistant-apsystems_ecur #59: Updating from 2022.1, now Inverter data is failing checksum

nectur, 2022-03-15:
Updated HA from 2021.12 to 2022.2 and this plugin from the versionless 2022.1 to the current 1.2.11 via HACS.  Now am unable to configure the plugin (was working fine via YAML)  When I add in the wifi IP address I get the following exceptions: 

```
2022-03-15 16:49:19 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b''

Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 212, in check_ecu_checksum
checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)

File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)

File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
data = self.process_inverter_data()

File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 287, in process_inverter_data
self.check_ecu_checksum(data, "Inverter data")

File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 217, in check_ecu_checksum
raise APSystemsInvalidData(error)

custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Error getting checksum int from 'Inverter data' data=b''
```

ECU has been rebooted multiple times to attempt diagnosis.  Responds properly to requests via Packet Sender.

JelleMarc, 2022-04-07:
Hi!

I am getting the same error. Did you ever get this resolved?

nectur, 2022-04-07:
Short of rolling back manually since it was before version numbers in HACS, I have not.  

I did just switch over to the fork from @tv3 where he is pulling data directly from the local webserver on the ECU.  That loaded up right away.

HAEdwin, 2022-04-15:
@nectur  The tv3 solution works great for ECU-R pro (with SunSpec logo on the back) owners. It's hard to solve the issue without owning one myself. Please close this issue if you're happy with it.

nectur, 2022-04-15:
@HAEdwin I'll give your debug mentioned in #67 a try and provide feedback on that issue.  Thanks.

closed 2022-04-15 by nectur

HAEdwin, 2022-04-17:
@JelleMarc You might want to try this https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/67#issuecomment-1100749752
It closes and reopens the socket inbetween the ecu query. Add lines 156 & 157, 164 & 165