ksheumaker/homeassistant-apsystems_ecur #60: Problem with setting ip adress ECU-R After install integration on new HA

RobertOz20, 2022-03-17:
Hello,

i install integration 1.2.11 by HACS, but i haven't any page, prompt to enter ip adress my ECU-R.  MY Home Assistant 2022.3.5. I don't know how configure it ...

HAEdwin, 2022-03-17:
You have to install the integration also, that's where the IP-Address is being entered.

RobertOz20, 2022-03-18:
I'v install ... but i havent any page to enter parameters ..

[image]

HAEdwin, 2022-03-18:
In HA go to [Settings] > [Devices and Services](Integrations, devices,entities and Areas) > on the right bottom select [Add Integration] and look for APSystems PV solar ECU, select it and you will be able to add an IP-Address
[image]

RobertOz20, 2022-03-18:
Thanks, it works !!! I'm new in HA. I have ECU-R, and 6 inverters.

gregyyc, 2022-04-04:
@HAEdwin you mention above if you have the IEEE 2030.5 CSIP logo you have ECU-R-Pro firmware and may have compatibility problems. That might be where I am at. I do have that logo on the on the back and when adding the IP address I get "Can't connect to ECU, check log" and the following in the logs ....

APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b''

Is there a work around for that firmware?

Thanks!

HAEdwin, 2022-04-05:
@gregyyc The challenge with this firmware is that the response to commands appears to be slightly different or unexpected. It is difficult to anticipate this if you as a developer do not have an ECU-R with ECU-pro firmware. 

The initial starting point for this integration was the (apparently) obsolete standard ECU-R without Sunspec logo on the back which also happens to have compatibility with the ECU-B.

Solving the problem requires a lot of patience and troubleshooting. Also, the owner of the ECU-R-Pro firmware must have some Python knowledge to indicate what to change in the code or configuration. Apart from this integration for HA, there are alternatives in the form of scraping or direct Zigbee reception from the inverters without ECU. I myself have thought about an ECU proxy where the data to EMA is first intercepted for use in HA.

HAEdwin, 2022-04-06:
@gregyyc The integration works great for me and other owners of an ECU-R (without Sunspec logo) or ECU-B so for me personally there is currently no need to change the approach.

We could give it a try. If you would like to raise a new issue with the subject something like: ECU-R with ECU-R-PRO firmware compatibility. Then start the discussion with the version of the integration you would like to work on and if possible your current ECU-R firmware. The first step is to solve "when adding the IP address I get Can't connect to ECU, check log and the following in the logs APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b'"

HAEdwin, 2022-04-10:
@RobertOz20  Since your issue is solved, can you close this issue please?

HAEdwin, 2022-04-15:
@gregyyc  Look at this issue if you'd like to try to solve it: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/67

HAEdwin, 2022-07-09:
@gregyyc Was the issue ever solved? If not you might want to try my fork, is has a lot of improvements: https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip

gregyyc, 2022-07-11:
> @gregyyc Was the issue ever solved? If not you might want to try my fork, is has a lot of improvements: https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip

Hey - last time I tried it worked out well. I've since moved to a different solution interacting with my Sense monitor but will probably dig back into this again soon. Thanks for the follow up!

closed 2022-07-14 by RobertOz20