ksheumaker/homeassistant-apsystems_ecur #61: [%key:common::config_flow::error::unknown%] upon setup

dewi-ny-je, 2022-03-31:
I have an ECU-R-PRO which I can access using the hotspot button.

I downloaded the master branch as 3166ad4 and installed it on HA 2022.1.

When I try to set it up I get in the interface:

`[%key:common::config_flow::error::unknown%]
`
and in the logs:

```
Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:93
Integration: apsystems_ecur ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur))
First occurred: 11:25:27 PM (5 occurrences)
Last logged: 11:30:17 PM

Unhandled exception: [Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 150, in query_ecu
    self.inverter_raw_data = self.send_read_from_socket(cmd)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 105, in send_read_from_socket
    return self.read_from_socket()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 93, in read_from_socket
    data = self.sock.recv(self.recv_size)
ConnectionResetError: [Errno 104] Connection reset by peer
```

I'm not sure how to proceed, it seems like a bug to me, since exceptions should be in principle caught and handled.

HAEdwin, 2022-03-31:
When the connection is reset by the peer, the connection is not accepted. Make sure you've got the IP-address right and it's the WiFi connected IP-Address. I'm not sure if this goed for all models but the standard ECU-R allows port 8899 only when it's WiFi connected not by ethernet.

dewi-ny-je, 2022-03-31:
I had both eternal and wifi set up. Now I disconnected the lan, rebooted, tried again. The IP is the same I used earlier and it's correct,  I manually assigned it in the router.
I get

```
Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:217
Integration: APSystems PV solar ECU (documentation)
First occurred: 00:10:29 (1 occurrences)
Last logged: 00:10:29

APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b''
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 212, in check_ecu_checksum
    checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 287, in process_inverter_data
    self.check_ecu_checksum(data, "Inverter data")
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 217, in check_ecu_checksum
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Error getting checksum int from 'Inverter data' data=b''
```

HAEdwin, 2022-03-31:
This is a known error with the latest ECU-R models (with SunSpec logo on the back). These indeed have the ECU-R-Pro firmware and should be modbus compatible.

Thetmar, 2022-04-03:
Hello Edwin and Dewi, 

As i read in your last post Edwin. Is it possible to read the ECU-R with Modbus? Do you know how? 
I have an ECU-R with an RJ45 output connector with discription RS485 on it.
Kind regards,
Thetmar

dewi-ny-je, 2022-04-03:
> Hello Edwin and Dewi,
> 
> As i read in your last post Edwin. Is it possible to read the ECU-R with Modbus? Do you know how? I have an ECU-R with an RJ45 output connector with discription RS485 on it. Kind regards, Thetmar

Look for the SUNSPEC module for home assistant

closed 2022-04-03 by dewi-ny-je