ksheumaker/homeassistant-apsystems_ecur #62: HA Release 2022.4 causes sensors to fail

HAEdwin, 2022-04-07:
sensor.py needs to be modified (no time to create pull request at this moment, this is my reminder here)
Add the following line: 
`from homeassistant.helpers.entity import EntityCategory`
Replace: 
`entity_category="diagnostic"` 
for 
`entity_category=EntityCategory.DIAGNOSTIC`

Need to work on the switch also, any additions welcome

otello86, 2022-04-10:
Hello
i do the update on sensor.py, it solved some problems but ECU_querying is still unavailable.
I already have this error in logs : 

Logger: homeassistant.components.switch
Source: helpers/entity_registry.py:529
Integration: Commutateur (documentation, issues)
First occurred: 08:43:44 (2 occurrences)
Last logged: 08:43:45

Error adding entities for domain switch with platform apsystems_ecur
Error while setting up apsystems_ecur platform for switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 382, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 510, in _async_add_entity
    entry = entity_registry.async_get_or_create(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 345, in async_get_or_create
    return self.async_update_entity(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_registry.py", line 529, in async_update_entity
    raise ValueError("entity_category must be a valid EntityCategory instance")
ValueError: entity_category must be a valid EntityCategory instance

Scholtz-i, 2022-04-10:
I have made all the changes from HAEdwin (Pull requests => File changed) and it works perfectly again.
Big thanks to HAEdwin.

otello86, 2022-04-10:
Yes i saw, thanks

HAEdwin, 2022-04-10:
For a quick reference I'll post the changes:
sensor.py
[image]

binary_sensor.py
[image]

switch.py
[image]

ksheumaker, 2022-04-18:
Compatibility with 2022.4 fixed in release v1.2.12

closed 2022-04-18 by ksheumaker