ksheumaker/homeassistant-apsystems_ecur #64: Cannot "Add Integration" after adding repository

clandmeter, 2022-04-08:
Not sure im doing something wrong here, but when I add the repository in HACS as described in the readme via custom repositories, it does not show when I visit Configuration -> Devices/Services -> Integrations -> Add Integrations. Typing apsystems does not show anything.

I am running: Home Assistant 2022.4.1

HAEdwin, 2022-04-08:
After adding via HACS you have to restart HA.

clandmeter, 2022-04-08:
Yup did that, still doesnt list it, it was red color before the restart

[image]

When searching for apsystems it does not show anything:

[image]

HAEdwin, 2022-04-08:
It might have to do with incompatibility (issue #09) Try to correct the 3 files using these directives https://gathering.tweakers.net/forum/insert_bookmark/2032302/71191574 then restart HA and try to add/find it again.

clandmeter, 2022-04-09:
The #9 PR is a year old, are you sure that's the related PR?
I did try your branch at https://github.com/HAEdwin/homeassistant-apsystems_ecur
But that didn't make a difference.

I was expecting to find some error in the log, but I don't see anything related.

clandmeter, 2022-04-09:
Ok I found the issue, I needed to clear my browsers cache and the integration popped up.

closed 2022-04-09 by clandmeter

HAEdwin, 2022-04-09:
I was referring to the wrong issue #9 but I meant to refer to issue #63. Great that you've found the solution.