ksheumaker/homeassistant-apsystems_ecur #66: Error when adding integration ECU-C

GibbseyNZ, 2022-04-15:
I tried adding my ECU-C to this. The ECU-C is connected to wifi when I go to add the device I get "Can't connect to ECU, check log"

The log is attached below.

This error originated from a custom integration.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:217
Integration: APSystems PV solar ECU (documentation)
First occurred: 16:03:02 (3 occurrences)
Last logged: 16:08:31

APSystemsInvalidData exception: Error getting checksum int from 'Inverter data' data=b''
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 212, in check_ecu_checksum
    checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 287, in process_inverter_data
    self.check_ecu_checksum(data, "Inverter data")
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 217, in check_ecu_checksum
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Error getting checksum int from 'Inverter data' data=b''

HAEdwin, 2022-04-15:
This integration was originally written for the ECU-R (without SunSpec logo on the back) and happened to be compatible with the ECU-B also.  Kyle and I do not own an ECU-C so it's hard to debug. You could try to add` self.sock.settimeout(self.timeout)` to APSystemsSocket.py on the lines where I discussed possible compatibility issues here: https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/50

HAEdwin, 2022-04-15:
@GibbseyNZ look at this post for directions if you whould like to try: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/67

HAEdwin, 2022-04-18:
@GibbseyNZ 1.2.12 was released, this should solve your issue. I received mail from another ECU-C owner that things are solved, can you confirm?

GibbseyNZ, 2022-04-19:
I haven't had a chance to test the solution you sent me yet, I will give it
a go tonight and update and see how it goes. If it does it will be great as
I can then program my EV slow charger to be turned on by HA when my solar
output gets above what the charger power usage would be.
Thanks,

GibbseyNZ, 2022-04-19:
Awesome. The update works. Thanks for that.

closed 2022-04-19 by GibbseyNZ