ksheumaker/homeassistant-apsystems_ecur #70: Basic Connection Help - Error [%key:common::config_flow::error::unknown%]

bugsbunny88, 2022-04-25:
Hello,
I have my ECU-R connected to my home wifi network.
It has been given a static dhcp lease 192.168.40.100.
I installed this integration via HACS.
I am able to login to ECU-R when I click on the AP button and access through its internal wifi network. However when it is connected to my home wifi I am unable to.

My Question is, are you guys essentially sending a signal to the ECU-R through the local AP command somehow and getting this from the local wifi . So Home Assistant is connected to the ECU-R ? 
ECU-R -> Home Assistant -> Router

OR are you guys able to do this like my setup ?
Home Assistant -> Router
ECU-R -> Router

I can't evn get past the first step and end up with [%key:common::config_flow::error::unknown%]

I have tried v1.12 and v1.11 from github
[image]

HAEdwin, 2022-04-25:
No, we do not connect via the local AP from the ECU itself but the WiFi connected ECU. You have to use the 1.2.12 release which is the latest. We found out that a WiFi connected ECU-R and ECU-B can use the same commands the ECUapp uses when connected to the ECU's AP but over WiFi. This goes for the "older/standard" ECU-R without  SunSpec logo and for the ECU-B. It is known that  ECU-R owners with SunSpec logo on the back might have compatibility issues, but should be able to connect over Modbus using the SunSpec specifications. There's also an initiative to read out this model using scraping from the local web UI.

Roald28, 2022-04-25:
Hello,

I'm hooking up on this issue.

Did a reinstall of HA  and the latest version from Github and ran into the same problem:

[image]

HAEdwin, 2022-04-25:
If you use Packet Sender and apply the data from the image below (but your WiFi connected IP-adress) does the ECU respond?
[image]

Roald28, 2022-04-25:
@HAEdwin 

1. If I am home I can do the Packet sender test
2. yes
3. Log File:

`2022-04-25 12:59:25 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] Unhandled exception: name 'APSsystemsInvalidData' is not defined
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 214, in check_ecu_checksum
    checksum = int(data[5:9])
ValueError: invalid literal for int() with base 10: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 135, in query_ecu
    self.process_ecu_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 248, in process_ecu_data
    self.check_ecu_checksum(data, "ECU Query")
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 219, in check_ecu_checksum
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Error getting checksum int from 'ECU Query' data=b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 136, in query_ecu
    except APSsystemsInvalidData as err:
NameError: name 'APSsystemsInvalidData' is not defined`

bugsbunny88, 2022-04-25:
My ECU-R exhibits some wierd behavior where I can't even ping it.
It connects to my wifi but doesn't show up on ARP table and sends signal and light shows up saying its connected. I created a static DHCP entry to capture it at 192.168.40.100 ; and i can't even ping it.

The only thing I can do is set the local AP and connect via the manager app locally OR view it from the user-end app.

In terms of cloud polling : are you referring to :
https://github.com/skelgaard/homeassistant-apsystems

bugsbunny88, 2022-04-25:
I just want to provide an update, I don't know what magic you guys did ... but I have an ECU-R-PRO with the sun label... 
this is what worked for me using the HACS integration... 
You connect the system to wifi
you also activate LAN ; and hardwire it in...
on the prompt you enter the LAN IP ADDRESS of the unit and voila ... it loads in all the inverters....

zylxpl, 2022-04-26:
Also have ecu-r-pro with sunspec logo, firmware 2.6. It was working flawlessly  for couple of days on both lan and wlan, however, since yesterday it stooped. I'm also getting [%key:common::config_flow::error::unknown%] when i trying to connect.

HAEdwin, 2022-04-26:
You might try to turn your ECU-R off and on again.

zylxpl, 2022-04-26:
@HAEdwin Yep, that was also my idea, however my ecu is 100km away. Is there a way to restart it remotely?

HAEdwin, 2022-04-26:
You can try the ethernet connected IP-address and send the string "reboot" to port 4540. This works for the 'older' ECU-R and does not flush settings, it's just a reboot.

zylxpl, 2022-04-26:
@HAEdwin something like "echo -n 'reboot' | nc ip 4540"?

HAEdwin, 2022-04-26:
echo 'reboot' | ncat [-u] [ip address] [port]

zylxpl, 2022-04-26:
@HAEdwin Ok, did not work, however changing settings of internal AP forced a restart and it is working again. Thx! Let see for how long :)

bugsbunny88, 2022-04-26:
someone needs to change the home-assitant add-on prompt that says it needs to be only on wifi for ECU-R to work i think, 
you need to have it both connected via lan and wifi ; and enter lan ip addy for whatever reason .

Roald28, 2022-04-26:
> someone needs to change the home-assitant add-on prompt that says it needs to be only on wifi for ECU-R to work i think, you need to have it both connected via lan and wifi ; and enter lan ip addy for whatever reason .

@bugsbunny88 How did you do this? Just had both Wifi en LAN connected and it wouldn't work. Did you restart something to get this to work?

Hasharderwyk, 2022-04-27:
Hello to you all.
Same issue here with a ecu-r with a sunspec logo verzie 2.06
At the moment it stops after 1,5 - 2 days.
I am figuring out whats going wrong, but is didn't try everything yet.
What i do know for now, if i unplug the power of the ecu for 10sec. And plug it in again it will work again for 1 or 2 days.
I

HAEdwin, 2022-04-27:
@Hasharderwyk 
I opened #71 for this issue

HAEdwin, 2022-04-27:
> someone needs to change the home-assitant add-on prompt that says it needs to be only on wifi for ECU-R to work i think,
> you need to have it both connected via lan and wifi ; and enter lan ip addy for whatever reason .

I can add it but how sure are you about this finding, does v1.2.12 and #71 make any difference for the situation and is it a finding based on longer term availability of the ECU and/or first installation when owning an ECU-R-pro?

Roald28, 2022-04-27:
@HAEdwin 

1. As you asked, I think I get a response. 
2. I also changed 106 as told in #71. 
3. Tried the Wifi and Lan option but doesn't seem to work. (Don;t think that should be te solution)

[image]

Any other thoughts on what can be the issue?

HAEdwin, 2022-04-27:
Roald28, The response you get from 192.168.1.16 is empty so there are two possibilities 1. You have to powercycle the ECU 2. This IP-address is not the correct WiFi connected IP-address. A correct response looks like the sample below. You'll receive an APSxxxx message back if done right. Try to solve this first using Packet sender and when this is correct add the same IP-address to the integration's settings.
[image]

Roald28, 2022-04-27:
> Roald28, The response you get from 192.168.1.16 is empty so there are two possibilities 1. You have to powercycle the ECU 2. This IP-address is not the correct WiFi connected IP-address. A correct [...]

A restart did it. Thanks!

HAEdwin, 2022-06-19:
Issue can be closed, thnx!

HAEdwin, 2022-08-06:
@bugsbunny88 If your issue is solved (also take a look at https://github.com/HAEdwin/homeassistant-apsystems_ecur/blob/main/custom_components/apsystems_ecur/APSystemsSocket.py) can you please close this issue for I'm not able to moderate this. Thank you.

closed 2022-08-30 by HAEdwin