ksheumaker/homeassistant-apsystems_ecur #74: Can't connect to ECU

aleg-nikue, 2022-04-28:
I found issue on log:
2022-04-28 23:45:19 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Unsupported inverter type 970 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 970 please create GitHub issue.

ksheumaker, 2022-04-29:
What model of inverters do you have, and where are you located? Once we know the inverter model we should be able to add support in the next release.

aleg-nikue, 2022-04-29:
I have 5 inverted QS1 boosted (1200/1400W) located in Poland near
Koszalin. Installation working with ECU-C.
Previous version of hassio integration was working perfectly - issue appear
after last upgrade... but I changed last wednesday all inverters from
version QS1 1200W to QS1 1200/1400W boosted
On photo my QS1 1200W/1400W boosted
[image: obraz.png]

HAEdwin, 2022-04-29:
In this case you can try adding it to APSystemsSocket.py Please let us know if this helps so Kyle can add it to the next release.

Modify line 45 in APSystemsSocket.py by adding 970 (after modification restart HA).
45         self.qs1_ids = [ "802", "801", "804", "805", "806", "970" ]

aleg-nikue, 2022-04-29:
i came up with the same and try it yesterday, but no success :)

HAEdwin, 2022-04-29:
What are the results in the log? We might have to change more code.

aleg-nikue, 2022-04-29:
Log I send is after modify in APSystemsSocket.py and few restarts HA and
off/on ECU-C power supply
I tried access by Ethernet cable IP and after that Wifi IP - no success :(

My experience with this integration in the previous version is:
1. after 2-3 days integration stopped working (all entities stay
unavailable)
2. when I tried restart and more deeply delete integration and then
installed new instance - integration cant connection with ECU-C, even
restart HA
3. only way to solved it is: turn off/20sec/on supply ECU-C, delete
integration in HA, restart HA, install new integration and perform connect
to ECU-C
4. few week ago I found easier solution - I installed zigbee relay near
ECU_C with automation reset supply ECU every 2 days at 1:00 - after that no
problem with stable work integration :)
5. but currently after new update integration is problem to connect I wrote
before....

HAEdwin, 2022-04-29:
Your log shows that the integration is able to connect to the ECU. You will also have added the new inverters to the ECU and updated EMA I guess. Stability of the ECU-C with this integration is still an issue but we try to solve this taking small steps.
If you can, install Packet Sender and post the result of the returned data here. I think we might have to take a closer look at the returned data. Come back to you later.

aleg-nikue, 2022-04-30:
After reinstall integration and add inwerter ID 970 I've got log:
2022-04-30 09:19:21 ERROR (MainThread)
[custom_components.apsystems_ecur.config_flow] APSystemsInvalidData
exception: Unable to convert binary to int location=181
data=b'415053313130313830303030323030303100092022043009133980200014221601303301f4007c008e00f10065011c011f80200014253401303301f4007e012c00f1012d012f012e802000142349013033028a00730000000500000004000480200014264201303301f4007500a000f200a200a6008d80200014258501303301f40076009f00f200a200a50095806000008104003030806000008668003030806000008897003030806000009733003030454e440a'
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
171, in aps_int
return int(binascii.b2a_hex(codec[(start):(start+2)]), 16)
ValueError: invalid literal for int() with base 16: b''
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in
async_step_user
test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in
run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
158, in query_ecu
data = self.process_inverter_data()
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
334, in process_inverter_data
(channel_data, location) = self.process_qs1(data, location)
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
404, in process_qs1
power.append(self.aps_int(data, location))
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
176, in aps_int
raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData:
Unable to convert binary to int location=181
data=b'415053313130313830303030323030303100092022043009133980200014221601303301f4007c008e00f10065011c011f80200014253401303301f4007e012c00f1012d012f012e802000142349013033028a00730000000500000004000480200014264201303301f4007500a000f200a200a6008d80200014258501303301f40076009f00f200a200a50095806000008104003030806000008668003030806000008897003030806000009733003030454e440a'

Foto005, 2022-05-13:
I have the same problem except the error states unsupported inverter type 000.  I'm in Mexico.

This error originated from a custom integration.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: APSystems PV solar ECU (documentation)
First occurred: 8:52:34 PM (3 occurrences)
Last logged: 9:09:13 PM

APSystemsInvalidData exception: Unsupported inverter type 000 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 000 please create GitHub issue.

I did a CURL test to the ECU-R and it returned data:

curl -X POST http://192.168.1.191/index.php/realtimedata/old_energy_graph -d "date=2022-05-10&period=monthly"
{"energy":[{"date":"04\/11","energy":0.12},{"date":"04\/12","energy":0.1},{"date":"04\/13","energy":0.03},{"date":"04\/14","energy":0},{"date":"04\/15","energy":0},{"date":"04\/16","energy":0},{"date":"04\/17","energy":0},{"date":"04\/18","energy":0},{"date":"04\/19","energy":0},{"date":"04\/20","energy":0},{"date":"04\/21","energy":16.72},{"date":"04\/22","energy":56},{"date":"04\/23","energy":55.87},{"date":"04\/24","energy":55.12},{"date":"04\/25","energy":56.84},{"date":"04\/26","energy":55.63},{"date":"04\/27","energy":56.4},{"date":"04\/28","energy":56.68},{"date":"04\/29","energy":56.75},{"date":"04\/30","energy":56.23},{"date":"05\/01","energy":56.23},{"date":"05\/02","energy":56.89},{"date":"05\/03","energy":56.64},{"date":"05\/04","energy":53.54},{"date":"05\/05","energy":51.9},{"date":"05\/06","energy":54.53},{"date":"05\/07","energy":43.69},{"date":"05\/08","energy":53.71},{"date":"05\/09","energy":0},{"date":"05\/10","energy":49.22}],"total_energy":998.84,"subtitle":""}

Any recommendations? Thanks!

aleg-nikue, 2022-05-17:
> In the past period I spent some days to see how stability can be
> increased. Of course I was only able to test with my own ECU (an older
> ECU-R without SunSpec logo on the back). In my case the number of timeouts
> has decreased enormously so it benefits me but I have absolutely no idea
> how the other ECU's react to this. In any case, you can download my code
> here.
>
> https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip
>
> Extract from the zip only the file: APSystemsSocket.py and overwrite the
> existing file at: config/custom_components/apsystems_ecur *make sure you
> restart HA afterwards*.
> (You can also rename the existing file first and then copy the new file
> into the folder so you can easily restore the old situation)
>
> Not sure if it will help you. What I did was trying to use another method
> of receiving complete dataframes, added some extra visual log informatie
> when timeouts or socket errors occur, added a delay between question and
> answer, added buffer flushing

Thank you for new version of APSystemsSocket.py. I've overwrite existing
file and restarted HA.
I have ECU-C and inverters QS1 with ID=970, then I added this ID to file
APSystemsSocket.py: line 45: self.qs1_ids = [ "802", "801", "804", "805",
"806" , "970" ] and restart HA.
Not connected to ECU-C - no success.

I found in log viewer:
2022-05-17 22:45:48 ERROR (MainThread)
[custom_components.apsystems_ecur.config_flow] APSystemsInvalidData
exception: Unable to convert binary to int location=181
data=b'415053313130313830303030323030303100092022051722412280200014221600303300000000000000000000000000008020001425340030330000000000000000000000000000802000142349003033000000000000000000000000000080200014264200303300000000000000000000000000008020001425850030330000000000000000000000000000806000008104003030806000008668003030806000008897003030806000009733003030454e440a'
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
168, in aps_int
return int(binascii.b2a_hex(codec[(start):(start+2)]), 16)
ValueError: invalid literal for int() with base 16: b''
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in
async_step_user
test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in
run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
157, in query_ecu
data = self.process_inverter_data()
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
309, in process_inverter_data
(channel_data, location) = self.process_qs1(data, location)
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
366, in process_qs1
power.append(self.aps_int(data, location))
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line
173, in aps_int
raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData:
Unable to convert binary to int location=181
data=b'415053313130313830303030323030303100092022051722412280200014221600303300000000000000000000000000008020001425340030330000000000000000000000000000802000142349003033000000000000000000000000000080200014264200303300000000000000000000000000008020001425850030330000000000000000000000000000806000008104003030806000008668003030806000008897003030806000009733003030454e440a'

Can you help me?

HAEdwin, 2022-05-17:
It's an issue in the dataframe, I don't think 970 is a valid inverter ID. I will have to decode the data first to discover how the info is compiled. Will take some time - sorry about that, I'll have to find time somewhere.

HAEdwin, 2022-06-19:
Is it correct that you have 6 QS1 inverters? You might try my latest APSystemsSocket.py
https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip

Extract from the zip only the file: APSystemsSocket.py and overwrite the existing file at: config/custom_components/apsystems_ecur make sure you restart HA afterwards.

Let me know if this solves your issue and plz let me know if I'm correct with 6 QS1 inverters.

HAEdwin, 2022-06-19:
Are you sure the ECU config is allright? I noticed something weird. All your inverter are QS1's starting with 802 but the last one starts with 806, and contains weird data, is offline also...
[image]

aleg-nikue, 2022-06-19:
As I wrote before on may,26 to ksheumaker/homeassistant-apsystems_ecur
after reinstalation firmware
on all 5 microinvertes in my instalation (performed by Service APsystems -
cloud instance) integration APSystems ECU-R started working correctly.
Thank you for helping and working on reported problem.

HAEdwin, 2022-08-06:
@aleg-nikue Also look at https://github.com/HAEdwin/homeassistant-apsystems_ecur/blob/main/custom_components/apsystems_ecur/APSystemsSocket.py 

If your problem is solved, please close this issue for I am not able to moderate this. Thank you.

closed 2022-08-06 by aleg-nikue

aleg-nikue, 2022-10-11:
After today reinstalation firmware on all microinvertes in my instalation
(performed by Service APsystems - cloud instance) integration APSystems
ECU-R started working correctly. Thank you for helping and working on
reported problem.

HAEdwin, 2022-10-11:
No problem, glad it helped. After all the data you posted was indeed not completely interpretable so there really was something wrong with the ECU/Inverter configuration. Because you own an ECU-C you might want to use the latest Beta version 1.2.20 because it has on-board restart capabilities in case the ECU fails to send data after a while. It is still in beta and I'm awaiting results. If they're positive I will release a new official version witch additional binairy-sensor so that users are able to see when the ECU did restart because of failure.