ksheumaker/homeassistant-apsystems_ecur #75: Another Can't Connect To ECU

Foto005, 2022-05-16:
This appears similar to a previous post where getting the error states unsupported inverter type, except the inverter type not working for me is "000". I'm in Mexico and have QS1A 1500W inverters. Here is the HA log output:

This error originated from a custom integration.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: APSystems PV solar ECU (documentation)
First occurred: 8:52:34 PM (3 occurrences)
Last logged: 9:09:13 PM

APSystemsInvalidData exception: Unsupported inverter type 000 please create GitHub issue.
Traceback (most recent call last):
File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
data = self.process_inverter_data()
File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 000 please create GitHub issue.

I did a CURL test to the ECU-R and it returned data:

curl -X POST http://192.168.1.191/index.php/realtimedata/old_energy_graph -d "date=2022-05-10&period=monthly"
{"energy":[{"date":"04/11","energy":0.12},{"date":"04/12","energy":0.1},{"date":"04/13","energy":0.03},{"date":"04/14","energy":0},{"date":"04/15","energy":0},{"date":"04/16","energy":0},{"date":"04/17","energy":0},{"date":"04/18","energy":0},{"date":"04/19","energy":0},{"date":"04/20","energy":0},{"date":"04/21","energy":16.72},{"date":"04/22","energy":56},{"date":"04/23","energy":55.87},{"date":"04/24","energy":55.12},{"date":"04/25","energy":56.84},{"date":"04/26","energy":55.63},{"date":"04/27","energy":56.4},{"date":"04/28","energy":56.68},{"date":"04/29","energy":56.75},{"date":"04/30","energy":56.23},{"date":"05/01","energy":56.23},{"date":"05/02","energy":56.89},{"date":"05/03","energy":56.64},{"date":"05/04","energy":53.54},{"date":"05/05","energy":51.9},{"date":"05/06","energy":54.53},{"date":"05/07","energy":43.69},{"date":"05/08","energy":53.71},{"date":"05/09","energy":0},{"date":"05/10","energy":49.22}],"total_energy":998.84,"subtitle":""}

Any recommendations? Thanks!

HAEdwin, 2022-05-17:
In the past period I spent some days to see how stability can be increased. Of course I was only able to test with my own ECU (an older ECU-R without SunSpec logo on the back). In my case the number of timeouts has decreased enormously so it benefits me but I have absolutely no idea how the other ECU's react to this. In any case, you can download my code here.
https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip

Extract from the zip only the file: APSystemsSocket.py and overwrite the existing file at: config/custom_components/apsystems_ecur make sure you restart HA afterwards.
(You can also rename the existing file first and then copy the new file into the folder so you can easily restore the old situation)

Not sure if it will help you. What I did was trying to use another method of receiving complete dataframes, added some extra visual log informatie when timeouts or socket errors occur, added a delay between question and answer, added buffer flushing.

theomuys, 2022-05-22:
Thanks @HAEdwin I have installed this patch and will keep you posted.

VikIborra, 2022-06-04:
And other:
```
Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: apsystems_ecur ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur))
First occurred: 08:29:34 (4 occurrences)
Last logged: 08:29:40

APSystemsInvalidData exception: Unsupported inverter type 705 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 705 please create GitHub issue.
```

The IP is 172.16.16.21 and it's visible from my computer and python apps that I made.

HAEdwin, 2022-06-04:
Thank you @VikIborra If you are able to update the integration from my repository using this information https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/75#issuecomment-1128781446 you should be fine. @ksheumaker will update the integration as soon as he reviewed my code and has some spare time left.

HAEdwin, 2022-06-19:
Please let me know if this https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/79#issuecomment-1159744832 solves your problem.

HAEdwin, 2022-08-06:
@Foto005 Was your issue solved with my latest version? https://github.com/HAEdwin/homeassistant-apsystems_ecur/blob/main/custom_components/apsystems_ecur/APSystemsSocket.py

If so please close this issue for I'm not able to moderate this, thank you!

HAEdwin, 2022-08-30:
Should be solved with the latest release, if not please open en new issue.

closed 2022-08-30 by HAEdwin