ksheumaker/homeassistant-apsystems_ecur #76: Unsuported ECU

JoeArny, 2022-05-17:
Hi, I'm getting the error below when attempting to activate the integration:

**Este erro originou-se de uma integração personalizada.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: APSystems PV solar ECU (documentation)
First occurred: 15:32:19 (3 occurrences)
Last logged: 21:21:20

APSystemsInvalidData exception: Unsupported inverter type 706 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 706 please create GitHub issue.**

My ECU model is ECU-R.
Is there any other information I could provide to help? Thanks

HAEdwin, 2022-05-17:
If it's indeed a DS3 inverter you can edit APSystemsSocket.py and modify the line for DS3 inverters like so:
`self.ds3_ids = [ "703", "704", "706" ] `
Just add 706 to the array. It will be included in the next release if it goes well. Please let us know.

JoeArny, 2022-05-17:
Wow! That was fast!!!
It's working now. Thank you very much!

closed 2022-05-17 by JoeArny