ksheumaker/homeassistant-apsystems_ecur #79: Can't connect to ECU-R

jeremiahsb, 2022-06-16:
ECU-R without the SunSpec logo
yc600 inverter with an ID of 216
Connected via WiFi only

I added 216 to the APSystemsSockets.py script (and restarted HA). Upon trying to add the IP address to apsystems_ecur in HA, I get this error in the logs:

This error originated from a custom integration.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:164
Integration: APSystems PV solar ECU (documentation)
First occurred: 2:57:55 PM (1 occurrences)
Last logged: 2:57:55 PM

APSystemsInvalidData exception: Unable to convert binary to int location=333 data=b'4150533131303333323030303230303031000f202206161452514090000942130130310258008f00c200f3002f00f34090000919060130310258009300be00f0003300f04090000917540130310258009300c300f200c200f24090000900150130310258009700c200f100c800f14090000895000130310258009500c100f300c000f34090000892680130310258008e00c300f300c700f34090000889350130310257009100c000f0002f00f04090000941000130310258008b00c400f1000000f14090000918590130310258009500c600f000c400f04090000913920130310258008400c400f3002e00f34090000897860130310257009400c500f200c300f24090000893310130310258008b00c100f0002e00f04090000889980030310000006400000000000000004090000885060130310257009400c100f200c200f2216000020769003030454e440a'
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 159, in aps_int
    return int(binascii.b2a_hex(codec[(start):(start+2)]), 16)
ValueError: invalid literal for int() with base 16: b''

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 148, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 296, in process_inverter_data
    (channel_data, location) = self.process_yc600(data, location)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 376, in process_yc600
    power.append(self.aps_int(data, location))
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 164, in aps_int
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unable to convert binary to int location=333 data=b'4150533131303333323030303230303031000f202206161452514090000942130130310258008f00c200f3002f00f34090000919060130310258009300be00f0003300f04090000917540130310258009300c300f200c200f24090000900150130310258009700c200f100c800f14090000895000130310258009500c100f300c000f34090000892680130310258008e00c300f300c700f34090000889350130310257009100c000f0002f00f04090000941000130310258008b00c400f1000000f14090000918590130310258009500c600f000c400f04090000913920130310258008400c400f3002e00f34090000897860130310257009400c500f200c300f24090000893310130310258008b00c100f0002e00f04090000889980030310000006400000000000000004090000885060130310257009400c100f200c200f2216000020769003030454e440a'

HAEdwin, 2022-06-17:
Hi @jeremiahsb, have you allready tried to use APSystemsSocket.py from my fork? Download it here: https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip

jeremiahsb, 2022-06-17:
@HAEdwin using your APSystemsSocket.py file, without the 216 in the line for YC600 gets this message:
APSystemsInvalidData exception: Unsupported inverter type 216 please create GitHub issue.
With 216 in it, I get the above error.

HAEdwin, 2022-06-19:
It seems that the ECU configuration does not contain a (valid) inverter. The data you posted should contain a value which specifies the number of inverters and that number is zero strange enough. Have you configured the ECU correctly?

HAEdwin, 2022-06-19:
I have updated my fork (quick and dirty), look if this this solves you issue. Download my repository from https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip to replace APSystemsSocket.py.
Let me know if this solves your issue.

jeremiahsb, 2022-06-19:
@HAEdwin yep, that resolved the issue. Thanks!

HAEdwin, 2022-06-23:
@jeremiahsb Have you ever removed/replaced this inverter 216000020769 from the ECU configuration? Just checking new code...
You can close this issue since it is solved. Thanks!

jeremiahsb, 2022-07-01:
@HAEdwin I haven't removed or replaced that inverter.

closed 2022-07-01 by jeremiahsb