ksheumaker/homeassistant-apsystems_ecur #80: ECU-B with DS3

juanka99, 2022-06-19:
Hi. This is my problem: 

Este error se originó a partir de una integración personalizada.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: APSystems PV solar ECU (documentation)
First occurred: 12:02:52 (2 occurrences)
Last logged: 12:03:20

APSystemsInvalidData exception: Unsupported inverter type 8f0 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 8f0 please create GitHub issue.

HAEdwin, 2022-06-19:
Does this post also solve your issue?
https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/79#issuecomment-1159744832
If not tell me more about your ECU model and connected Inverter(s) types

juanka99, 2022-06-19:
> Does this post also solve your issue? [#79 (comment)](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/79#issuecomment-1159744832) If not tell me more about your ECU model and connected Inverter(s) types

Solved. Thanks a lot!

HAEdwin, 2022-06-20:
Ok, I'll check the combination ECU-B and DS3. Can you download my repository once more and uncomment line 234, 251 and 268 then restart HA? You should be able to see some hex data in home-assistant.log can you post me those lines?

juanka99, 2022-06-20:
> Ok, I'll check the combination ECU-B and DS3. Can you download my repository once more and uncomment (remove only the #) line 234, 251 and 268 then restart HA? You should be able to see some hex data in home-assistant.log can you post me those lines?

Hi,
What file do I have to edit?
In APSystemsSocket.py that lines are not commented

HAEdwin, 2022-06-20:
So interpretation of the ECU data is correct, that's great. In this case only uncomment line 268. But first you need to redownload my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip replace APSystemsSocket.py uncomment (remove the # in front of _LOGGER at line 268) and restart HA. Then post the data you'll find in home-assistant.log

juanka99, 2022-06-20:
> So interpretation of the ECU data is correct, that's great. In this case only uncomment line 268. But first you need to redownload my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip replace APSystemsSocket.py uncomment (remove the # in front of _LOGGER at line 268 and 251) and restart HA. Then post the data you'll find in home-assistant.log

This is the log after uncomment line 268:

_b'415053313130303539303030323030303100022022062009510370400000025400303070400000688801303101f40085005800e8005600e8454e440a'
09:54:25 – (ADVERTENCIA) APSystems PV solar ECU (integración personalizada)_

HAEdwin, 2022-06-20:
Thank you, will see if I can find the error

HAEdwin, 2022-06-20:
You have two DS3 inverters and the net frequency is 50Hz and net voltage is 230v right?
Can you post the datastring once again (to see which values are rising/lowering)

juanka99, 2022-06-20:
> You have two DS3 inverters and the net frequency is 50Hz and net voltage is 230v right? Can you post the datastring once again (to see which values are rising/lowering)

Actually I have only one DS3. This one replaced another one that broke down.

juanka99, 2022-06-21:
Ok. I´ll wait for the solution.
Thanks a lot!

HAEdwin, 2022-06-23:
Update, I've created an ECU test utility in which I can also paste all data strings for manual analysis. It's interpreting things correct now. Next step is to rewrite the Python code, hope I'll get to that this weekend.

juanka99, 2022-06-23:
> Update, I've created an ECU test utility in which I can also paste all data strings for manual analysis. It's interpreting things correct now. Next step is to rewrite the Python code, hope I'll get to that this weekend.

Ok. Then, I'll wait you finish the new update.
Thanks a lot!

HAEdwin, 2022-06-23:
This utility will show your data correct if all is good. Would you like to check? https://github.com/HAEdwin/ECUreader

juanka99, 2022-06-23:
> Are you sure you entered the correct IP-address (the same you used in the integration)?

Totally!

[image]

HAEdwin, 2022-06-23:
Did you allready powercycle the ECU? Are you able to connect using Packet Sender with that same IP-address?
[image]

juanka99, 2022-06-23:
> 
[image]

HAEdwin, 2022-06-24:
The errorbox refers to error 10051 (socket error 10051 is commonly caused by problems with a router or modem). But I can see that packet sender is able to connect. Are both programs on the same computer? Do you have a firewall or some virusscanner which might block the ECU reader from connecting to the ECU? Maybe whitelist ECU reader? Sorry and annoying that you get this error message, I'm stepping from one problem to another and of course it works like a charm here... :(
[image]

juanka99, 2022-06-24:
> The errorbox refers to error 10051 (socket error 10051 is commonly caused by problems with a router or modem). But I can see that packet sender is able to connect. Are both programs on the same [...]

I've tried the ECU test utility in another PC, with ethernet connection, and it works fine! ;-))

juanka99, 2022-06-24:
Now I just need the integration for HA to work too. 

Do you thing the problem is the second inverter that is not connected at this moment?

HAEdwin, 2022-06-24:
Great! I think first time you tried, you where on another network segment. Well the data pops-up nicely proving the code is doing the right thing now. I have to convert this to Python. It was indeed the second inverter that causes the error. The current code does not take that into account.

juanka99, 2022-06-28:
> Update, I've created an ECU test utility in which I can also paste all data strings for manual analysis. It's interpreting things correct now. Next step is to rewrite the Python code, hope I'll get to that this weekend.

Hi!
Could you take a look at this?

HAEdwin, 2022-07-06:
Hi @juanka99 could you please try to redownload my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip and replace APSystemsSocket.py ?
It's a bit of a wild guess if this will work for you because I'm not able to test on all hardware variations. For me, owning a  QS1 this works fine. Hope it also works for you.

juanka99, 2022-07-06:
> Hi @juanka99 could you please try to redownload my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip and replace APSystemsSocket.py ? It's a bit of a [...]

Hi! 
It works fine again!
Thank you very much for your work.
You are a machine! ;-)

[image]

HAEdwin, 2022-07-06:
Thanks, hope it will also hold when the inverter is offline (haven't tested that yet). I haven't cleaned up redundant code, will do that if all proves to be working fine.

juanka99, 2022-07-07:
> @juanka99 It didn't so I've updated my repository and removed the redundant check on online status of the inverter. Please redownload my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip and replace APSystemsSocket.py. Sorry for the inconvenience, should be allright now.

Ok. Working...

juanka99, 2022-07-15:
> @juanka99 It didn't so I've updated my repository and removed the redundant check on online status of the inverter. Please redownload my repository [...]

Hello
Today I found another problem. Since I updated HA yesterday, neither the panel nor the temperature values are updated automatically. I have to restart HA so that it reads these values again and they remain fixed until I restart it again.

HAEdwin, 2022-07-15:
Does the integration use the cache? (which is an entity so you can check) If so you might have to powercycle the ECU (turn it off and on again).

juanka99, 2022-07-15:
> Does the integration use the cache? (which is an entity so you can check) If so you might have to powercycle the ECU (turn it off and on again).

I think it isn't using the cache, but I have powercycled the ECU and all is equal

[image]

HAEdwin, 2022-07-16:
> Hello Today I found another problem. Since I updated HA yesterday, neither the panel nor the temperature values are updated automatically. I have to restart HA so that it reads these values again [...]

What do you mean by "...and they remain fixed" / "...until I restart it again"? Do the values not change? What do you restart again? If you use my Windows application does it display the correct data?

juanka99, 2022-07-16:
> What do you mean by "...and they remain fixed" / "...until I restart it again"? Do the values not change? What do you restart again? If you use my Windows application does it display the correct data?

Exactly, the values doesn't change until I restart Home Assistant, and then values remain fixed until I reset HA again.

And yes, your Windows application display the correct data:

[image]

HAEdwin, 2022-07-16:
I noticed the Inverter which is offline and is marked Removed. I struggle with the same issue since I installed my YC600 again yesterday. In fact the EMA configuration must be in sync with the ECU. So make sure the ECU and EMA configuration is correct and is synched with eachother.

juanka99, 2022-07-16:
> I noticed the Inverter which is offline and is marked Removed. I struggle with the same issue since I installed my YC600 again yesterday. In fact the EMA configuration must be in sync with the ECU. So make sure the ECU and EMA configuration is correct and is synched with eachother. The 254 inverter should be removed from the ECU configuration if invalid.

The inverter that is offline and removed doesn't exist because it broke down and I had to replace it with another. This second one is reading the values correctly with your windows application but they doesn't update in Home Assistant.
If I delete inverter 254 I would lose its historical values. Do you think it's the only solution?

HAEdwin, 2022-07-16:
No not sure if this is the only solution. If you reinstalled the integration settings component it overwrites APSystemsSocket.py so you'll have to upload my version again and restart HA.

juanka99, 2022-07-16:
> No not sure if this is the only solution. If you reinstalled the integration settings component it overwrites APSystemsSocket.py so you'll have to upload my version again and restart HA.

I have done what you say (I reinstalled the integration completely even) and and the problem remains

HAEdwin, 2022-07-16:
All the values remain the same since the last succesful update? I suppose they're not zero... and the interval is 300 seconds? Inverters only update once every five minutes so there's no need to read the ECU with a larger interval. Does home-assistant.log show any errors?

juanka99, 2022-07-16:
> All the values remain the same since the last succesful update? I suppose they're not zero... and the interval is 300 seconds? Inverters only update once every five minutes so there's no need to read the ECU with a larger interval. Does home-assistant.log show any errors?

No. All inversor values are read correctly only once when Home Assistant is restarted and do not change until HA is restarted again.

HAEdwin, 2022-07-16:
So might there be a refresh problem in the browser you use to monitor HA? It's hard to discover where the issue might be. I'll be leaving for a two week holiday here. Have to get back to you later.

juanka99, 2022-07-16:
> So might there be a refresh problem in the browser you use to monitor HA? It's hard to discover where the issue might be. I'll be leaving for a two week holiday here. Have to get back to you later.

I don't think the problem is with the browser since the values read from the ECU are updated. Only those read from the inverter do not.
Have a nice holidays!

juanka99, 2022-07-19:
> So might there be a refresh problem in the browser you use to monitor HA? It's hard to discover where the issue might be. I'll be leaving for a two week holiday here. Have to get back to you later.

I've deleted the entities of the 254 inverter in Home Assistant and all it's all right now!

Thanks again!

HAEdwin, 2022-08-06:
@juanka99 Great that all works fine now. Could you please close this issue because I'm not able to moderate.

closed 2022-08-06 by juanka99