ksheumaker/homeassistant-apsystems_ecur #81: issues with ecu-c

doreks, 2022-06-23:
hi!

thank you so much with your support. Im having issues trying to add my ECU-C unit into HA. Here you have the error i got from HA:

`2022-06-23 17:58:07 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS120111000121500001846901\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x0010014ECU-C-Z_C1.2.1012Europe/Paris\x80\x97\x1b\x03$\x8b`\xc5\xa8z\xcb\xfd00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00END\n'
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 134, in query_ecu
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: ECU returned 0 for lifetime energy, raw data=b'APS120111000121500001846901\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00\x0010014ECU-C-Z_C1.2.1012Europe/Paris\x80\x97\x1b\x03$\x8b`\xc5\xa8z\xcb\xfd00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00END\n'
`

any idea? 
thanks for your time and support!

br

HAEdwin, 2022-06-23:
It is more of a coincidence that the ECU-C can be read out with the integration and therefor not future proof. The integration works according to the ECUapp concept that is used when configuring the ECU-R and ECU-B. The ECU-C is configured with the integrated web interface and therefore gives quite different answers that are not anticipated. I don't own an ECU-C myself, so anticipating the incoming data is difficult. Option is to use the utility https://github.com/HAEdwin/ECUreader and see how far you get. For reference, it is then useful to also share current values ​​such as lifetime energy, etc. Maybe better not here but via HA's mailbox. Unfortunately it seems that the integration is only compatible and stable with the ECU-R (older model) and ECU-B.

doreks, 2022-06-24:
@HAEdwin
thank you so much for your support! but a sad news!

I will try to get more info, but I ask, is there a way to be able to have this utility:

https://github.com/HAEdwin/ECUreader

in a linux, phyton or .sh version.
I don't have a windows computer to be able to run that .exe. 
:-(

doreks, 2022-07-17:
Finally now it's working, thanks for your support. I would like to know if there is any way to get values of grid export import or used energy. Currently on app apsystem I can get this info but not into home assistant using your hacs integrations.

Thanks for your time and support?!

HAEdwin, 2022-08-06:
It's a bit offtopic but if you have a smart meter you can use a p1 sensor and put the energy dashboard into work for home assistant. So you have to configure energy management.

doreks, 2022-08-06:
Hi @HAEdwin,

Thank you so much for your reply.

Already I have shelly smart meter. I'm getting all the meters from there but I'm so interested in go through your comment.. Could you pleae explain a little bit more the setup.
Just for your info shelly is not able to read any meter directly connecting to ecu c.

Thanks!

HAEdwin, 2022-08-06:
If you would like to setup your energy dashboard, go to [Settings] > [Dashboards] > [Energy] there you should be able to add parameters like consumed energy and generated energy (even costs and forecast production). So the consumed energy parameters are supplied by your Shelly sensor, the generated parameters should be supplied by the integration. At the solar panels section add the sensor.ecu_lifetime_energy parameter. Your dashboard could look something like this.
[image]

HAEdwin, 2022-08-30:
> Finally now it's working, thanks for your support. I would like to know if there is any way to get values of grid export import or used energy. Currently on app apsystem I can get this info but not into home assistant using your hacs integrations.
> 
> Thanks for your time and support

Closed issue. If you have any issues please open a new issue.

closed 2022-08-30 by HAEdwin