ksheumaker/homeassistant-apsystems_ecur #85: Please add inverter type 702 für DS3-S

alpapst, 2022-07-17:
Adding ECU-B gave me the following log entry:

`ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Unsupported inverter type 702 please create GitHub issue.`

imonlinux, 2022-07-26:
My APsystem solar was just installed and I'm getting the same HA log entry as @alpapst above.

APSystemsInvalidData exception: Unsupported inverter type 702 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 702 please create GitHub issue.

Thanks for your work.

HAEdwin, 2022-07-29:
Sorry for my late response, I was on vacation. There are two ways for solving this.
1. Edit APSystemsSocket.py on line 48 add the 702 inverter type (restart home assistant afterwards)
2. recommended method: download my repository at https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip and replace APSystemsSocket.py (restart home assistant afterwards)

HAEdwin, 2022-08-06:
@alpapst @imonlinux  Did my response solve you issues? If so @alpapst please close this issue for I'm not able to moderate this. Thank you!

imonlinux, 2022-08-07:
Thanks for the instructions. I just made the change and it is now working as expected.

Thanks again for your work.

HAEdwin, 2022-08-12:
@alpapst please close this issue for I'm not able to moderate this. Thank you!

alpapst, 2022-08-12:
@HAEdwin Sorry for the late response, I was on vacation and had no access to my GitHub account. Thank you for your support!

closed 2022-08-12 by alpapst