ksheumaker/homeassistant-apsystems_ecur #88: New Device Type 702 for  AP-Systems DS3-S-EU

andreassolarteur, 2022-07-31:
2022-07-31 16:44:21 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Unsupported inverter type 702 please create GitHub issue.
Traceback (most recent call last):
  File "/var/snap/home-assistant-snap/509/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/snap/home-assistant-snap/509/usr/local/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/var/snap/home-assistant-snap/509/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/var/snap/home-assistant-snap/509/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 702 please create GitHub issue.

HAEdwin, 2022-08-01:
Hi, this issue was solved here: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/85#issuecomment-1198982084

andreassolarteur, 2022-08-02:
Thank you, excuse me i am a new bee.

closed 2022-08-02 by andreassolarteur