ksheumaker/homeassistant-apsystems_ecur #94: Error at Configuration - Unsupported Inverter

Patte67, 2022-08-13:
2022-08-13 10:55:58.069 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: Unsupported inverter type 702 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 702 please create GitHub issue.

HAEdwin, 2022-08-13:
Issue is solved here: https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/85#issuecomment-1198982084

HAEdwin, 2022-08-14:
@Patte67 Please close this issue if it did solve your problem. I'm not able to moderate this. Thank you!

Patte67, 2022-08-14:
It works :)

closed 2022-08-14 by Patte67