ksheumaker/homeassistant-apsystems_ecur #96: ECU-C not responding after xx time. + solution intent

12christiaan, 2022-08-16:
I am a bit new to the ECU-C integration (less than 1 week).
but this morning the value in HASS flatlined. and when i looked in HASS, it say that the values where not undated for 3 hours.

I had PCAP running, so if someone is intrested, i can make them availible. but my first conclusion is that the TCP session to the ECU is accepted, but the ECU just does not respond with the expected data.

i was able to recover from the situation WITHOUT power cycle the ECU !!!!

just 1 simple command is suficiant to reboot the ECU

`curl "http:// < IP > /index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttpRequest" --data-raw "SSID=ECU-WIFI_local&channel=0&method=0&psk_wep=&psk_wpa="`

hope that some one can automate this in the scripts. Let's say if the ECU did not respond for 5 times?

HAEdwin, 2022-08-16:
That is very good news, you might have to create a switch. After this you can call it in an automation. https://community.home-assistant.io/t/automation-execute-curl-command/134909/2

I'm interested in the PCAP logs, I never had a chance to analyze the traffic to the ECU-C because I only own the good old classic ECU-R. The ECU-R with ECU-ID's starting with 2162..... have the same problem (not responding after a while).

HAEdwin, 2022-08-16:
You can choose to scedule a restart once a week or shorter interval or you might check on {{ states('switch.ecu_query_device') }}
When all fails and cache limit is reached it will turn off if I'm not mistaken.

HAEdwin, 2022-08-16:
> 

Got the .zip thnx

12christiaan, 2022-08-17:
where did you see duplicate IP-address

HAEdwin, 2022-08-18:
In the ECU-C_Responding log: [Duplicate IP address detected for 192.168.0.70 (60:c5:a8:79:a0:b8) - also in use by 80:97:1b:02:1d:91 (frame 4)]
60:c5:a8:79:a0:b8 = A node from Beijing LT Honway Technology Co.,Ltd
80:97:1b:02:1d:91 = A node from Altenergy Power System,Inc. (your ECU)
The first could well be the bluetooth device in the Rpi, I'm not sure - just noticed this, might be nothing.

12christiaan, 2022-08-18:
That is a intressting mistack from the ECU.
ECU Eth0 Mac Address | 80:97:1B:02:1D:91 (ip 192.168.0.70)
ECU Wlan0 Mac Address | 60:C5:A8:79:A0:B8 (ip 192.168.0.71)

HAEdwin, 2022-08-21:
@12christiaan Are you still testing the trigger on switch.ecu_query_device or are you scheduling the reset? I might add an sensor if switch.ecu_query_device isn't working.

12christiaan, 2022-08-22:
yes, but i have to wait for the ECU to stop, to see if it works. can take a few days

12christiaan, 2022-08-23:
my automation did not work (detecting the 'unav") needed to re-start manualy.

HAEdwin, 2022-08-23:
Have you looked at the state of switch.ecu_query_device?

12christiaan, 2022-08-23:
automation

`- id: '1661155335281'
  alias: Reboot ECU if unavailable readings.
  description: ''
  trigger:
  - platform: state
    entity_id:
    - switch.ecu_query_device
    to: unavailable
    for:
      hours: 0
      minutes: 3
      seconds: 0
  condition: []
  action:
  - service: shell_command.shell_reboot_ecu
    data: {}
  mode: single`

HAEdwin, 2022-08-24:
Did some testing also (ECU-R powered off). After using the cache five times, the ecu_query_device switch flips the 6th time but will not trigger the automation. Using the switch manually does trigger the automation. I'll look into this later.

12christiaan, 2022-08-24:
my observation:

1) disconnect cable.
2) next polling of the ECU (0-5 minute):
ECU Using Cached Data turned on = binary_sensor.ecu_using_cached_data 

3) 25 minute later:
ECU Inverters became unavailable = sensor.ecu_inverters
ECU Inverters Online became unavailable = sensor.ecu_inverters_online
ECU Using Cached Data became unavailable = binary_sensor.ecu_using_cached_data
ECU Query Device became unavailable = switch.ecu_query_device

4) Cable back.

that is why i made the automation for "binary_sensor.ecu_using_cached_data"

HAEdwin, 2022-08-26:
You can also use (might be better to use because it is more generetic):

```
alias: Reboot ECU if unavailable readings
description: ""
trigger:
  - platform: state
    entity_id:
      - binary_sensor.ecu_using_cached_data
        to: unavailable
        condition: []
action:
  - service: shell_command.shell_reboot_ecu
    data: {}
    mode: single
```

HAEdwin, 2022-08-29:
@12christiaan Any news on the automation? I might include it in the next release and/or add your finding to the readme.

12christiaan, 2022-08-30:
Last week I was testing with disconnecting the cable. now running for (almost) 3 days, so expect it to fail soon. (and hopefuly recover automaticly).

12christiaan, 2022-08-31:
To day morning it started with failure @7:58. "Using cached data"
However, the system DID respond next Query and everything looked fine.
My automation did NOT trigger because it need 16 minute (3 query) to be active.

2022-08-31 07:53:19 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU
2022-08-31 07:53:34 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Got data from ECU
2022-08-31 07:53:34 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', ...............
2022-08-31 07:53:34 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.035 seconds (success: True)

2022-08-31 07:58:34 DEBUG (SyncWorker_4) [custom_components.apsystems_ecur] Querying ECU
2022-08-31 07:58:39 WARNING (SyncWorker_4) [custom_components.apsystems_ecur.APSystemsSocket] b'4150...............
2022-08-31 07:58:39 WARNING (SyncWorker_4) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Error: ECU returned 0 for lifetime energy, raw data=b'AP-C-Z_C1.2.2016Europe/Amsterdam\x80\x97\x1b\x02\x1d\x91`\xc5\xa8y\xa0\xb800\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00END\n'
7:58:39 AM - ECU Using Cached Data turned on
2022-08-31 07:58:39 DEBUG (SyncWorker_4) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', ...............
2022-08-31 07:58:39 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 5.011 seconds (success: True)

2022-08-31 08:03:39 DEBUG (SyncWorker_4) [custom_components.apsystems_ecur] Querying ECU
2022-08-31 08:03:54 DEBUG (SyncWorker_4) [custom_components.apsystems_ecur] Got data from ECU
2022-08-31 08:03:54 DEBUG (SyncWorker_4) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', ...............
2022-08-31 08:03:54 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.068 seconds (success: True)
8:03:54 AM - ECU Using Cached Data turned off

This afternoon, things did go wrong.

2022-08-31 15:08:54 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Querying ECU
2022-08-31 15:08:59 WARNING (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313230313135303030313231353030303031323530323031000006e100000562000002f4202208311506030003000331303031344543552d432d5a5f43312e322e323031364575726f70652f416d7374657264616d80971b021d9160c5a879a0b8303000000000000000000000454e440a'
2022-08-31 15:09:09 WARNING (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303932303030323030303100032022083115032470400002423901303101f40097019300f3019700f370400002343501303101f40094019700f2005400f270400002368201303101f40088002c00f1002100f1454e440a'
2022-08-31 15:09:09 WARNING (SyncWorker_6) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303339303033303030704000024239cf704000023435da704000023682de454e440a'
2022-08-31 15:09:09 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Got data from ECU
2022-08-31 15:09:09 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [44, 33], 'voltage': [241, 241], 'inverters': {'704000024239'_id': '215000012502', 'today_energy': 7.56, 'lifetime_energy': 176.1, 'current_power': 1378, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-08-31 15:09:09 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.028 seconds (success: True)

2022-08-31 15:14:09 DEBUG (SyncWorker_8) [custom_components.apsystems_ecur] Querying ECU
2022-08-31 15:14:14 WARNING (SyncWorker_8) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313230303939303030313231353030303031323530323031000006e3000004ec00000309202208311511020003000331303031344543552d432d5a5f43312e322e3230303080971b021d9160c5a879a0b8303000000000000000000000454e440a'
2022-08-31 15:14:24 WARNING (SyncWorker_8) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303138303030323031454e440a'
2022-08-31 15:14:24 WARNING (SyncWorker_8) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Error: 'NoneType' object does not support item assignment
2022-08-31 15:14:24 DEBUG (SyncWorker_8) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', ...............
2022-08-31 15:14:24 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.085 seconds (success: True)
3:14:24 PM - ECU Using Cached Data turned on

2022-08-31 15:19:24 DEBUG (SyncWorker_3) [custom_components.apsystems_ecur] Querying ECU
This is strange... There is NO responce. Nore end of the query. It looks IT DOES NOT time out.

Than at 15:30. My automation kicked in.
Triggered by the state of binary_sensor.ecu_using_cached_data at August 31, 2022 at 3:30:24 PM
Call service shell_command.shell_reboot_ecu
Finished at August 31, 2022 at 3:30:25 PM (runtime: 1.85 seconds)

2022-08-31 15:30:25 DEBUG (MainThread) [homeassistant.components.shell_command] Stdout of command: `/usr/bin/curl "http://192.168.0.70/index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttre `, return code: 0:
2022-08-31 15:30:25 DEBUG (MainThread) [homeassistant.components.shell_command] Stderr of command: `/usr/bin/curl "http://192.168.0.70/index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttre `, return code: 0:
[... cut ...] ': 220, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-08-31 18:20:00 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.068 seconds (success: True)

6:20:00 PM - ECU Using Cached Data turned off

From this moment the data is normaly arriving and looks working again.

My conclusion is:
The reset of the ECU DID work fine.
however the query proccess was hanging, so that the querying did not start.

HAEdwin, 2022-08-31:
Thanks for the comprehensive report, I see plenty of things to investigate/address again.

HAEdwin, 2022-09-03:
One consideration is not to mark the ECU as unavailable on the 6th attempt to read data but to toggle the query switch instead. The integration will now just cache indefinitely until the switch is flipped again. The switch thus also retains its original purpose; optionally stopping the ECU querying when the sun has set.

An additional advantage is that the entities remain intact and do not fill the dashboard with unavailable entity statuses, that might cause some panic.

Your automation could then detect that the switch is off, reset the ECU and then flip the switch back on.

Does this sound logical?

By the way, the switch bounced so I had to debounce that first. Maybe that's why the automation didn't work. In the meantime also corrected the raises on invalid data.

HAEdwin, 2022-09-03:
Ok, ready testing and updated my repository. Automation now works but under the condition that the querycache is also enabled. If you don't add the condition the automation will be triggered when you manually flip the switch. Automation would look like this:

```
- id: '[automation ID]'
  alias: When the ECU does not repond anymore
  description: ''
  trigger:
  - platform: state
    entity_id:
    - switch.ecu_query_device
    to: 'off'
  condition:
  - condition: and
    conditions:
    - condition: state
      entity_id: binary_sensor.ecu_using_cached_data
      state: 'on'
  action:
  - service: notify.mobile_app_iphone_from_Edwin
    data:
      message: ECU unavailable
      title: Error
  mode: single
```

The switch is not bounching anymore and the entities remain intact/enabled/available for automations.
If you'd like to test, you can download the sources https://github.com/HAEdwin/homeassistant-apsystems_ecur/archive/refs/heads/main.zip and replace the following files:
__init__.py
APSystemsSocket.py
switch.py

12christiaan, 2022-09-04:
I will test.

12christiaan, 2022-09-04:
I have updated the items you said. and chenged my automation to this.
Now wait for 3 or 4 day till automation is triggered. 

- id: '1661155335281'
  alias: Reboot ECU if unavailable readings.
  description: ''
  trigger:
  - platform: state
    entity_id:
    - binary_sensor.ecu_using_cached_data
    to: 'on'
    for:
      hours: 0
      minutes: 16
      seconds: 0
    from: 'off'
  condition: []
  action:
  - service: notify.mobile_app_21081111rg
    data:
      message: ECU unavailible
      title: ECU
  - service: shell_command.shell_reboot_ecu
    data: {}
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - service: switch.turn_off
    data: {}
    target:
      entity_id: switch.ecu_query_device
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.ecu_query_device
  mode: single

nielssteenvoorde, 2022-09-05:
New versions here freezes also after a couple of days (ECU-C). Looks promising guys.

12christiaan, 2022-09-06:
Yes, connected to ethernet. however the same service is availible via WiFi. I disconnected the WiFi, due to duplicated MAC error.

12christiaan, 2022-09-08:
Just came home and saw that the ECU failed today.
both Automation where triggered.
1:05:20 PM - ECU Using Cached Data turned on
1:22:22 PM - ECU Query Device turned off triggered by automation ECU Reboot if unavailable readings. triggered by state of ECU Using Cached Data
1:23:22 PM - ECU Query Device turned on triggered by automation ECU Reboot if unavailable readings. triggered by state of ECU Using Cached Data

Neverhteless, the "ECU Using Cached Data" whent on, and never when off again.
I see that the Quering of the ECU stopped after the automation is triggered.
The automation switched "ECU Query Device" Off and ON but did not start Queriing.
I switched manualy "ECU Query Device" Off and ON but did not start Queriing.

2022-09-08 11:57:05 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 11:57:05 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [368, 353], 'voltage': [238, 238], 'inverters': {'70400002423_id': '215000012502', 'today_energy': 1.02, 'lifetime_energy': 235.4, 'current_power': 1520, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 11:57:05 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.022 seconds (success: True)
2022-09-08 12:02:05 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:02:20 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:02:20 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [66, 58], 'voltage': [236, 236], 'inverters': {'704000024239'_id': '215000012502', 'today_energy': 1.05, 'lifetime_energy': 235.4, 'current_power': 399, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 12:02:20 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.550 seconds (success: True)
2022-09-08 12:07:20 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:07:35 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:07:35 DEBUG (SyncWorker_7) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [171, 158], 'voltage': [240, 240], 'inverters': {'70400002423_id': '215000012502', 'today_energy': 1.17, 'lifetime_energy': 235.5, 'current_power': 908, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 12:07:35 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.027 seconds (success: True)
2022-09-08 12:12:35 DEBUG (SyncWorker_9) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:12:50 DEBUG (SyncWorker_9) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:12:50 DEBUG (SyncWorker_9) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [181, 171], 'voltage': [239, 239], 'inverters': {'70400002423_id': '215000012502', 'today_energy': 1.25, 'lifetime_energy': 235.6, 'current_power': 902, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 12:12:50 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.581 seconds (success: True)
2022-09-08 12:17:50 DEBUG (SyncWorker_8) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:18:05 DEBUG (SyncWorker_8) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:18:05 DEBUG (SyncWorker_8) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [291, 276], 'voltage': [242, 242], 'inverters': {'70400002423_id': '215000012502', 'today_energy': 1.36, 'lifetime_energy': 235.7, 'current_power': 1385, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 12:18:05 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.032 seconds (success: True)
2022-09-08 12:23:05 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:23:20 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:23:20 DEBUG (SyncWorker_6) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [217, 203], 'voltage': [236, 236], 'inverters': {'70400002423_id': '215000012502', 'today_energy': 1.44, 'lifetime_energy': 235.8, 'current_power': 1080, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying': True}
2022-09-08 12:23:20 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.610 seconds (success: True)
2022-09-08 12:28:20 DEBUG (SyncWorker_13) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 12:28:35 DEBUG (SyncWorker_13) [custom_components.apsystems_ecur] Got data from ECU
2022-09-08 12:28:35 DEBUG (SyncWorker_13) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [316, 300], 'voltage': [241, 241], 'inverters': {'7040000242_id': '215000012502', 'today_energy': 1.56, 'lifetime_energy': 235.9, 'current_power': 1531, 'qty_of_inverters': 3, 'qty_of_online_inverters': 3, 'data_from_cache': False, 'querying [... cut ...]
2022-09-08 13:10:20 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU
2022-09-08 13:21:22 DEBUG (MainThread) [homeassistant.components.shell_command] Stdout of command: `/usr/bin/curl "http://192.168.0.70/index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttre `, return code: 0:
2022-09-08 13:21:22 DEBUG (MainThread) [homeassistant.components.shell_command] Stderr of command: `/usr/bin/curl "http://192.168.0.70/index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttre `, return code: 0:
No more entries (from ECU) after this.

HAEdwin, 2022-09-09:
@12christiaan Thanks for the feedback. I'll have to check with you if expectations are correct. The integration never stops the process of updating with the set interval.

While the update process is running, it checks if caching is active, the ECU is reachable, the switch is on, etc. If you flip the ECU_Query_Device switch, the status of everything is only updated at the next interval that the update process is running. So don't expect a disabled cache right away, but wait for the minimum set interval before this changes. 

I kind of assume that issues are handled unattended where you don't look at the dashboard. Perhaps you concluded too early that the process is returning to normal operations and the error in the log was an accidental interruption of the ongoing process.

So reproduce, flip the switch wait and see what is happening with the "ECU Using Cached Data" status after the set interval.
I would recommend testing after sunset by unplugging the ethernet cable and then waiting to see what happens without putting an automation on it. When you reconnect the Ethernet cable, the normal situation will return after the interval.

HAEdwin, 2022-09-12:
@12christiaan Your automation contains
```
- service: switch.turn_off
    data: {}
    target:
      entity_id: switch.ecu_query_device
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.ecu_query_device
```
But this is redundant because the integration allready flips the ECU Query Device switch off. The automation should therefor only check if [ECU Using Cached Data = On] AND [ECU Query Device = Off]. If these conditions are true, send the cURL and turn the switch [ECU Query Device] to on.

```
- id: '1661155335281'
  alias: Reboot ECU if unavailable readings.
  description: ''
  trigger:
  - platform: state
    entity_id:
    - binary_sensor.ecu_using_cached_data
    to: 'on'
  condition:
  - condition: and
    conditions:
    - condition: state
      entity_id: switch.ecu_query_device
      state: 'off'
  action:
  - service: notify.mobile_app_21081111rg
    data:
      message: ECU unavailible
      title: ECU
  - service: shell_command.shell_reboot_ecu
    data: {}
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.ecu_query_device
  mode: single

```

Firegun1988, 2022-09-13:
I have seen that after re-installing the ECU-R integration in HACS en rebooting HA, the ECU-R will give the information just without any reboot or adjustments.

HAEdwin, 2022-09-14:
@Firegun1988 It depends on your ECU-R model whether you need to apply the automation. Older ECU-R models that start with 2160xxxxxx indeed have no problem that they get stuck. I will improve the readme with all the info supplied. Thanks all!

HAEdwin, 2022-09-17:
@12christiaan I'll add your findings and my automation to the readme and release a new version by the end of this month. Is the automation using the shell command correct the way I described it here?

In automations.yaml
```
- id: '1661155335281'
alias: Reboot ECU if unavailable readings
description: ""
trigger:
  - platform: state
    entity_id:
      - switch.ecu_query_device
    to: "off"
condition:
  - condition: and
    conditions:
      - condition: state
        entity_id: binary_sensor.ecu_using_cached_data
        state: "on"
action:
  - service: notify.mobile_app_iphone_van_edwin
    data:
      message: ECU unavailable
      title: ECU
  - service: shell_command.reset_ecu
    data: {}
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - service: switch.turn_on
    data: {}
    target:
      entity_id: switch.ecu_query_device
mode: single
```

In configuration.yaml (don't forget to specify your ECU IP-Address):
```
shell_command:
  reset_ecu: "curl 'http://<ip-address>/index.php/management/set_wlan_ap' -H 'X-Requested-With: XMLHttpRequest' --data-raw 'SSID=ECU-WIFI_local&channel=0&method=0&psk_wep=&psk_wpa='"
```

12christiaan, 2022-09-18:
Sorry, i am very bussy, no time to work on this. 
for now i have disabled my automations, and send shell command very night, this works fine (for now).

My shell command
#
shell_command:
  shell_reboot_ecu: '/usr/bin/curl "http://192.168.0.70/index.php/management/set_wlan_ap" -H "X-Requested-With: XMLHttpRequest" --data-raw "SSID=ECU-WIFI_local&channel=0&method=0&psk_wep=&psk_wpa=" --compressed  --insecure '
#

HAEdwin, 2022-09-25:
Added to release v1.2.18 thank you again @12christiaan. I will close this issue. Should there be any questions or comments on this issue, you are welcome to open a new issue and refer to this issue.

closed 2022-09-25 by HAEdwin

HAEdwin, 2022-10-05:
@12christiaan I could as well add an option to the configuration if you own the ECU-R (pro) or ECU-C model and call the URL from within the integration when reboot is applicable, that's much easier. Are you willing to test if I do?