ksheumaker/homeassistant-apsystems_ecur #97: Unsupported Inverter Type 702

Porsti, 2022-08-19:
I get an error when i am trying to connect the ecu-b with inverter APSystem DS3-S.
The log shows:
Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:348
Integration: APSystems PV solar ECU ([documentation](https://github.com/ksheumaker/homeassistant-apsystems_ecur))
First occurred: 14:27:25 (5 occurrences)
Last logged: 14:39:59

APSystemsInvalidData exception: Unsupported inverter type 702 please create GitHub issue.
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 45, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 158, in query_ecu
    data = self.process_inverter_data()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 348, in process_inverter_data
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: Unsupported inverter type 702 please create GitHub issue.

Could you help?

Thank you

Thomas

HAEdwin, 2022-08-19:
Hello @Porsti,

Read this part to solve your issue: https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/95
I've done a pull request and waiting for it to include it in the new release.

HAEdwin, 2022-08-21:
@Porsti Thomas, if it solved you issue, can you please mark this as solved/closed for I'm not able to moderate this. Thank you.

Porsti, 2022-08-22:
@HAEdwin  Thank you. This solved the problem. The integration was able to start and found my inverter.

closed 2022-08-22 by Porsti