ksheumaker/homeassistant-apsystems_ecur #103: IndexError: list index out of range

JohnnyPicnic, 2022-08-26:
When toggling the query switch I get an error "IndexError: list index out of range"
It looks like the main ECU updates but the individual inverters are not updating.
There are three separate errors:
`This error originated from a custom integration.

Logger: homeassistant
Source: custom_components/apsystems_ecur/sensor.py:133
Integration: APSystems PV solar ECU (documentation)
First occurred: August 21, 2022, 5:03:03 PM (1300 occurrences)
Last logged: 9:01:38 AM

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 151, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 110, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 348, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/config/custom_components/apsystems_ecur/sensor.py", line 133, in state
    return self.coordinator.data.get("inverters", {}).get(self._uid, {}).get("voltage", [])[0]
IndexError: list index out of range
`

`Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/apsystems_ecur/sensor.py:133
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: August 24, 2022, 7:12:32 PM (3 occurrences)
Last logged: 9:04:11 AM

[140299712788080] list index out of range
[140298448364224] list index out of range
[140298920775056] list index out of range
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1713, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/config/custom_components/apsystems_ecur/switch.py", line 77, in async_turn_off
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 158, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 82, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 110, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 348, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/config/custom_components/apsystems_ecur/sensor.py", line 133, in state
    return self.coordinator.data.get("inverters", {}).get(self._uid, {}).get("voltage", [])[0]
IndexError: list index out of range
`

`This error originated from a custom integration.

Logger: homeassistant.core
Source: custom_components/apsystems_ecur/sensor.py:133
Integration: APSystems PV solar ECU (documentation)
First occurred: August 24, 2022, 7:12:51 PM (3 occurrences)
Last logged: 9:04:32 AM

Error executing service: <ServiceCall switch.turn_on (c:01GB98XRP8V8F5CE6B1F9F4T1P): entity_id=['switch.ecu_query_device']>
Error executing service: <ServiceCall switch.turn_on (c:01GB9D4WAA2MBMSMBPC1YBB82V): entity_id=['switch.ecu_query_device']>
Error executing service: <ServiceCall switch.turn_on (c:01GBDAXC1BS94P176PJJNGNXR2): entity_id=['switch.ecu_query_device']>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1731, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1750, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 204, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 676, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 930, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 713, in _handle_entity_call
    await result
  File "/config/custom_components/apsystems_ecur/switch.py", line 81, in async_turn_on
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 158, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 82, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 182, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 110, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 348, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/config/custom_components/apsystems_ecur/sensor.py", line 133, in state
    return self.coordinator.data.get("inverters", {}).get(self._uid, {}).get("voltage", [])[0]
IndexError: list index out of range
`

HAEdwin, 2022-08-26:
Are you using the latest beta version? If you are not, please try: https://github.com/ksheumaker/homeassistant-apsystems_ecur#beta-program

JohnnyPicnic, 2022-08-27:
Yes I switched to the beta so I'm running v1.2.15. It does seem to be working better than the main version but I'm still getting the index out of range error. 

Firmware: ECU_R_1.2.22
Inverters are YC600

[home-assistant.log](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/9437019/home-assistant.log)

HAEdwin, 2022-08-27:
Thank you, you can stop the logging again. For my understanding and impact, for what purpose do you use the switch? With your ECU-R there is no need to stop logging when the inverters are offline. I leave it querying 24/7.

JohnnyPicnic, 2022-08-27:
When I first loaded the integration I was using the main version and although it loaded and populated all the sensors, the inverter data would not updated. So I thought it was related to the query switch.
Now I see I don't need to use the switch.
Thanks!

closed 2022-08-27 by JohnnyPicnic