ksheumaker/homeassistant-apsystems_ecur #105: ECU-R not able to add integrationi

clardima, 2022-08-30:
Hey Guys,

Finally had some time to play with this integration. I setup my ECU-R over my WIFI and the IP address is available.
Now I want to add the integration but I get an error message after filling in the IP address [%key:common::config_flow::error::unknown%]

Must I setup something else first?

Thanks.

regards
Maarten

HAEdwin, 2022-08-30:
Did you follow the instructions https://github.com/ksheumaker/homeassistant-apsystems_ecur#setup ? You might have to restart HA also.

clardima, 2022-08-30:
Yes, I did restart HA, I also restarted ECU-R, I see data coming in on the webportal of APSystems, so everything is connected.
IP-address is filled in correctly.

HAEdwin, 2022-08-30:
The ultimate test to see if you've got the IP-address right is to use Packet Sender. Enter the data marked red and see if you get a response after selecting [Send]
[image]
If the IP-address is correct then you'll have to check if HA can reach the ECU or check the network segment.

clardima, 2022-08-30:
Packet Sender gives an error. So I changed the IP adres from DHCP to fixed IP, just to try. Now it still did not work.

I installed Advanced Portscanner and after a while port 6668 seems to be open, connecting with Packet Sebder is possible, but I am not sure if this is the correct way.  I do not see port 8899 open.

clardima, 2022-08-30:
Question: Is APS1100160001END a default value, or should I change it, depending on my own ECU-R?

HAEdwin, 2022-08-30:
Leave it as it is, it's the first command to retreive ECU info (like firmware version etc.) do not use random commands. 
You'll have to use 8899 others are just temporary open to push data to EMA and cannot be used. When you got Packet Sender working, the integration will also work.

clardima, 2022-08-30:
But I can connect to Packet Sender, but not on the correct port, so connectivity is okay to the ECU-R. Only hte port is somehow blocked. Is it possible that this port is disabled on the ECU-R?

HAEdwin, 2022-08-31:
Make absolutely shure you have the ip-address from the ECU. You are not the first to find out that the wrong IP address was used because the ESP interface in the router resembled it. In your router, find the ECU-R which can be seen as ESP-xxxxxx. MAC starts with D8:BF:C0: (Espressif Inc.)

clardima, 2022-08-31:
I am definitely sure I have the correct IP address. After setting up the fixed IP address in my Fritz.box and restarting the ECU-R, the fixed IP address became active on my WiFi network. 
But.... my MAC start with A4:E5;7C, so it's different from yours. But... the manufacturer states: Espressif Inc., so that should be right.

Are there different ECU-R versions available?

If I look in the EMA Manager app on my mobile, I see version 1.2.22, ID starts with 2160*.

HAEdwin, 2022-08-31:
We both have the same ECU-R with the same firmware. The ECU is still only WiFi connected to your LAN right? For this specific ECU (and ECU-B) the integration only works on Wifi connected ECU. If I use Packet Sender as specified above, (my ECU is at 172.16.0.4) I get immediate responses after pressing the Send button. That's the top line in the log: from ECU (8899) to PC (21607). This tool is the ultimate test to see if the integration would work for you (assuming only one network segment exists).

If you are absolutely sure you've got the correct ip-address and the firmware is the same with both of us, the only thing I can think of is that some firewall software on the PC running Packet Sender or router blocks the port and we are using a flaw in the older type of ESP interfaces that is revised in a later version you own. Maybe silly because it's not using 8899 but do a ping [ECU ip-address] -t on the PC to the WiFi connected ECU. You get a response right? Now pull the power plug from the ECU, responses are timed out right?

There is also a newer ECU-R with SunSpec logo on the back. Unit ID starts with 2162xxxxx and is supposed to be modbus compliant. It holds ECU-R-Pro firmware although there officially is no ECU-R pro model released. The integration works on this model when it is attached to the LAN by ethernet cable. Unfortunately this model stops responding after a while (days). Hopefully there will be a solution in issuing a curl command URL (because opposed to our ECU-R model, the ECU-R-pro has a webinterface).

HAEdwin, 2022-08-31:
Check: You have choosen you WiFi SSID in the ECU and provided the WiFi password from that SSID. I've also had someone that was connecting to the temporary hotspot from the ECU. But in your case you wrote that data is being received on the EMA site so this should be allright then.

HAEdwin, 2022-09-16:
@clardima Did you solve your connection issue? Because we use the same firmware/hardware the only issue here might be your network.

closed 2022-10-09 by HAEdwin

letrouf, 2025-03-08:
Hi, I seem to have the exact same issue.
Everything was working fine but I had to change my router, I manage to get everything back on track except for this integration. 
Ecu-R have a fixed IP adress, is well connected to the network.
Can't manage to get data from port 8899. In terminal and with pack sender on PC. Other ports seems to work fine though like @clardima mentionned.
If @clardima is around, I would love to see if he fixed it.

[image]

HAEdwin, 2025-03-09:
Hi @letrouf tell me what ECU-R you own (2162xxxx or 2160xxxx). Make sure you have the correct node IP and the ECU is able to access the internet freely. When you turn off the ECU does a ping stop also? Please note that with the new ECU firmware you will have to wait 10 minutes after a restart of the ECU. 

Try https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader because I continued development there. I've also a proxy method https://github.com/HAEdwin/homeassistant-apsystems_ecu_proxy you could use when you own a 2160xxxx

letrouf, 2025-03-09:
Hi, It's an Ecu-R 2162, firmware is ECU_R_PRO_2.1.23. Ecu is uploading fine to Apsystems servers.
I tried the new integration but without access to port 8899, it cannot load the integration properly too. Ping is down when offline.

[image]

HAEdwin, 2025-03-09:
I have the same ECU (and firmware). I assume the ECU is connected wirelessly to your network via WiFi? Nothing is blocking traffic on the network and HA is on the same network as the ECU (same vlan and/or subnet)? It might indeed happen that the integration is unable to connect (open the socket). It will then retry the number of times specified. In most cases where users are unable to connect to the ECU they made some kind of mistake in the network configuration. Hard to tell what it is in your case but because you mention the new router I would expect some sort of firewall issue on the internal network.

letrouf, 2025-03-16:
Hi. I'm back after multiple testing. I set my old router back, but in fact what I thought was the Ecu was not. I have tested ith packet sender almost all my no-named devices and it appears my ecu does not have its mac adress beginning with D8:BF:C0:... but ac:48:2d...
All works fine now, the new integration too.
Thanks

HAEdwin, 2025-03-16:
Thanks for the feedback, glad it's resolved for you.