ksheumaker/homeassistant-apsystems_ecur #107: No integration in Energy Management

arduinopjotr, 2022-09-14:
Hello,
I have succesfuly setup the integration in Home Assistant. However, i want to integrate the ECU-R in the Energy monitor. Within the energy monitor i already configured the smartmeter. 
https://www.home-assistant.io/docs/energy/electricity-grid/

When i want to add the Solar production i need to add the entity from the ECU-R named "sensor.ecu_today_energy" but i it not visible. Home Assistant gives the clue because within the integration there must be told that this entity must me configured as an entity which to track statistics 
https://www.home-assistant.io/more-info/statistics/

Is this a known issue ? Is ther maybe an workarround ?
Thanks

HAEdwin, 2022-09-14:
You'll have to use sensor.ecu_lifetime_energy
It takes around two hours before statistics become available.

arduinopjotr, 2022-09-14:
Hello Edwin,  I just configured the entity "sensor.ecu_lifetime_energy" in the recorder and after that the entity "sensor.ecu_lifetime_energy" is visible when i select :
Settings -> Dashboard -> Energy ->Add Solar production
But when i select this i get the message : 
statistics_not_defined
sensor.ecu_lifetime_energy

HAEdwin, 2022-09-15:
Does it need fixing? Go to Developer Tool > STATISTICS and search for sensor.ecu_lifetime_energy

arduinopjotr, 2022-09-15:
It works ! Even with the warning that the statistics is not defined it works. As a mather of fact, why do i need to add sensor.ecu_lifetime_energy instead of sensor.ecu_today_energy ?

synack666, 2022-09-16:
I use ECU Today Energy, works too.

HAEdwin, 2022-09-16:
You are right, both show the same increase per day and therefore the production per day. It indeed takes some patience to see the energy dashboard working.

closed 2022-09-16 by HAEdwin