ksheumaker/homeassistant-apsystems_ecur #108: After updating Hassio OS to 9 it stop working after some minutes

synack666, 2022-09-15:
First ECU caching ist activated, then ECU is lost, restart HA and the same, try several times. With OS before it works a lot of month.

synack666, 2022-09-15:
ups, ECU-B and YC600

HAEdwin, 2022-09-15:
Have you allready powercycled the ECU (turn off and on)?

synack666, 2022-09-15:
Hi, not, have done ist just now.
HA Log is not much:
 Logger: custom_components.apsystems_ecur
Source: helpers/update_coordinator.py:151
Integration: APSystems PV solar ECU (documentation)
First occurred: 17:22:19 (1 occurrences)
Last logged: 17:22:19
Error fetching apsystems_ecur data: Error using cached data more than 5 times. See log, and try power cycling the ECU.

Version 1.2.16

synack666, 2022-09-15:
Just went back to 8.5, some minutes after powercycle it happens again, will see what is with 8.5 again to make sure that depends to HA and not something other.

synack666, 2022-09-16:
With 8.5 no more problems, everything was fine so i could say OS 9 was the problem.

HAEdwin, 2022-09-16:
No problem here. (ECU-R, QS1 + YC600)
[image]
Fails the integration immediately after the home-assistant reboot? Differences are IP6 support and socket receive and send buffer size which I could very possibly relate to the problem.

synack666, 2022-09-16:
No, it tooks some minutes, first caching activatet and some minutes later all sensors lost, ECU offline. App on handy works at same time.

HAEdwin, 2022-09-16:
What about changing the value to 1024 on line 34 of APSystemsSocket.py (self.recv_size = 1024) ? Are you using a Raspberry Pi (which model)?

synack666, 2022-09-17:
Will test this next week, i use a Intel Nuc.

HAEdwin, 2022-09-18:
See if others with an intel NUC also have problems with the new OS 9 release. There are few to no options in the integration to fix this unless you come up with ideas. Troubleshooting takes a lot of time if you don't have the hardware yourself unfortunately.

synack666, 2022-09-19:
Morning :-)

Test results:

It works since 2h
First i changed the value in 8.5 -> works
then upgrade to 9.0, reboot
No caching in log and no faults!

Thanks a lot, this integration is much important for me.
Problems with NUC and 9.0 i found are mostly with USB sticks and driver things.

I close here and wish a pleasant week

closed 2022-09-19 by synack666