ksheumaker/homeassistant-apsystems_ecur #109: This entity is no longer being provided by the apsystems_ecur integration?

bloemkool001, 2022-09-15:
As per a few days I get the following message in my HA:

"This entity is no longer being provided by the apsystems_ecur integration. If the entity is no longer in use, delete it in settings."

Trying to re-configure the IP address results in an error. The panels ARE working as the AP Systems app shows all the data.

- Tried to update HA, but still no results. In the Overview (Dashboard)
- In Configurations and Settings the status remains on: Initializing
- Tried to delete the config and re-install the integration, but this also results in an error (and remains non-responsive)

Is the entire integration down?

HAEdwin, 2022-09-16:
Have you allready tried to powercycle the ECU?

bloemkool001, 2022-09-16:
@HAEdwin thank you for this one! I removed the integration completely, rebooted HA. Reinstalled the integration, rebooted the ECU again, and now...finally...it's there! Great work!!! Thx

closed 2022-09-16 by bloemkool001