ksheumaker/homeassistant-apsystems_ecur #110: Task exception was never retrieved

ruudkoeyvoets, 2022-09-19:
Hello 
I get these error between the retrievals. I am on current latest build v1.2.17.
below log file
Regards
Ruud

2022-09-19 12:36:45.008 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Querying ECU
2022-09-19 12:37:00.664 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Got data from ECU
2022-09-19 12:37:00.679 DEBUG (SyncWorker_0) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [158, 160], 'voltage': [229, 229], 'inverters': {'[704000037390](tel:704000037390)': {'uid': '[704000037390](tel:704000037390)', 'online': True, 'signal': 84, 'frequency': 50.0, 'temperature': 36, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [159, 159], 'voltage': [227, 227]}, '[704000047117](tel:704000047117)': {'uid': '[704000047117](tel:704000047117)', 'online': True, 'signal': 89, 'frequency': 50.0, 'temperature': 35, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [168, 167], 'voltage': [228, 228]}, '[704000039603](tel:704000039603)': {'uid': '[704000039603](tel:704000039603)', 'online': True, 'signal': 89, 'frequency': 50.0, 'temperature': 35, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [165, 165], 'voltage': [228, 228]}, '[704000046411](tel:704000046411)': {'uid': '[704000046411](tel:704000046411)', 'online': True, 'signal': 80, 'frequency': 50.0, 'temperature': 35, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [158, 160], 'voltage': [229, 229]}, '[704000038131](tel:704000038131)': {'uid': '[704000038131](tel:704000038131)', 'online': False, 'signal': 0}, '[704000045962](tel:704000045962)': {'uid': '[704000045962](tel:704000045962)', 'online': False, 'signal': 0}, '704000045916': {'uid': '704000045916', 'online': False, 'signal': 0}, '704000046425': {'uid': '704000046425', 'online': False, 'signal': 0}}, 'ecu_id': '[216200069349](tel:216200069349)', 'today_energy': 2.41, 'lifetime_energy': 81.9, 'current_power': 1301, 'qty_of_inverters': 8, 'qty_of_online_inverters': 4, 'data_from_cache': False, 'querying': True}
2022-09-19 12:37:00.682 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.680 seconds (success: True)
**2022-09-19 12:37:00.716 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved**
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 151, in _handle_refresh_interval
await self._async_refresh(log_failures=True, scheduled=True)
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
self.async_update_listeners()
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 110, in async_update_listeners
update_callback()
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 348, in _handle_coordinator_update
self.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 533, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 571, in _async_write_ha_state
state = self._stringify_state(available)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in _stringify_state
if (state := self.state) is None:
File "/config/custom_components/apsystems_ecur/sensor.py", line 133, in state
return self.coordinator.data.get("inverters", {}).get(self._uid, {}).get("voltage", [])[0]
IndexError: list index out of range

HAEdwin, 2022-09-19:
Because this is the beta version, do you also have this issue with the 1.2.16 version? You indicate; between queries,.. so it doesn't happen continuously? Is the ECU configuration correct?

ruudkoeyvoets, 2022-09-19:
Yes same issue with previous release

HAEdwin, 2022-09-19:
What strikes me is that only 4 of the 8 DS3 inverters are online and that in the middle of the day is a bit strange. 
How many inverters do you have, or how many panels? Is there a reason they are offline?
704000038131
704000045962
704000045916
704000046425

ruudkoeyvoets, 2022-09-19:
I have 8 inverters of which 4 are installed :-) Remainder will be installed soon. (They are still in my basement)
I will test your suggestion and answer asap

ruudkoeyvoets, 2022-09-19:
still same issue, but now I see 2 different error before. They might have been there before as well.
ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform apsystems_ecur
ERROR (MainThread) [homeassistant.components.sensor] Error while setting up apsystems_ecur platform for sensor

HAEdwin, 2022-09-19:
I think the fasted way to get rid of these errors is to remove the not yet installed inverters. Make sure your can scan them later to add them to the ECU again. I assume the "invalid" ECU configuration is causing this error. Better restore that line 133 back also. I might take a look at what's happening if you want to post the data by uncommenting line 215, 238 and 256 in APSystemsSocket.py

ruudkoeyvoets, 2022-09-19:
Thanks for the support:

2022-09-19 22:35:28.533 WARNING (SyncWorker_1) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533132303131373030303132313632303030363933343930310000037d00000000000003d6202209192026070008000031303031364543555f525f50524f5f322e302e37413031364575726f70652f416d7374657264616d80971b0415f160c5a87f24d9303000000000000000000000454e440a'
2022-09-19 22:35:39.576 WARNING (SyncWorker_1) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533131303134393030303230303031000820220919222805704000037390003031000000000000000000000000704000047117003031000000000000000000000000704000039603003031000000000000000000000000704000046411003031000000000000000000000000704000038131003030704000045962003030704000045916003030704000046425003030454e440a'
2022-09-19 22:35:39.582 WARNING (SyncWorker_1) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303734303033303030704000037390d8704000047117e4704000039603e2704000046411ca70400003813100704000045962007040000459160070400004642500454e440a'

————————————

Attached the log after un_commenting the proposed lines

HAEdwin, 2022-09-19:
Thanks - the data is correct right now (sunset so the values are zero). If you want please post the data when the four inverters are online so that I can check how the data is interpreted.

ruudkoeyvoets, 2022-09-19:
I think you are right the not online inverters cause the problem.
The number of channels you read on a not active inverter is none. Shouldn’t this be 0

2022-09-19 23:26:26.576 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000037390 2]
2022-09-19 23:26:26.625 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000047117 2]
2022-09-19 23:26:26.640 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000039603 2]
2022-09-19 23:26:26.649 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000046411 2]
2022-09-19 23:26:26.684 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000038131 None]
2022-09-19 23:26:26.688 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000045962 None]
2022-09-19 23:26:26.693 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter 704000045916 None]
2022-09-19 23:26:26.699 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter 704000046425 None]

ruudkoeyvoets, 2022-09-20:
Below log file info during day time.

last part is from the debug info of Sensor line 65.

2022-09-20 14:04:40.013 WARNING (SyncWorker_2) [custom_components.apsystems_ecur.APSystemsSocket] b'4150533132303131373030303132313632303030363933343930310000039e000000e30000014e202209201401530008000431303031364543555f525f50524f5f322e302e37413031364575726f70652f416d7374657264616d80971b0415f160c5a87f24d9303000000000000000000000454e440a'
2022-09-20 14:04:51.672 WARNING (SyncWorker_2) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313439303030323030303100082022092014011570400003739001303101f40084001c00de001b00de70400004711701303101f40085001e00df001c00df70400003960301303101f40085001e00e0001d00e070400004641101303101f40085001b00de001c00de704000038131003030704000045962003030704000045916003030704000046425003030454e440a'
2022-09-20 14:04:51.679 WARNING (SyncWorker_2) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130303734303033303030704000037390d8704000047117e3704000039603e1704000046411cc70400003813100704000045962007040000459160070400004642500454e440a'
2022-09-20 14:04:51.684 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Got data from ECU
2022-09-20 14:04:51.685 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [27, 28], 'voltage': [222, 222], 'inverters': {'[704000037390](tel:704000037390)': {'uid': '[704000037390](tel:704000037390)', 'online': True, 'signal': 84, 'frequency': 50.0, 'temperature': 32, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [28, 27], 'voltage': [222, 222]}, '[704000047117](tel:704000047117)': {'uid': '[704000047117](tel:704000047117)', 'online': True, 'signal': 89, 'frequency': 50.0, 'temperature': 33, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [30, 28], 'voltage': [223, 223]}, '[704000039603](tel:704000039603)': {'uid': '[704000039603](tel:704000039603)', 'online': True, 'signal': 88, 'frequency': 50.0, 'temperature': 33, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [30, 29], 'voltage': [224, 224]}, '[704000046411](tel:704000046411)': {'uid': '[704000046411](tel:704000046411)', 'online': True, 'signal': 80, 'frequency': 50.0, 'temperature': 33, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [27, 28], 'voltage': [222, 222]}, '[704000038131](tel:704000038131)': {'uid': '[704000038131](tel:704000038131)', 'online': False, 'signal': 0}, '[704000045962](tel:704000045962)': {'uid': '[704000045962](tel:704000045962)', 'online': False, 'signal': 0}, '704000045916': {'uid': '704000045916', 'online': False, 'signal': 0}, '704000046425': {'uid': '704000046425', 'online': False, 'signal': 0}}, 'ecu_id': '[216200069349](tel:216200069349)', 'today_energy': 3.34, 'lifetime_energy': 92.6, 'current_power': 227, 'qty_of_inverters': 8, 'qty_of_online_inverters': 4, 'data_from_cache': False, 'querying': True}
2022-09-20 14:04:52.149 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 17.538 seconds (success: True)
2022-09-20 14:04:54.765 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000037390 2]
2022-09-20 14:04:54.783 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000047117 2]
2022-09-20 14:04:54.793 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000039603 2]
2022-09-20 14:04:54.799 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000046411 2]
2022-09-20 14:04:54.804 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000038131] None
2022-09-20 14:04:54.812 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter [704000045962] None
2022-09-20 14:04:54.818 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter 704000045916 None
2022-09-20 14:04:54.828 WARNING (MainThread) [custom_components.apsystems_ecur.sensor] Inverter 704000046425 None

HAEdwin, 2022-09-20:
Interpretation of the data is done correct but of course the status is also updated of the sensors that do not yet exist.
I'm having a hard time (and lack of that) reproducing it all but maybe it will help to disable those inverter entities until panels are attached to them?  Hope HA will pick that up. 

I'll add a note to the prerequisites that a valid ECU configuration is necessary for the integration to work ;) it would take me a lot of time to anticipate incomplete or invalid ECU configurations, and difficult to reproduce for all models and options.

HAEdwin, 2022-09-21:
It's a bit of an uncontrolled test after sunset but does this help? Line 66 was added. Pay close attention to the indent!
```
    inverters = coordinator.data.get("inverters", {})
    for uid,inv_data in inverters.items():
        _LOGGER.warning(f"Inverter {uid} {inv_data.get('channel_qty')}")
        if inv_data.get('channel_qty') != None:
            sensors.extend([
                    APSystemsECUInverterSensor(coordinator, ecu, uid, 
                        "temperature", label="Temperature", 
                        devclass=DEVICE_CLASS_TEMPERATURE, unit=TEMP_CELSIUS, 
                        entity_category=EntityCategory.DIAGNOSTIC),
                    APSystemsECUInverterSensor(coordinator, ecu, uid, 
                        "frequency", label="Frequency", unit=FREQUENCY_HERTZ, 
                        devclass=None, icon=FREQ_ICON, entity_category=EntityCategory.DIAGNOSTIC),
                    APSystemsECUInverterSensor(coordinator, ecu, uid, 
                        "voltage", label="Voltage", unit=ELECTRIC_POTENTIAL_VOLT, 
                        devclass=DEVICE_CLASS_VOLTAGE, entity_category=EntityCategory.DIAGNOSTIC),
                    APSystemsECUInverterSensor(coordinator, ecu, uid, 
                        "signal", label="Signal", unit=PERCENTAGE, 
                        icon=SIGNAL_ICON, entity_category=EntityCategory.DIAGNOSTIC)

            ])
            for i in range(0, inv_data.get("channel_qty", 0)):
                sensors.append(
                    APSystemsECUInverterSensor(coordinator, ecu, uid, f"power", 
                        index=i, label=f"Power Ch {i+1}", unit=POWER_WATT, 
                        devclass=DEVICE_CLASS_POWER, icon=SOLAR_ICON)
                )
    add_entities(sensors)
```

ruudkoeyvoets, 2022-09-21:
Thanks, this solved the log file entries!!!

Thanks for the great support

HAEdwin, 2022-09-22:
Great, It was not that bad in the end, keep me informed when you connect the inverters - the channels should then be automatically discovered and the other parameters filled. I'll add this to the new release and change the topic name for better reference in the future.

closed 2022-09-22 by HAEdwin