ksheumaker/homeassistant-apsystems_ecur #113: HACS integration not able to connect to ECU-B

TimWinkler, 2022-09-30:
This error originated from a custom integration.

Logger: custom_components.apsystems_ecur.config_flow
Source: custom_components/apsystems_ecur/APSystemsSocket.py:116
Integration: APSystems PV solar ECU (documentation, issues)
First occurred: September 29, 2022 at 9:10:32 PM (5 occurrences)
Last logged: 8:35:14 AM

APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS110094000121630003664601\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x01\x00\x0010012ECU_B_1.2.23009Etc/GMT-8\x80\x97\x1b\x04`\xbf\x00\x00\x00\x00\x00\x00END\n'
APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS110094000121630003664601\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x01\x00\x0110012ECU_B_1.2.23009Etc/GMT-8\x80\x97\x1b\x04`\xbf\x00\x00\x00\x00\x00\x00END\n'
APSystemsInvalidData exception: ECU returned 0 for lifetime energy, raw data=b'APS110094000121630003664601\x00\x00\x00\x00\x00\x00\x00\x0c\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x01\x00\x0110012ECU_B_1.2.23009Etc/GMT-8\x80\x97\x1b\x04`\xbf\x00\x00\x00\x00\x00\x00END\n'
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 46, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 116, in query_ecu
    raise APSystemsInvalidData(error)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: ECU returned 0 for lifetime energy, raw data=b'APS110094000121630003664601\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd0\xd0\xd0\xd0\xd0\xd0\xd0\x00\x01\x00\x0010012ECU_B_1.2.23009Etc/GMT-8\x80\x97\x1b\x04`\xbf\x00\x00\x00\x00\x00\x00END\n'

TimWinkler, 2022-09-30:
answer taken from packet sender:

41 50 53 31 31 30 30 39 34 30 30 30 31 32 31 36 33 30 30 30 33 36 36 34 36 30 31 00 00 00 00 00 00 00 2E 00 00 00 06 D0 D0 D0 D0 D0 D0 D0 00 01 00 01 31 30 30 31 32 45 43 55 5F 42 5F 31 2E 32 2E 32 33 30 30 39 45 74 63 2F 47 4D 54 2D 38 80 97 1B 04 60 BF 00 00 00 00 00 00 45 4E 44 0A

HAEdwin, 2022-09-30:
Hi Tim, a brand new ECU-B... there's a few lines of code you could comment out in APSystemsSocket.py by placing a # in front of it. Don't forget to restart! Hope you are able to do that. 
It applies to line 113, 114, 115, 116 or... you can wait until lifetime energy is not zero anymore. 

I will patch these lines in the next release. Sorry for the inconvenience. Sample how it should look like:
[image]

TimWinkler, 2022-09-30:
works perfect - thank you very much!

closed 2022-09-30 by HAEdwin