ksheumaker/homeassistant-apsystems_ecur #121: Integration stopped working

didigrand, 2022-10-08:
Hello,
After few days of normal behavior, the integration stop working...

The last message in the log (debug) is : 
2022-10-08 04:27:48.001 DEBUG (SyncWorker_5) [custom_components.apsystems_ecur] Querying ECU

And the previous update cycle was:
2022-10-08 04:22:33.001 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Querying ECU
2022-10-08 04:22:48.016 WARNING (SyncWorker_2) [custom_components.apsystems_ecur] Using cached data from last successful communication from ECU. Exception error: 'NoneType' object does not support item assignment
2022-10-08 04:22:48.022 DEBUG (SyncWorker_2) [custom_components.apsystems_ecur] Returning {'model': 'YC600/DS3', 'channel_qty': 2, 'power': [0, 0], 'voltage': [0, 0], 'inverters': {'xxx': {'uid': 'xxx', 'online': False, 'signal': 69, 'frequency': 0.0, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [0, 0], 'voltage': [0, 0]}, 'xxx': {'uid': 'xxx', 'online': False, 'signal': 69, 'frequency': 0.0, 'model': 'YC600/DS3', 'channel_qty': 2, 'power': [0, 0], 'voltage': [0, 0]}}, 'ecu_id': 'xxx', 'today_energy': 0.0, 'lifetime_energy': 63.4, 'current_power': 0, 'qty_of_inverters': 2, 'qty_of_online_inverters': 0, 'data_from_cache': True, 'querying': True}
2022-10-08 04:22:48.024 DEBUG (MainThread) [custom_components.apsystems_ecur] Finished fetching apsystems_ecur data in 15.023 seconds (success: True)

I tried to trigger the ECU reboot automation but without success.
Then I restarted HomeAssistant and the integrated started to work again.

ViperRNMC, 2022-10-08:
Same as https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/115

HAEdwin, 2022-10-18:
@didigrand I'll close this issue and continue on https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/115
I think I'm close to an answer with this one. Please install v1.2.22 where it might be solved.

closed 2022-10-18 by HAEdwin