ksheumaker/homeassistant-apsystems_ecur #123: Unable to add ECU-R PRO

dewi-ny-je, 2022-10-12:
I am using @HAEdwin fork and I posted the issue there, but at some point issues were disabled without warning there (I just noticed) and my bug report/request of assistance disappeared so I am writing it again here, I hope I haven't forgotten any detail.

I have an ECU-R PRO (2162xxxx) which worked fine until beginning of September.
Then it couldn't access the ECU anymore. I tried to remove the ECU and add it again, nothing. This happened not at the same time, but also not so long after the ECU got updated to firmware 2.0.7.

I removed the custom component, rebooted, reinstalled the custom component, rebooted, I can see the APSystems in the list of available components, but when I try to add it, the process stops after I enter the IP address (tried both wired and wireless, I have both). It says "timeout, cannot connect" (or similar) and there's no way to get data from it.

Needless to say, accessing the ECU via browser is just fine.

I cannot find any logs in the HA logs.

Please let me know what and how I can provide more data so we can diagnose the issue and, if needed, correct the bug/adapt the component to the new firmware.

HAEdwin, 2022-10-13:
@dewi-ny-je Without prior notice is of course not neat. My apologies. I wanted to keep the communication in one place, my repository is not always reliable to use. You wrote: _This happened not at the same time, but also not so long after the ECU got updated to firmware 2.0.7_ From what I remind in earlier discussions is that the ECU firmware might have changed the MAC address, if you have an IP address bound to a MAC address, that could be the cause. Also an expired DHCP lease could cause the issue because it happened somewhat later. 

To rule out HA or the integration, I would advice to use Packet Sender to try to connect to the ECU.
[image]
If you are able to connect using Packet Sender you could send me the hex results marked yellow. In some cases I noticed a invalid ECU configuration on which the integration did not anticipate.

dewi-ny-je, 2022-10-13:
The MAC hasn't changed in my case, the ECU still gets the same assigned IPs: .227 for wired and .228 for wifi.
[image]
[image]
but wait!
I just restarted it (cut the power physically) and I get a reply!
It is the second row from the BOTTOM:
[image]

I paste here the reply (I can't see any private data in it, beside that I'm in the NL):
APS120117000121620004722301\00\00|I\00\00\00\00\00\00\00\04 "\10\13\t4T\00\06\00\0010016ECU_R_PRO_2.0.7B016Europe/Amsterdam\80\97\1b\03\88P`\c5\a8~\f8\1400\00\00\00\00\00\00\00\00\00\00END\n

Important! while I was doing that, the web interface still showed NO data from the inverters, no read had yet taken place.
I waited a little bit to get data in the web interface, then I run again the packet sending (next 4 lines in the screenshot).
No more replies!

Now I tried again and it works again (TOP 4 lines, check the timestamps). I wonder.
After performing more tests, I can see that replies are sometimes missing.
Still, I got it in HA!

I have an automation which physically cuts the power to the ECU for 10 seconds every evening, 1 hour after sunset. If the issue repeats itself I will check whether it actually runs. It should, it's very simple.

I close the ticket for now.

closed 2022-10-13 by dewi-ny-je