ksheumaker/homeassistant-apsystems_ecur #125: Re-Configuration issue

HAEdwin, 2022-10-18:
The ip is wrong, changed again on HA:
[image]

still the same wrong ip on .log

tried including ip's that don't exists on HA, got errors:
[image]

for some reason its trying to get data from an ip that is not possible to include on config, while ignoring the ip i wrote on config.

_Originally posted by @doug-last in https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/115#issuecomment-1281698493_

doug-last, 2022-10-18:
on file ".storage/core.config_entries" i noted that:

`
    {
        "entry_id": "d13f5e064b06cc06409c682383443a45",
        "version": 1,
        "domain": "apsystems_ecur",
        "title": "ECU: 216200056339",
        "data": {
          "host": "192.168.0.52",
          "scan_interval": 300
        },
        "options": {
          "host": "192.168.0.44",
          "scan_interval": 321
        },

`

the one with ip ended in 44 scan_interval 321 is my input, the one ended in 52 with scan 300 is the one that is being read and should have being replaced.

doug-last, 2022-10-18:
> on file ".storage/core.config_entries" i noted that:
>
> ` { "entry_id": "d13f5e064b06cc06409c682383443a45", "version": 1, "domain": "apsystems_ecur", "title": "ECU: 216200056339", "data": { "host": [...]

on that part of __init__.py:
```
async def async_setup_entry(hass, config):
    """ Setup the APsystems platform """
    
    _LOGGER.debug(f"config={config.data}")

    hass.data.setdefault(DOMAIN, {})

    host = config.data[CONF_HOST]
    interval = timedelta(seconds=config.data[CONF_SCAN_INTERVAL])
```

lines 174 and 175 replaced to:

```
    host = config.options[CONF_HOST]
    interval = timedelta(seconds=config.options[CONF_SCAN_INTERVAL])
```

i think this got it ... testing

doug-last, 2022-10-18:
> on that part of **init**.py:
>
> ```
> async def async_setup_entry(hass, config):
>     """ Setup the APsystems platform """
>
>     _LOGGER.debug(f"config={config.data}") [...]

reinstalled to v1.2.22, rebooted, got the bug back, changed the 2 lines above again, rebooted, it's working.

probably is not the proper fix tho, i just tried to make it get the data from "options" instead of "data", should not have both.

doug-last, 2022-10-18:
i noted also that the ip of my ECU changes on power outages.
having the feature of scanning the local network for the ECU can be nice, easier for the user. (i'm new with all this, but HA detected upnp and suggested the integration automatically, i don't know if this behavior is ideal for all integrations with devices)

HAEdwin, 2022-10-18:
You are so right finding the issue, there are two active schemas: config.options and config.data. The one to use is config.data but settings are stored in config.options when you reconfigure the integration not in config.data. I'll have to run this through carefully. Thank you so much for finding this! When I have some time I'll fix it.

If your ECU IP changes it might be in the DHCP pool. In your router or node that runs the DHCP server you should be able to bind a fixed IP-Address to the MAC of your ECU so it gets the same IP-Address all the time. Common practice is to reserve part of the subnet for nodes that need a fixed IP-Address.

dkwolf1, 2022-10-21:
Same problem. Just had to add the host and scan interval under options in file .storage/core.config_entries. It is working again

HAEdwin, 2022-11-04:
This issue is fixed in pre-release 1.2.24

closed 2022-11-16 by HAEdwin