ksheumaker/homeassistant-apsystems_ecur #126: ECU-R shows entire lifetime energy consumed during one hour in the energy

arduinopjotr, 2022-10-19:
I have the same issue her as mackowskim since a couple of days. I did an upgrade of the ecu-r plugin. After the upgrade it looks like the ECU plugin sometimes write a zero in the entity "sensor.ecu_lifetime_energy" . If the next data is correct again (e.g. 400kWh) Then the statistics will giva a value of 400 because of the delta from 400-0 = 400.
Is it possible that the ECU-R plugin writes a zero when there is no connection to the gateway ? Is it a plan to block the updates of the entity "sensor.ecu_lifetime_energy" if the value is zero.
PS The entity sensor.ecu_today_energy" does nog have any gliches.

[image]

HAEdwin, 2022-10-20:
In version 1.2.19 I removed the check because out-of-the-box ECU's do have zero lifetime energy. I will reintroduce the check and figure out a way around that for freshly connected ECU's. The zero values are under some circumstances received from the ECU they are not produced by the intergration.

arduinopjotr, 2022-10-20:
Thanks for your reply and your hard work !!

HAEdwin, 2022-10-30:
Released v1.2.23 where I re-introduced the check on zero values for life-time energy.

closed 2022-10-30 by HAEdwin