ksheumaker/homeassistant-apsystems_ecur #131: Is this integration working for you guys?

andrebbruno, 2022-11-04:
[image]

This was working fine since last update, now it stopped working. Tried to reload, power cycle, disable and enable. Nothing goes.

I´m afraid of removing the integration and re-adding and lose the integration data from the past.

thanks!

HAEdwin, 2022-11-04:
Is this the pre-release v1.2.24? Does the homeassistant.log give any information?

andrebbruno, 2022-11-04:
> Is this the pre-release v1.2.24? Does the homeassistant.log give any information? Removing the integration and reinstalling an earlier version does not cause data loss.

I´m very sorry, should´ve included this information. No I´m using 1.2.23 - logs didn´t say much besides the error message.

HAEdwin, 2022-11-04:
So you migrated from 1.2.22 to 1.2.23? You can easily revert back to an older version by going to HACS and in the overflow menu on the right choose [Redownload]. After a restart you are up-n-running again. I have had this message once earlier, HA will retry and might recover from it.

andrebbruno, 2022-11-05:
Tried to revert back, didn´t work ... I was afraid of losing historic data for solar production, so in the end performed a full backup, removed the integration completely, restarted etc. Reinstalled it and it worked. Sorry for the alarm.

HAEdwin, 2022-11-05:
No problem, as I get to know HA better I see how solid the architecture is. HA sometimes just needs some TLC, patience and sometimes you have to think carefully about what you're doing. I have never lost data, at most daily totals are reset. A Redownload via HACS only takes care of replacing the excisting integration files. These only become active after a restart. I will close this issue. Might another issue arise you are always welcome to open a new one.

closed 2022-11-05 by HAEdwin