ksheumaker/homeassistant-apsystems_ecur #132: several sensors are still missing

MP3Chris2712, 2022-11-08:
Would it be possible to add the sensor total energy yesterday in the ECU and in the inverter there is still total dc power. Since I'm just switching from Iobroker to Homeassistant, I noticed that there are more sensors or objects there.

HAEdwin, 2022-11-08:
Tell me more about your hardware

MP3Chris2712, 2022-11-08:
ECU-B with a DS3-S could show you the objects from IoBroker what is possible. Greetings from Northern Germany

HAEdwin, 2022-11-13:
The integration is written to be as generic as possible, without owning an ECU-B it is hard to test what has been written/added to facilitate this option. What component did you use in IoBroker and indeed what are the objects?

MP3Chris2712, 2022-11-14:
hey,

I could send you an excerpt of all objects that are available in IoBroker. I use the following adapters https://github.com/npeter/ioBroker.apsystems-ecu

[image]
[image]

HAEdwin, 2023-01-29:
Some of the additional values are being calculated in the ecu.js file used in ioBroker. I could add them to the integration but for example the total power of every inverter could be calculated using a "helper" in HA. I currently use it this way.

To calculate the total power from an inverter from HA: Settings > Devices and Services > Helpers (top of the screen) > +Create Helper > +/- Combine the state of several sensors
Then name the new sensor you've created from the sum of each inverter channel's power.

To transfer the daily power to the total_energy_yesterday you could use something like what is being used here: https://community.home-assistant.io/t/statistic-sensor-reset-clear-at-midnight-for-daily-min-max-temperature/501688/8

HAEdwin, 2023-11-12:
I close this issue. I will keep Total Energy for each week, month and year in mind as it brings a number of benefits. If a question arises, please feel free to open a new issue.

closed 2023-11-12 by HAEdwin