ksheumaker/homeassistant-apsystems_ecur #135: Support for DS3D-L inverter

dclobato, 2022-11-14:
Today, I added 4 more panels to my setup. Now, I have 4 micro inverters generating 8.64kWp

[image]

I had 3 QS1 microinverters, and they have all details on the integration

[image]

The new inverter has no details at all...

[image]

The new inverter is a [**DS3D-L**](https://latam.apsystems.com/wp-content/uploads/2021/11/4303601202_APsystems-Microinverter-DS3D-For-Brazil-Datasheet_-Rev1.0_2021-11-19.pdf).

I saw, on the code, that there is support for DS3 inverters. Is it possible to add support for the DS3D model?  What information should I collect/provide to allow this new model be added to the integration?

HAEdwin, 2022-11-14:
Have you tried uninstalling the integration (Settings>Devices & Services>Integrations and then using the menu to delete the integration) restart HA and add the integration again. The entities might than get added/discovered.

dclobato, 2022-11-14:
> Have you tried uninstalling the integration (Settings>Devices & Services>Integrations and then using the menu to delete the integration) restart HA and add the integration again.

I understand that the integration detected the new entity.

I believe that the problem lies on the fact that the DS3D-L inverter has a different code. From the [code](https://github.com/ksheumaker/homeassistant-apsystems_ecur/blob/acbfe25b9329f92e1d3d98cfc13545cb1db23cc7/custom_components/apsystems_ecur/APSystemsSocket.py#L324), each inverter has a different code: 01 for YC600/DS3, 02 for YC1000 and 03 for QS1. Maybe the DS3D-L has a different code on the response string...

HAEdwin, 2022-11-15:
Yes, that's a good starting point if entities from the inverter are not discovered. You can uncomment line 262 by removing the first # . In /config/home-assistant.log you will find the results from the inverter query. Can you share that string for further analysis?

dclobato, 2022-11-15:
Here it is

```
2022-11-15 17:33:17.001 WARNING (SyncWorker_1) [custom_components.apsystems_ecur.APSystemsSocket] b'415053313130313139303030323030303100042022111517284180400004919001303302580089001800d90025002c00448040000512700130330258008d006900d9006e001d000a8040000513600130330259008d000e00da000c000d000b7060000305790130340258009500e400d7009700d7454e440a'
```

HAEdwin, 2022-11-16:
Daniel, I've made to changes to APsystemsSocket.py to add compatibility with this new DS3D-L inverter. Before a new release could you check if this works out fine for you? Follow link for changes.
https://github.com/ksheumaker/homeassistant-apsystems_ecur/commit/bcd62f98a08fdbbc5a9d3ed1deb11fab37a035c2

dclobato, 2022-11-16:
> Before a new release could you check if this works out fine for you?

I believed it's working now!

[image]

[image]

Can I close the issue?

Thanks!

HAEdwin, 2022-11-16:
Yes you can, glad it worked out fine. Thanks for supplying info on this relatively new inverter so that I could add it. It will be included in the new release.

closed 2022-11-16 by HAEdwin