ksheumaker/homeassistant-apsystems_ecur #136: Add options to secure ECU AP

HAEdwin, 2022-11-16:
The automated restart , resets the AP with a default name and no security. After this the AP is available for 1 hour. Maybe better to set a WPA password. Is it possible to make a config in the integration to set an own SSID, Channel, Method and Password?

data for requests.post would be:
`data= {'ssid': 'ECU-WIFI_local', 'channel'; 0, 'method': 2, 'psk_wep': 'pwd', 'psk_wpa': 'pwd'})`

ssid = ssid
channel = 0-13 (0=auto)
method = 0,1,2 (no/wep/wpa)
psk_wep = password wep
psk_wpa = password wpa

_Originally posted by @ViperRNMC in https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/115#issuecomment-1316797653_

HAEdwin, 2022-12-11:
@ViperRNMC  Please help me out with the correct data string. Would this represent it correctly?
```
SSID = "MyNetworkSSID"
Password = "MyPassword123"
data = "{'SSID': '" + SSID + "', 'channel': 0, 'method': 2, 'psk_wep': '', 'psk_wpa': '" + Password + "'}"

```
Output result would be:
`{'SSID': 'MyNetworkSSID', 'channel': 0, 'method': 2, 'psk_wep': '', 'psk_wpa': 'MyPassword123'}`

But note that 0 and 2 are not within single-parentheses like you specified above, is that correct? Is it allright to default to method 2 (WPA) because WEP is considered not secure.

ViperRNMC, 2022-12-11:
Channel and method can be send as number/int, without the single quotes. I also think it would be better to only use wpa.

HAEdwin, 2022-12-12:
Great thanks, I'll have to do some testing now on this and other features now.

HAEdwin, 2022-12-16:
If you want, you can test the changes from my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur I'm done testing but can't really test __init__.py line 67 specific on how the ECU reacts.

ViperRNMC, 2022-12-18:
> If you want, you can test the changes from my repository https://github.com/HAEdwin/homeassistant-apsystems_ecur I'm done testing but can't really test **init**.py line 67 specific on how the ECU reacts.

I’ll check it tomorrow, what files are changed over the last release?

HAEdwin, 2022-12-18:
All files have changed except: binairy_sensor.py, diagnostics.py, switch.py

- For added flow options:
nl.json 
en.json
- Easier way to test connection:
readme.md
- Removed warning messages where debug applies:
APsystemsSocket.py
- Added additional options ssid, wpa, retry count and added flow update listener:
__init__.py
- Added fix to apply interval change immediately and added ssid, wpa, retry count to options flow:
config_flow.py
 

In detail:
https://github.com/ksheumaker/homeassistant-apsystems_ecur/compare/main...HAEdwin:homeassistant-apsystems_ecur:main

ViperRNMC, 2022-12-19:
Using it now, look fine I think. When I unplug the rj45. It looks like it would reset the ECU. Must only check if the soft reset is really working with SSID info. Let's wait a few days.

ViperRNMC, 2023-01-03:
add a fix https://github.com/HAEdwin/homeassistant-apsystems_ecur/commit/3ff967f9f1fafab4d79c4b5b78df93dfa1e62a96

HAEdwin, 2023-01-04:
Released v1.2.26-beta to solve this issue. I'll close the issue on production release later on.

ViperRNMC, 2023-01-04:
I saw the new beta release, thanks!

HAEdwin, 2023-01-23:
Just published the latest v1.2.26 I will close this issue. If a problem occurs, please open a new issue.

closed 2023-01-23 by HAEdwin