ksheumaker/homeassistant-apsystems_ecur #139: ECU-R error "Can't find ECU at this IP-Address"

Salhorgne, 2022-11-28:
Hi,

In Hacs i installed the last 1.12.25 last beta version component of APS ECU-R. But i still can't add my ECU with the same error (listed in title)

My ECU-R-EU 's IP address (ECU is from august 2022) is responding from ha with ping commannd (i got DS3 cconverters btw, new install)

And i wanted to test an older version but i can't uninstall the last beta version of EcU-R component using hacs to see if it's resolve the problem...

Any ideas guys why ? PS:  no firewall security, ssh with putty works from my PC and from HA ...

Regards,

ViperRNMC, 2022-11-28:
Got the same message, power cycle the ecu and it could find the ecu again.

Salhorgne, 2022-11-28:
tried so many times, no way...

HAEdwin, 2022-11-28:
Ultimate test is to use PacketSender, send the string APS1100160001END to the correct IP-Address using port 8899 over TCP.
[image]

If you don't get any responses and you are sure you're on the right node (use ping while the ECU is off) there might be a serious compatibility issue. What firmware is running on the ECU? Did you use the EMA manager or the webinterface from the ECU?

Verguldebarman, 2022-11-29:
Same problem here. My ECU-R (2160xxxxx) is connected via LAN and the integration just cannot connect. Below is the error in the log (enabled debug hoping to see more, but this is it):

```
2022-11-29 12:38:36.951 DEBUG (MainThread) [custom_components.apsystems_ecur.config_flow] Starting config flow class...
2022-11-29 12:38:36.951 DEBUG (MainThread) [custom_components.apsystems_ecur.config_flow] Starting step_user
2022-11-29 12:38:36.951 DEBUG (MainThread) [custom_components.apsystems_ecur.config_flow] Returning to show form
2022-11-29 12:38:42.338 DEBUG (MainThread) [custom_components.apsystems_ecur.config_flow] Starting step_user
2022-11-29 12:38:42.338 DEBUG (MainThread) [custom_components.apsystems_ecur.config_flow] User Input is not none
2022-11-29 12:38:42.358 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: [Errno 111] Connection refused
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 101, in open_socket
    self.sock.connect((self.ipaddr, self.port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 46, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 107, in query_ecu
    self.open_socket()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 104, in open_socket
    raise APSystemsInvalidData(err)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: [Errno 111] Connection refused
```

Steps done:
1. checked if the IP address is correct, by checking my router. The MAC address connected with the IP address corresponds with a device manufactured by Altenergy Power System,Inc., as MAC addresses starts with 80-97-1B
2. reserved IP address on my router, so it stays fixed
3. rebooted device many times

p.s. I have a working solar setup with DS3 inverters. My panels produce energy and I am able to see produced energy in HA as well as in the APsystems app and dashboard.

Salhorgne, 2022-11-29:
> Ultimate test is to use PacketSender, send the string APS1100160001END to the correct IP-Address using port 8899 over TCP. [image]
> 
> If you don't get any responses and you are sure you're on the right node (use ping -t while the ECU is off/on) there might be a serious compatibility issue. What firmware is running on the ECU? Did you use the EMA manager or the webinterface from the ECU to set it up? Make sure to assign a fixed IP-Address to the ECU.

Thanks you Guy! 
Here is my packet sender result screenshot : 
[image]

HAEdwin, 2022-11-29:
@Verguldebarman Did you allready try the PacketSender method? How did this work out for you?

Verguldebarman, 2022-11-29:
No good... and didn't expect it to work as an earlier performed Portscan did not show port 8899 open:

[image]

HAEdwin, 2022-11-29:
@Verguldebarman Let's continue your issue at #141

HAEdwin, 2022-11-30:
@Salhorgne Can you look and post what's in your home-assistant.log? Is the lifetime energy on the ECU still zero?

Salhorgne, 2022-11-30:
> @Salhorgne Can you look and post what's in your home-assistant.log? Is the lifetime energy on the ECU still zero? Might be the same issue like [#113 (comment)](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/113#issuecomment-1263436330)

HI HAEdwin, by commenting the lines and restart HA. I succeeded in integrating my ECU-R :) Thks a lotttttttttttttttttt. issue resolved :)

HAEdwin, 2022-11-30:
Great to hear that, thank you for the feedback. When you notice that lifetime energy is not equal to zero you can uncomment these lines again because of issue https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/126
I really have to find a solution which suits all users fine :)