ksheumaker/homeassistant-apsystems_ecur #142: Strange behaviour when using 2 ECU in same network

syl2222, 2022-12-19:
Hello

I have strange behavious with this integration.

Everything work fine, but, time to time, i lost my graph and, when i check entity , i good the message:
This entity is no longer provided by the apsystems_ecur integration. If the entity is no longer used, delete it from the settings.

I remove the ecu and add it again, it's enough.
Then, later (1 minute, 1 hour, 12 hours, 1 day, depending) the second ECU got the same problem

I'm newbie with Homeassistant and unable to fully undestand everything. I attached few debug logs .
I have the impression that with 2 ecu and the absence of unique ID, the integration is lost?

Where can i search to find where is the problem . 
[home-assistant_apsystems_ecur_2022-12-19T16-17-04.314Z.log](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/10260928/home-assistant_apsystems_ecur_2022-12-19T16-17-04.314Z.log)

Joyeuses fêtes , merry christmas !

HAEdwin, 2022-12-19:
From your log I extracted that you own 1x ECU-B and 2x DS3-L. One of the DS3-L inverters has an somewhat strange ID 703000012345 was this ID really scanned from the inverters bar-code and is it valid? Make sure the ECU has a valid configuration first and the EMA site represents the correct hardware. Then it will also run fine in HA.

Are you using the latest 1.2.25 version?

For Home Assistant the best thing you can do is remove all (un-)related entities by uninstalling the integration:
Go to [Settings] > [Devices & Services] > Select the three dots above eachother (overflow menu) and choose [Delete].

The entities will now be removed (the software will remain in place, that's ok) but you will notice some entities will remain visible when you navigate to [Settings] > [Devices & Services] tab [Entities]. Select these entities and choose [Remove Selected].  
Restart HA

If all "APSystems PV solar ECU" associated entities are removed you can add the Integration again by going to [Settings] > [Devices & Services] > [+ Add Integration] and choose the "APSystems PV solar ECU integration". Enter the IP-Address of the ECU and wait for the integration to complete setup. Hopefully you will now only have valid entities that you can add to the dashboard. If not your ECU might be misconfigured.

syl2222, 2022-12-20:
Hello, 
Thank you for your answer. (English is not my natural language, so, sorry for errors in typo).

All my id (Ecu and ds3) were (and are) changed before i send the logs, so, don't worry about the strange numbers :)

I try, without problem, to remove all ecu and reinstall them. Everything works fine (data show up and graphs ok).
As soon as i restart HomeAssistant (i need it many times a day as i  keep migrating from domoticz to Ha) , i have one Ecu which stop working (not all the time the same one). 
In Ema website , everything still ok (graph)
From this git, i'm using v1.2.25 and, it seem than my Ecu(s) are in version 1.2.23

Attached , few logs at each step when i remove /restart HA/reinsert one Ecu.

when I encounter this problem and I try with my old bash script from domoticz, both ecu respond and give me data.
But, I do not know how to program in python and unable to debug :(

[logs.txt](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/10267330/logs.txt)

Maybe, with little luck, somebody should see something in this logs ..

Best regards
Sylvain

HAEdwin, 2022-12-21:
If you really own 2 ECU-B units, the integration has not anticipated on this because settings are stored under the same key. But still it is a bit confusing to see ECU ID's 216300067890 and 216300012345 but these are not the real numbers are they? I know the ECU-B can handle max 4 panels, are you planning to connect more than 4 panels I'd recommend using the ECU-R not ECU-B.

HAEdwin, 2022-12-21:
Unfortunately it's way above my head to adapt the integration to work with multiple ECU's (and thus multiple instances of the integration). Hopefully you find a solution by buying the ECU-R cheaper abroad for example.

syl2222, 2022-12-22:
Thank you for your investigations . I understand than this should be too much work to integrate this use case . Maybe you could add this limitation (only one ecu by integration) in the readme ?
A last question: 
I wonder if , by copying this integration into a new one (by changing the name), and , by changing the domain value in the const.py file would allow me to have a workaround in my case?

HAEdwin, 2022-12-22:
That might work because settings are then hopefully stored under the other domain. You can see the settings file by going to the terminal screen then type: cd config > cd .storage > cat core.config_entries but be carefull !!! If you change or remove values you might end up with an inaccessible HA. Make sure you make a full backup before experimenting so you can restore easy from the HA console prompt.

syl2222, 2022-12-22:
Thank you for everything. 
I close this issue as, this is not an issue but a limitation which is excepted 

FYi: It seem to working good with the workaround :copy/paste old folder to new name , change Domain in const.py and manifest.json. Now i have on ecu on legacy integration and one Ecu on this duplicate integration.
On core.config_entries , id are different , so ... cross-finger than it will continue working for next days/month/years ! If not, I would search for ecu-r because it's frustrating by don't see everything

closed 2022-12-22 by syl2222