ksheumaker/homeassistant-apsystems_ecur #152: warning in Home Assistant

MrLivelong, 2023-02-02:
Today I have got a warning in Home Assistant:

Logger: homeassistant.helpers.frame
Source: helpers/frame.py:77 
First occurred: 21:55:47 (2 occurrences) 
Last logged: 21:55:56

Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for apsystems_ecur using this method at custom_components/apsystems_ecur/__init__.py, line 187: hass.config_entries.async_setup_platforms(config, PLATFORMS)
Detected integration that called async_setup_platforms instead of awaiting async_forward_entry_setups; this will fail in version 2023.3. Please report issue to the custom integration author for hacs using this method at custom_components/hacs/__init__.py, line 171: hass.config_entries.async_setup_platforms(

I guess something needs to be changed, but I have no idea what. Could you please fix this?

btw: thanks for all your work! I am using your integration and watching the stats of my PV every day through HA.

HAEdwin, 2023-02-03:
Thanks for noticing, I'll fix it asap.

HAEdwin, 2023-02-03:
Fixed in release 1.2.27. Might any issue rise, please let me know by submitting a new issue.

closed 2023-02-03 by HAEdwin