ksheumaker/homeassistant-apsystems_ecur #153: Release APSystemsLibrary on PyPi

spacemanspiff2007, 2023-02-03:
Hi, I just bought an Ecu-B and I'd like to use part of your library.
Since I am not running HA I'll be writing an Ecu-B to MQTT wrapper.
I'd like to publish it on PyPi so it's easy for other users to reuse the wrapper.
Would you be willing to separate the communication part (to the ECU-B) from the HA integration part and publish it as a package on PyPi? That way I could use it as a dependency in the mqtt wrapper.

HAEdwin, 2023-02-03:
Can this fork help you out a little? https://github.com/Doudou14/Domoticz-apsystems_ecu

HAEdwin, 2023-02-24:
@spacemanspiff2007 I'm closing this topic as it's not really a problem just a question. If you have any more questions, please visit the discussions section.

closed 2023-02-24 by HAEdwin