ksheumaker/homeassistant-apsystems_ecur #156: Support for ECU-3

rohrsh, 2023-02-08:
Hi there

I have an old ECU-3 device, installed by "ZEN Energy" in Australia. It looks old, maybe 2014 era. 

It is connected via ethernet, has a webpage, and sends data to the APS cloud servers. 

But I don't get any response when I try the netcat command from the readme. 

Happy to help with any diagnostics?

Cheers

[image]

[image]

[image]

HAEdwin, 2023-02-08:
The ECU-3 is not compatible. I've once thought of using a proxy method to capture the data being sent to EMA. This would be great for compatibility but will require that users can do DNS-rewrites to enable traffic to EMA to pass HA. For example AdGuard and Pi-Hole can do DNS redirects. This would be a dependancy, besides that, some info will be missing like inverter temperature which is for some reason not sent to EMA.

rohrsh, 2023-02-11:
Would you believe I am running pi-hole here .. so I'm happy to be your tester if you ever get excited about it!

I thought I could make it work [all I want is the total power generated] using the web scraper tool, but I can't wrangle the CSS. 

So I think my best shot might be the APS cloud tools.

JSteunou, 2023-02-14:
If you have easy access to a local webpage, you can totally scrap that page to get your data. I can help you with that if needed.

rohrsh, 2023-02-14:
It does have a local webpage. I tried to scrape that but never quite got it to work, so I would very much welcome any help. 

This is [home.txt](https://github.com/ksheumaker/homeassistant-apsystems_ecur/files/10731682/home.txt) source for the bottom frame, which has the url http://1.2.3.4/cgi-bin/home: 
[image]

I tried to get this into  Home Assistant with this scrape sensor.  

```
multiscrape:
  - resource: http://1.2.3.4/cgi-bin/home
    scan_interval: 3600
    sensor:
      - unique_id: solar_lifetime_generation
        name: Solar Total
        select: ".current-version > body > form > table > tr:nth-child(2) > td:nth-child(2)"
        value_template: '{{ ((value.split(" ")[0]) | replace ("KW", "")) }}'
        unit_of_measurement: "kWh"
```

But it generates this error: _"Updating failed with exception: illegal header line: bytearray(b'debug9')"_ This thread shows other folks reporting the same problem. https://community.home-assistant.io/t/error-scraping-apsystems-ecu-3-webpage/298242/6

Thanks for any help!

JSteunou, 2023-02-14:
It seems the page headers cannot be parsed by *multiscrape* plugin. Others solve that by using AppDaemon and manual BeautifulSoup parser. Did you try to log the returned page with `log_response` option ? Did you try to sniff your network to see the content of the request ? It might be linked to this kind of error https://github.com/danieldotnl/ha-multiscrape/issues/134  You should open a ticket there to have a better support on your issue, but I think the `httpx` versus `request` python package is a good start.

rohrsh, 2023-02-14:
Indeed... 

Postman says "Error: Parse Error: Invalid header token"
Curl says " (8) Header without colon"

HAEdwin, 2023-02-18:
I've the proxy up and running as an integration. Basically it works, but there is a lot of back and forth communication between the ECU en EMA site currently being captured and to be analysed. Todo: a translation has to be made from the data, checks have to be filtered out and sensors have to be filled with data. Work in progress, because I'm pretty busy with all sorts of other things it may take a while. It's nice to see the ECU is happy with the responses though. In the past I've seen the ECU going bananas because it could not receive the right response.

TFHFony, 2023-02-20:
I'd also be interested in getting the old ECU working in Home Assistent.

[image]

 If you need some help testing: Feel free to ask. Might need some help though.

HAEdwin, 2023-02-20:
11 inverters - that a lot! I'll let the proxy run for 24 hours and if all is well I'll open the github repository so we can continue there. Hopefully one translation tempate will apply to all inverters.

HAEdwin, 2023-03-07:
I'm pretty busy so I can't do much with it right now. I need help with the data interpretation and the further development to sensors. I'm currently running this proxy 24/7 alongside the integration and it works super stable. https://github.com/HAEdwin/APsystems_ECU2EMA_proxy

jeeshofone, 2023-07-15:
I came across this post while looking for any updates to the APC-3 status of this repo. I have just pushed my unshared code on how I'm doing this with my APC-3.  I hope this helps https://github.com/jeeshofone/ha-apc-ecu-3

HAEdwin, 2023-10-29:
@rohrsh  Closing this question, this integration was intended to be as generic as possible. Hope the alternative solution from jeehofone helps you out. Maybe some day the proxy solution will become available.

closed 2023-10-29 by HAEdwin