ksheumaker/homeassistant-apsystems_ecur #157: Add consumption sensor

commonslibs, 2023-02-08:
Hi,

Ecu-c have two clamp meters. One that measures the generation of the solar panels and another that measures the consumption of the house.

It would be nice if the sensor that measures the consumption of the house could be included. Would it be possible to include that sensor?

If I need to become a tester, no problem.

Thank you so much

HAEdwin, 2023-02-08:
If the data can be called with an APS command I could add it, even without owning an ECU-C myself testers could confirm all is good when implementing. At the moment it is unknown with which APS command this data can be called.

tediroca, 2023-02-09:
Just dropping a note that I'd be happy to help with the testing too as I own an ECU-C.

tediroca, 2023-02-10:
I've tried the steps from https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/91 to grab the data from my ECU-C, and command 53 does not seem to include the consumed power (only produced):

[image]

[image]

I guess the key thing would be to find which command retrieves the consumed energy.

tediroca, 2023-02-10:
This could also be useful: http://<ecu_ip>/index.php/hidden/ . It list a set of internal URLs which provide additional info.

This is quite interesting as it seems to display a record of all the internal transmissions: http://<ecu_ip>/index.php/display/record

tediroca, 2023-02-13:
Actually the response already includes the Produced and Exported values:

Consumed:
[image]

Produced:
[image]

Consumed:
[image]

And the response:

```
{
	"APS": "ANSWER",
	"VERSION": "1.1",
	"error": 0,
	"ECUID": "<my-ecu-id>",
	"Command_Id": "53",
	"Command": "1",
	"meter_data": {
		"subtitle": "",
		"power1": [
			{
				"time": "20230213000011",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213000511",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213001011",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213001511",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213002011",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213002511",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213003011",
				"powerA": 0,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213003511",
				"powerA": 0,
				"powerB": 0,
[... 1787 lines cut ...]
				"time": "20230213123555",
				"powerA": 783,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213124055",
				"powerA": 725,
				"powerB": 0,
				"powerC": 0
			},
			{
				"time": "20230213124555",
				"powerA": 537,
				"powerB": 0,
				"powerC": 0
			}
		]
	}
}
```

Produced = meter_data > power1 >  powerA = 2156 W 
Exported = meter_data> power2 > powerA = 537 W
Consumed = Produced + Exported = 2693 W

gregoo31, 2023-03-06:
Bonjour, 
j'ai le plugin APSystem sour jeedom et le développeur a réussi à récupérer la puissance consommée.
Comment peut on faire pour HA ? 
(ne comptez pas sur moi ;-)  , je ne sais pas codé !)

HAEdwin, 2023-03-07:
I toyed with the same idea as this is probably the most stable and generic solution. Maybe you could use some of it. I'm pretty busy so I can't do much with it. I need help with the data interpretation and the further development to sensors. I'm currently running this proxy 24/7 alongside the integration and it works super stable. https://github.com/HAEdwin/APsystems_ECU2EMA_proxy

tediroca, 2023-03-08:
> I toyed with the idea to implement a proxy between the ECU and EMA website as this is probably the most stable and generic solution. Maybe you could use some of it. I'm pretty busy so I can't do much with it. I need help with the data interpretation and the further development to sensors. I'm currently running this proxy 24/7 alongside the integration and it works super stable. https://github.com/HAEdwin/APsystems_ECU2EMA_proxy

@HAEdwin I'm giving it a try. So far I've installed Adguard and followed the instructions from your link to install the proxy, although I'm getting an error after restarting HA. I've raised an issue on the proxy project itself to avoid missing stuff.

I'm new to your proxy, Adguard, APsystems and HA, so be patient please :) Is the info from the sensor supposed to appear somewhere in HA or shoul that be directly in the logs?

supertin, 2023-03-31:
I'm also interested in this.

Not sure if anyone is aware, but there's straight JSON available at http://IP/index.php/meter/old_meter_power_graph
Looks to be the same format content as @tediroca posted above... Might be useful to someone 🤷
I've been using that URL for about 12 months in a dodgy Arduino project to show the import/export on an OLED. ECU seems to struggle sometimes with it though.

tediroca, 2023-03-31:
> I'm also interested in this.
>
> Not sure if anyone is aware, but there's straight JSON available at http://IP/index.php/meter/old_meter_power_graph Looks to be the same format content as @tediroca [...]

Thats's great. Thanks for sharing!

I ended up using the internal display records from http://<ECU_IP>/index.php/display/record and a scrape sensor:

Select: body > section > div > article > div > div.panel-body > div.table-responsive > table > tbody > tr:nth-child(1) > td:nth-child(2)

Value template: {{ value | regex_findall_index('(?![\s\S]*"I")(?<="I":).*(?=,"J")')}}

commonslibs, 2023-04-12:
> > I'm also interested in this.
> > Not sure if anyone is aware, but there's straight JSON available at http://IP/index.php/meter/old_meter_power_graph Looks to be the same format content as @tediroca posted above... Might be useful to someone 🤷 I've been using that URL for about 12 months in a dodgy Arduino project to show the import/export on an OLED. ECU seems to struggle sometimes with it though.
> 
> Thats's great. Thanks for sharing!
> 
> I ended up using the internal display records from http://<ECU_IP>/index.php/display/record and a scrape sensor:
> 
> Select:
> 
> ```
> body > section > div > article > div > div.panel-body > div.table-responsive > table > tbody > tr:nth-child(1) > td:nth-child(2)
> ```
> 
> Value template:
> 
> ```
> {{ value | regex_findall_index('(?![\s\S]*"I")(?<="I":).*(?=,"J")')}}
> ```
> 
> This gives me the imported/exported energy, and based on that I enable/disable some appliances: [image]

Hi, Tediroca.

Could you tell us how to add that sensor to HA? I go over the top in HA but I don't know how to include that sensor.

On the other hand, I notice that the sensor that you have included gives you the imported/exported energy but at least I am looking for the energy that your house is consuming. Do you know how to get it?

Regards, and thank you very much.

>>>>>>>>>>>>>>>>>>>>>>>>>>> Spanish
¿Podrías indicarnos cómo añadir ese sensor a HA?. Me manejo por encima en HA pero no se cómo incluir ese sensor.

Por otro lado, observo que el sensor que has incluido te da la energia importada/exportada pero al menos yo estoy buscando la energia que esta consumindo tu casa. ¿Sabes cómo obtenerla?

Un saludo y muchas gracias.

wildekek, 2023-11-21:
I'm in the process of creating power and energy sensors for HA from the CT clamps.
So far I have all the power sensors I want, the energy sensors need a bit more work.
[Here's a gist](https://gist.github.com/wildekek/4e19fb049eb7cea3a35fe801e3ed8f57).

wildekek, 2023-11-21:
> If the data can be called with an APS command I could add it, even without owning an ECU-C myself testers could confirm all is good when implementing. At the moment it is unknown with which APS command this data can be called.

I'd be happy to help you test implementing this via the TCP socket. Here's a dump of the ECU-C with CT clamps enabled:
```
Connection to 192.168.0.6 port 8899 [tcp/*] succeeded!
APS1100160001END
APS120113000121500001550901`�j #!	Q4

10015ECU-C-Z_C1.2.10013America/Aruba��5�`Ũy��00END
```
Maybe we can chat on discord and debug this in realtime.

wildekek, 2023-11-21:
Found the CT energy meters:
http://ip/index.php/display/historical_data/meter_lifetime_energy/
Chip 1 = Grid import/export
Chip 2= Solar production
I'll add this data to my gist above in a minute, then you can use it for the HA Energy Dashboard.

HAEdwin, 2023-11-21:
> Connection to 192.168.0.6 port 8899 [tcp/*] succeeded!
> APS1100160001END
> APS120113000121500001550901`�j #!	Q4
> 
> 10015ECU-C-Z_C1.2.10013America/Aruba��5�`Ũy��00END
> ```
> 
> Maybe we can chat on discord and debug this in realtime.

This queries the ECU:
[image]

wildekek, 2023-11-22:
What I'm looking for (ordered by prio) is:
* Energy data to use in the HA Energy Dashboard
  * "Grid consumption": Needs a sensor with the total, non-negative, energy consumed. 
  * "Return to grid": Needs a sensor with the total, non-negative, energy produced.
  * "Solar production": Currently I use the "ECU Lifetime Energy" sensor, which is great, but the CT clamps are a bit more accurate, since they measure the net energy after losses, instead of the sum of the reported inverters energy.
* Power data, which I use to create a Sankey diagram to monitor real-time consumption. This is solved by implementing [my comment](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/157#issuecomment-1820961476)

> This queries the ECU: (image) But it's not what you are looking for. 

Thanks for the mapping table, this could help give clues to find the "energy consumption total" in the data the socket returns.

> How is the ECU-C being configured? Only by web UI or also with a telephone app?

Both app and web UI. 

> So this solution is not what you are looking for? [#91 (reply in thread)](https://github.com/ksheumaker/homeassistant-apsystems_ecur/discussions/91#discussioncomment-6582176)

This solution returns the last measured power per day, not energy. There is however a way to get the energy per day in the same way (http://<ECU_IP>/index.php/meter/old_meter_energy_graph), but since it is both a net meter (can go up and down) and daily (so it resets to 0 at night) it is really hard to get a reliable total production and consumption meter (which the Energy Dashboard requires) from that data. Could not find a rest interface for the totals unfortunately. The [web page in my comment above](https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues/157#issuecomment-1821121437) does provide this, so this with a few scrape sensors is a reasonable way to solve the need and create a reliable energy dashboard.
Having said that, I would much rather have the values come from the socket, so we only have to query once, I can make a PR and everyone with CT clamps gets this.

wildekek, 2023-11-22:
Updated [my Gist](https://gist.github.com/wildekek/4e19fb049eb7cea3a35fe801e3ed8f57). 

It now also contains energy sensors for use in the [Energy Dashboard of Home Assistant](https://demo.home-assistant.io/#/energy).
- **Grid Energy Consumed**: You can use this sensor in the "Grid consumption" section
- **Grid Energy Produced**: You can use this sensor in the "Return to grid" section
- **Solar Energy Produced**: You can use this sensor in the "Solar production" section

All these sensors are have thestate_class: total_increasing. This means they contain positive, increasing values and will never reset. Have fun with your new Energy Dashboard!

supertin, 2023-11-23:
> Updated [my Gist](https://gist.github.com/wildekek/4e19fb049eb7cea3a35fe801e3ed8f57).

If I was to copy this for my own use, can you think of what values I'd need to change? Obviously the ECU IP will likely be different. I also see ECU ID which I imagine needs to be changed... Any others?

wildekek, 2023-11-23:
> > Updated [my Gist](https://gist.github.com/wildekek/4e19fb049eb7cea3a35fe801e3ed8f57).
> 
> If I was to copy this for my own use, can you think of what values I'd need to change? Obviously the ECU IP will likely be different. I also see ECU ID which I imagine needs to be changed... Any others?
Nope, that's it!

nTildeStudio, 2023-11-23:
Hi! sorry for the beginner question, but I'm trying to apply that YAML file and I'm not able to do it. Could someone explain to me how to do it? I've looked into everything I could in the Home Assistant documentation (I've tried including it in configuration.yaml in all possible ways). Thanks in advance.

wildekek, 2023-11-23:
No problemo.
The way I do it (and would recommend you do this too), is to use [packages](https://www.home-assistant.io/docs/configuration/packages/): 
- Copy this file to `/homeassistant/packages/ecu.yaml`
- And then add the following lines to the `/homeassistant/configuration.yaml`:
```
homeassistant:
  packages:
    system: !include packages/ecu.yaml
```
If this does not work, I would recommend checking the logs and removing some of the contents to see what is causing the error.

nTildeStudio, 2023-11-23:
It worked!! Thank you very much!!!!

didierst, 2023-11-28:
@wildekek

You are the boss 👍 
Of the different solutions tested so far, yours is probably the most accomplished and functional.
Thank you for this work and this much appreciated sharing

BR

didierst, 2023-11-29:
Indeed, there is reason to be a little confused.

It is possible to receive a lot of data via Modbus, but I am not sure it is possible to get the values of the CT, they are however "essential".

However, with the current script, I have a strange behavior, at night, the system records a very light solar output 0.01KW while this value is 0 in the EMA APP.
For some reason I don’t know the entity: sensor.solar_power_produced_l1 gives the value of 0.011

Any idea what that behavior was?

[image]

[image]

wildekek, 2023-11-29:
> However, with the current script, I have a strange behavior, at night, the system records a very light solar output 0.01KW while this value is 0 in the EMA APP. For some reason I don’t know the entity: sensor.solar_power_produced_l1 gives the value of 0.011
> 
> Any idea what that behavior was?
> 
I have the same issue, but for me the value is 0,002kW. The reason is that the ECU rest API still really does report this value at night:
```
{"time":"20231129060113","powerA":2,"powerB":0,"powerC":0},{"time":"20231129060613","powerA":2,"powerB":0,"powerC":0},{"time":"20231129061113","powerA":2,"powerB":0,"powerC":0},{"time":"20231129061613","powerA":2,"powerB":0,"powerC":0},{"time":"20231129062113","powerA":2,"powerB":0,"powerC":0},{"time":"20231129062613","powerA":2,"powerB":0,"powerC":0},{"time":"20231129063113","powerA":2,"powerB":0,"powerC":0},{"time":"20231129063613","powerA":2,"powerB":0,"powerC":0},{"time":"20231129064113","powerA":2,"powerB":0,"powerC":0},{"time":"20231129064613","powerA":2,"powerB":0,"powerC":0},{"time":"20231129065113","powerA":0,"powerB":0,"powerC":0},{"time":"20231129065613","powerA":0,"powerB":33,"powerC":20},
```
This is really just a calibration issue in the ECU, so I don't really want to fix that in my scripts, since it can vary per ECU.  There are CT clamp calibration settings in the ECU at http://[ip]/index.php/configuration/meter, but I don't feel comfortable editing those myself. I learned to live with it, but if you are able to get the values calibrated I'd love to hear it.

didierst, 2023-11-29:
Thank you for the feedback, it would be interesting to get some feedback from other users to determine the importance / average of "drift" in the worst case, we could via the code, consider that anything below the drift value is considered to be 0

Calibration requires a constant and calibrated source of voltage/current, which is not within everyone’s reach, I guess APS use a filter that probably only considers a value above a certain threshold to be considered "real"

That said, the script works very well as it is, and you can easily live with this micro production at night, we only have to say that it is induced by the light of the full moon :)

didierst, 2023-11-29:
{"time":1701213125000,"powerA":11,"powerB":0,"powerC":0},{"time":1701213427000,"powerA":11,"powerB":0,"powerC":0},{"time":1701213728000,"powerA":11,"powerB":0,"powerC":0},{"time":1701214029000,"powerA":11,"powerB":0,"powerC":0}

didierst, 2023-11-29:
MODBUS via Nodered 

[image]

chuppito, 2024-07-27:
Hello, I am interested in integrating consumption and production values. Have you managed to do it? Thank

mtthidoteu, 2024-08-01:
> Hello, I am interested in integrating consumption and production values. Have you managed to do it? Thank

read above :)

HAEdwin, 2025-05-02:
@mtthidoteu @chuppito 
Solved in latest version @ https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader

closed 2025-05-02 by HAEdwin