ksheumaker/homeassistant-apsystems_ecur #159: Is the QT2 supported?

TheRedBull205, 2023-02-15:
I am currently looking into solar panels in combination with a microinverter. But this also needs to work with Home Assistant.

Is the new QT2 also supported by this project in combination with the ECU-R?
https://emea.apsystems.com/nl/portfolio-item/apsystems-qt2/

HAEdwin, 2023-02-18:
It might but I cannot guarantee it. We've seen other inverter models being added to this integration, these needed adoption in the code and so after a while the integration was made more generic to suit more models automatically.

Slamkryp, 2023-03-01:
Yes, i can confirm qt2 is working. just type in the ecu-r ip and wifi password and its ready to go!  (only tested with 1 qt2)

HAEdwin, 2023-03-25:
Thanx for confirmation @Slamkryp

closed 2023-03-25 by HAEdwin